zwave-js/zwave-js #4: Weiterentwicklung des Adapters

husky-koglhof, 2018-11-21:
Hallo, hab mir mal deinen Source geladen und ein wenig mit rumgespielt.
Kann schon Temperaturen auf meinen Danfoss setzen.

Jetzt aber die Frage, hast du bereits eine Weiterentwicklung deines Codes oder soll
ich meinen Teil mal sauber und einen Pull Request machen.

LG Christian

asciidisco, 2018-11-21:
Ich wäre da Prinzipiell auch interessiert dran, bzw. was zu contributen.
Hab ne Handvoll Z-Wave Geräte Zuhause (nutze an sich eher andere Protokolle), wäre aber durchaus daran interessiert das mal auszubauen. Dazu brauchts aber ne non Openzwave Lösung für mich.

AlCalzone, 2018-11-21:
Ich hab lange nichts mehr gemacht, würde das Projekt aber gerne voran treiben. Was vor allem noch fehlt ist das Node-Interview beim Starten, da wollte ich mir das beste aus OZW und OpenHAB zusammenklauen.
Wenn du Verbesserungen hast, gerne einen PR erstellen, dann schaue ich mir das an.

AlCalzone, 2018-11-22:
Ein paar Hinweise:
* https://github.com/openhab/org.openhab.binding.zwave/tree/master/src/main/java/org/openhab/binding/zwave/internal/protocol ist ein guter Startpunkt - ich weiß leider gerade nicht mehr in welcher Klasse das definiert war. 
* https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Node.cpp#L238 hier beginnt in OZW das Interview
* https://github.com/AlCalzone/node-zwave-js/blob/master/src/lib/node/Node.ts#L287 definiert die Interview-Reihenfolge wie in OZW, OpenHab scheint es kürzer hin zu bekommen - daran sollten wir uns orientieren.

AlCalzone, 2019-03-27:
Habe inzwischen einiges ergänzt und überarbeitet. Das Interview ist deutlich vollständiger (Security, Associations und Configuration fehlt noch, sowie die Abfrage einiger statischer und dynamischer Werte).
Und die Änderungen aus Husky's Fork sind nun sauber implementiert:
* BatteryCC
* ClimateControlScheduleCC
* Reading and writing floats
* ManufacturerSpecificCC
* MultiLevelSensorCC (needs verification)
* ThermostatSetpointCC

closed 2019-04-25 by AlCalzone