zwave-js/zwave-js #6: [Tracking] Implementation status of Command Classes

AlCalzone, 2019-03-19:
The following command classes are not implemented or incomplete:
(generated with `npm run implemented_ccs -- --onlyIncomplete --flavor=github`)

|  | Command class name | Implemented version | max. |
| --- | --- | --- | --- |
| ✗ | Alarm Sensor | not implemented | 1 |
| ✗ | Alarm Silence | not implemented | 1 |
| ✗ | All Switch | not implemented | 1 |
| ✗ | Anti-theft | not implemented | 2 |
| ✗ | Application Capability | not implemented | 1 |
| ✗ | Application Status | not implemented | 1 |
| ✗ | Association | not implemented | 2 |
| ✗ | Association Command Configuration | not implemented | 1 |
| ✗ | Association Group Information (AGI) | not implemented | 3 |
| ✗ | Barrier Operator | not implemented | 1 |
| ✗ | Basic Tariff Information | not implemented | 1 |
| ✗ | Basic Window Covering | not implemented | 1 |
| ✗ | Binary Toggle Switch | not implemented | 1 |
| ✗ | Climate Control Schedule | not implemented | 1 |
| ✗ | Clock | not implemented | 1 |
| ✗ | Color Switch | not implemented | 3 |
| ✗ | Configuration | not implemented | 4 |
| ✗ | Controller Replication | not implemented | 1 |
| ✗ | CRC-16 Encapsulation | not implemented | 1 |
| ✗ | Demand Control Plan Configuration | not implemented | 1 |
| ✗ | Demand Control Plan Monitor | not implemented | 1 |
| ✗ | Device Reset Locally | not implemented | 1 |
| ✗ | Door Lock | not implemented | 4 |
| ✗ | Door Lock Logging | not implemented | 1 |
| ✗ | Energy Production | not implemented | 1 |
| ✗ | Entry Control | not implemented | 1 |
| ✗ | Firmware Update Meta Data | not implemented | 5 |
| ✗ | Geographic Location | not implemented | 1 |
| ✗ | Grouping Name | not implemented | 1 |
| ✗ | Hail | not implemented | 1 |
| ✗ | HRV Status | not implemented | 1 |
| ✗ | HRV Control | not implemented | 1 |
| ✗ | Humidity Control Mode | not implemented | 2 |
| ✗ | Humidity Control Operating State | not implemented | 1 |
| ✗ | Humidity Control Setpoint | not implemented | 2 |
| ✗ | Inclusion Controller | not implemented | 1 |
| ✗ | Indicator | not implemented | 3 |
| ✗ | IP Association | not implemented | 1 |
| ✗ | IP Configuration | not implemented | 1 |
| ✗ | Irrigation | not implemented | 1 |
| ✗ | Language | not implemented | 1 |
| ✗ | Lock | not implemented | 1 |
| ✗ | Mailbox | not implemented | 1 |
| ✗ | Manufacturer Proprietary | not implemented | 1 |
| ✍ | Manufacturer Specific | Version 1 | 2 |
| ✗ | Support/Control Mark | not implemented | N/A |
| ✗ | Meter | not implemented | 5 |
| ✗ | Meter Table Configuration | not implemented | 1 |
| ✗ | Meter Table Monitor | not implemented | 3 |
| ✗ | Meter Table Push Configuration | not implemented | 1 |
| ✗ | Move To Position Window Covering | not implemented | 1 |
| ✍ | Multi Channel | Version 3 | 4 |
| ✗ | Multi Channel Association | not implemented | 3 |
| ✗ | Multi Command | not implemented | 1 |
| ✗ | Multilevel Sensor | not implemented | 11 |
| ✗ | Multilevel Toggle Switch | not implemented | 1 |
| ✗ | Network Management Basic Node | not implemented | 2 |
| ✗ | Network Management Inclusion | not implemented | 3 |
| ✗ | Network Management Installation and Maintenance | not implemented | 2 |
| ✗ | Network Management Primary | not implemented | 1 |
| ✗ | Network Management Proxy | not implemented | 2 |
| ✗ | Node Naming and Location | not implemented | 1 |
| ✗ | Node Provisioning | not implemented | 1 |
| ✗ | Powerlevel | not implemented | 1 |
| ✗ | Prepayment | not implemented | 1 |
| ✗ | Prepayment Encapsulation | not implemented | 1 |
| ✗ | Proprietary | not implemented | 1 |
| ✗ | Protection | not implemented | 2 |
| ✗ | Pulse Meter | not implemented | 1 |
| ✗ | Rate Table Configuration | not implemented | 1 |
| ✗ | Rate Table Monitor | not implemented | 1 |
| ✗ | Remote Association Activation | not implemented | 1 |
| ✗ | Remote Association Configuration | not implemented | 1 |
| ✗ | Scene Activation | not implemented | 1 |
| ✗ | Scene Actuator Configuration | not implemented | 1 |
| ✗ | Scene Controller Configuration | not implemented | 1 |
| ✗ | Schedule | not implemented | 4 |
| ✗ | Schedule Entry Lock | not implemented | 3 |
| ✗ | Screen Attributes | not implemented | 2 |
| ✗ | Screen Meta Data | not implemented | 2 |
| ✗ | Security | not implemented | 1 |
| ✗ | Security 2 | not implemented | 1 |
| ✗ | Security Mark | not implemented | N/A |
| ✗ | Sensor Configuration | not implemented | 1 |
| ✗ | Simple AV Control | not implemented | 4 |
| ✗ | Sound Switch | not implemented | 1 |
| ✗ | Supervision | not implemented | 1 |
| ✗ | Tariff Table Configuration | not implemented | 1 |
| ✗ | Tariff Table Monitor | not implemented | 1 |
| ✗ | Thermostat Fan Mode | not implemented | 4 |
| ✗ | Thermostat Fan State | not implemented | 2 |
| ✗ | Thermostat Mode | not implemented | 3 |
| ✗ | Thermostat Operating State | not implemented | 2 |
| ✗ | Thermostat Setback | not implemented | 1 |
| ✗ | Thermostat Setpoint | not implemented | 3 |
| ✗ | Time | not implemented | 2 |
| ✗ | Time Parameters | not implemented | 2 |
| ✗ | Transport Service | not implemented | 2 |
| ✗ | User Code | not implemented | 2 |
| ✗ | Window Covering | not implemented | 1 |
| ✗ | Z/IP | not implemented | 4 |
| ✗ | Z/IP 6LoWPAN | not implemented | 1 |
| ✗ | Z/IP Gateway | not implemented | 1 |
| ✗ | Z/IP Naming and Location | not implemented | 1 |
| ✗ | Z/IP ND | not implemented | 1 |
| ✗ | Z/IP Portal | not implemented | 1 |

jprates, 2021-02-05:
Aeotec Ltd. Bulb 6 Multi-Color (ZWA002) implementation for changing color is not working.
Please see [this post](https://community.home-assistant.io/t/zwave-js-cant-change-aeotec-zwa002-led-lamp-color/277134) for further details including a log snapshot.

AlCalzone, 2021-02-05:
Could you make a new issue with that? This is off-topic.