zwave-js/zwave-js #291: Typescript error in definitions - Strict mode compatibility?

RoboPhred, 2019-09-14:
I am getting an error on build trying to use this in a project with strict mode turned on

```
node_modules/zwave-js/build/lib/driver/Driver.d.ts:34:22 - error TS2420: Class 'Driver' incorrectly implements interface 'IDriver'.
  Property 'options' is missing in type 'Driver' but required in type 'IDriver'.

34 export declare class Driver extends EventEmitter implements IDriver {
                        ~~~~~~

  node_modules/zwave-js/build/lib/driver/IDriver.d.ts:13:5
    13     options: ZWaveOptions;
           ~~~~~~~
    'options' is declared here.
```

I am using typescript 3.6.3.  My tsconfig is
```
{
  "compilerOptions": {
    "target": "es6",
    "lib": ["es6"],
    "module": "commonjs",
    "moduleResolution": "node",
    "esModuleInterop": true,
    "experimentalDecorators": true,
    "stripInternal": true,
    "strict": true,
    "forceConsistentCasingInFileNames": true,
    "outDir": "../lib",
    "incremental": true,
    "declaration": false,
    "sourceMap": true
  }
}
```

AlCalzone, 2019-09-15:
I'll check as soon as I'm back from vacation

AlCalzone, 2019-09-25:
Fixed and should soon be released as v1.3.1

commit d5b49915 (2019-09-25): fix: mark `options` in `IDriver` internal

fixes: #291
closed 2019-09-25 by AlCalzone