zwave-js/zwave-js #398: node "value updated" event not called when GE switch status changes

RoboPhred, 2019-11-08:
I have a GE wall toggle switch I am experimenting with.  I can change its value reliably, but when I use the switch manually, I do not get any "value updated" event on the node.

I am aware that some devices do not transmit when their values change, but in this case I believe this particular switch is doing so, as every time I change its state I get log output:

```
>18:04:57.724 SERIAL « 0x0110004984020a04100125277370867277bd                              (18 bytes)
18:04:57.725 SERIAL   Buffer := 0x0110004984020a04100125277370867277bd                    (18 bytes)
18:04:57.727 SERIAL » [ACK]                                                                   (0x06)
18:04:57.729 CNTRLR « [Node 002] Received updated node info
18:04:57.731 SERIAL   The receive buffer is empty, waiting for the next chunk...
```

In this case, node 2 is indeed the switch I am toggling, so something is being transmitted in response to the changing of the switch value,

I believe the switch is a "GE Z-Wave In-Wall Smart Light Switch", and it is connecting through an Aeotech Gen 5 usb receiver.

AlCalzone, 2019-11-08:
The log output you have shown is just a node information frame (NIF), which many devices send when they are woken up. 
It might be that the controller needs to be assigned to the Lifeline association group to receive those updates. Is your device a Z-Wave+ device? Those should be automatically configured. Can you send me an excerpt of the cache file (`cache/<something hexadecimal>.json`)? The part under node 2 should be enough.

RoboPhred, 2019-11-09:
It's an older switch so its possible it doesn't.  It definitely sends an update every time I press the switch though, so perhaps that was throwing me off.

```js
        "2": {
            "id": 2,
            "interviewStage": "Complete",
            "deviceClass": {
                "basic": 4,
                "generic": 16,
                "specific": 1
            },
            "isListening": true,
            "isFrequentListening": false,
            "isRouting": true,
            "maxBaudRate": 40000,
            "isSecure": false,
            "isBeaming": true,
            "version": 4,
            "commandClasses": {
                "0x25": {
                    "name": "Binary Switch",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 1,
                    "values": [
                        {
                            "endpoint": 0,
                            "propertyName": "interviewComplete",
                            "value": true
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "currentValue",
                            "value": false
                        }
                    ]
                },
                "0x27": {
                    "name": "All Switch",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 0
                },
                "0x70": {
                    "name": "Configuration",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 1
                },
                "0x72": {
                    "name": "Manufacturer Specific",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 1,
                    "values": [
                        {
                            "endpoint": 0,
                            "propertyName": "interviewComplete",
                            "value": true
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "manufacturerId",
                            "value": 99
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "productType",
                            "value": 18770
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "productId",
                            "value": 12338
                        }
                    ]
                },
                "0x73": {
                    "name": "Powerlevel",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 0
                },
                "0x77": {
                    "name": "Node Naming and Location",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 0
                },
                "0x86": {
                    "name": "Version",
                    "isSupported": true,
                    "isControlled": false,
                    "version": 1,
                    "values": [
                        {
                            "endpoint": 0,
                            "propertyName": "interviewComplete",
                            "value": true
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "libraryType",
                            "value": 6
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "protocolVersion",
                            "value": "3.40"
                        },
                        {
                            "endpoint": 0,
                            "propertyName": "firmwareVersions",
                            "value": [
                                "3.27"
                            ]
                        }
                    ]
                }
            }
        },
```

AlCalzone, 2019-11-10:
Ok that's definitely something I need to handle. Thanks for the info

AlCalzone, 2019-11-10:
@RoboPhred just to be sure - is there anything else in the log related to this node when you press a switch?

AlCalzone, 2019-11-10:
@RoboPhred I created a PR with a potential fix. Please change your dependency to `AlCalzone/node-zwave-js#refreshValues` and try with that version. You will have to install devDependencies and build the sources aswell:
```
cd node_modules/zwave-js
npm i
npm run build
```

The driver now checks when it receives a NIF if the node supports Z-Wave+. If not, it assumes that it won't receive updates from that node and refreshes all dynamic values, including the switch state. That logic will need some fine-tuning but I need to verify that this is somewhat correct behavior.

RoboPhred, 2019-11-24:
Installing that branch gives me a package that contains no code, just a config folder and package file.
NPM version 6.10.2

My guess is its being affected by the files entry in your package.json
```
  "files": [
    "build/**/*.js",
    "build/**/*.d.ts",
    "config/"
  ],
```

AlCalzone, 2019-11-25:
Ok then you'll need a bit more patience. I'm working on a rather large update that will a lot of configuration issues for a lot of devices. Afterwards, I'll include this PR.

AlCalzone, 2019-11-26:
I'll release version 2.0 soon. If you could test if that solves your problem, I would be happy.

RoboPhred, 2019-12-01:
I don't think this is entirely working.  With version 2.0.1, I cannot get any value updated events on this switch, and `.getValue({commandClass: 0x25, endpoint: 0, property: "currentValue"})` only updates if I manually send a .get command on the command class.

When I change the switch manually, the following log output is generated:
```
10:37:34.262 SERIAL « 0x01                                                                 (1 bytes)
10:37:34.263 SERIAL   [incomplete] Buffer := 0x01                                          (1 bytes)
10:37:34.264 SERIAL « 0x10004984020a04100125277370867277bd                                (17 bytes)
10:37:34.264 SERIAL   Buffer := 0x0110004984020a04100125277370867277bd                    (18 bytes)
10:37:34.283 SERIAL » [ACK]                                                                   (0x06)
10:37:34.283 CNTRLR « [Node 002] Received updated node info
10:37:34.284 SERIAL   The receive buffer is empty, waiting for the next chunk...
```
No other effect occurs aside from this.  The `value updated` event never fires, and getValue shows no new value being received.

AlCalzone, 2019-12-02:
Alright, thanks for testing. I'll add some more logging specifically for this case. Once that is in, I'll let you know.

AlCalzone, 2019-12-08:
I think the problem is that I wasn't refreshing the values on every NIF, but only when the node state switches from asleep to awake.
I have released a beta version for you, where the the values are refreshed for every NIF, so hopefully that finally resolves this issue.
`v2.1.2-beta.0` will be available shortly, please test with that version.

AlCalzone, 2019-12-11:
Hey @RoboPhred did you already have an opportunity to test `v2.1.2-beta.0`?

AlCalzone, 2019-12-13:
FYI: I've released this as part of `v2.2.0`

AlCalzone, 2020-01-02:
Closing for housekeeping purposes. Please reopen if this continues to be an issue

PR #408 (2019-11-26): On receipt of NIF, refresh values if node does not support Z-Wave+
fixes: #398
closed 2020-01-02 by AlCalzone