zwave-js/zwave-js #414: Import OpenZWave device config DB

AlCalzone, 2019-11-13:
Like #411 but with the files from the OpenZWave repo.

robertsLando, 2020-11-02:
@AlCalzone Could we speak about this? I could create a script that parses the xml and creates the desired JSON files but I would need some help to understand the format and what you expect, also I think that we would need to create a github action that runs scheduled to check for updated config files

AlCalzone, 2020-11-02:
I'll see if I can spare some time later

robertsLando, 2020-11-03:
For reference: https://github.com/zwave-js/node-zwave-js/blob/6274e4b34bb9f34d7687b12a994ee71574c62ae6/maintenance/importConfig.ts#L328

robertsLando, 2020-11-03:
OZW Manufacturer specific: https://github.com/OpenZWave/open-zwave/blob/master/config/manufacturer_specific.xml
Config files folder: https://github.com/OpenZWave/open-zwave/tree/master/config

IMO we could use [xml2json](https://www.npmjs.com/package/xml2json) to read manufacturer.xml and then for each manufacturer all it's products (the product file path is specified there). Also the product xml should be readed using xml2json

AlCalzone, 2020-11-03:
Are these notes for you or questions for me? ;)

robertsLando, 2020-11-03:
Both 😂 What do you think about it?

AlCalzone, 2020-11-04:
Sounds good so far

robertsLando, 2020-11-04:
@AlCalzone I will try to write some code today, would you like me to write it that file or create a new one? How do you run that script actually?

AlCalzone, 2020-11-04:
I run it through npm (lerna) commands
* `lerna run config -- manufacturers` downloads the manufacturer list from the OpenHAB config
* `lerna run config -- download` downloads the entire OpenHAB device config DB
* `lerna run config -- download deviceid` downloads the OpenHAB config for a single device
* `lerna run config -- import` imports previously downloaded config files (so you can delete or change them before importing)
* `lerna run config -- index` updates the config file index
* `lerna run lint:config` checks the config files for errors and common problems

We should probably separate the commands for OpenHAB and OZW imports or add an additional param to distinguish. Feel free to work in that file, as it all belongs together IMO.

robertsLando, 2020-11-04:
@AlCalzone Nice! I think that list of commands would be good to place somewhere in developers docs :)

robertsLando, 2020-11-04:
Just for reference:

`lerna run <script>` runs `npm run <script>` in all `/packages` that contains `<script>` in their `package.json`, in this case it runs [this](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/package.json#L48)

AlCalzone, 2020-11-04:
Correct - there is only one package with the `config` command, so it just runs the one.
It would probably make sense to refactor this - after all the script is in the root `maintenance` folder, so we could just put the script into the root package.json

robertsLando, 2020-11-04:
Yes I think we could make it `npm run config` and move it to main `package.json`

robertsLando, 2020-11-04:
@AlCalzone about my idea, If you want the download of ozw things to be made by the script we should use [download](https://www.npmjs.com/package/download) that's a handy script that downloads files and also allows to decompress them. Using that we could download the openzwave `tar.gz` available from https://github.com/OpenZWave/open-zwave/archive/master.tar.gz and make the parsing

Alternatively we could simply add the parsing script there and make the download from github action using svn like my previous example  what would you prefer?

AlCalzone, 2020-11-04:
I already have `axios` in the devDependencies, so there's no need for another download utility. Decompressing could be done by executing the system utility `tar`, which should be available on all relevant platforms (Unix and Windows)

robertsLando, 2020-11-04:
Ok so we could use axios for the download and the command `tar -xzf master.tar.gz open-zwave-master/config  --strip-components=1` to extract the config files from the archive. I would create a `.tmp` folder to store downloaded config files that can be empty once finished

PR #1093 (2020-11-11): feat(config): add support for importing the OZW devices database
fixes: #414
closed 2020-11-11 by AlCalzone