zwave-js/zwave-js #476: Interview issues

CliffS, 2019-12-10:
So pleased to have found this library after struggling with openzwave-shared. Thank you @AlCalzone for all the hard work you've put in on it.  However I have a question regarding node interviewing.

I have various types of device.  The controller and wall sockets report `interview completed` after a short while.  However I have door sensors an movement sensors that never report completed.  The cache file shows the `interviewStage` as `ProtocolInfo`.

I have tried setting `node.keepAwake = true` on each node and then triggering the sensors but I still don't see `interview completed`.

Am I doing something wrong?

AlCalzone, 2019-12-10:
The log output might show what's happening. Can you redirect that to a file and send it to me?

CliffS, 2019-12-10:
Hmm... Winston seems to be suppressing output if it's not directed to a terminal.

AlCalzone, 2019-12-10:
That behavior is in this library. I haven't found a good solution yet. Basically I want to suppress the output when it won't be seen anywhere.

You can trick it by setting the `NODE_ENV` environment variable to "test".

CliffS, 2019-12-10:
Setting `NODE_ENV` to "test" suppresses all logging.  See `shared.ts` line 239.

However, I did managed to get it to log with:

```typescript
export function createConsoleTransport(): Transport {
	return new winston.transports.File({
		level: transportLoglevel,
                filename: 'winston.log'
          });
}
```

CliffS, 2019-12-10:
Please see attached.  For completeness I've attached the output, the cache file and my test code.

For reference, the types are:

1. Controller
2.  Movement sensor
3.  Door sensor
4. -
5. Wall socket
6. -
7. -
8. Not in use (needs deleting)
8. Not in use (needs deleting)
10. -
11. Wall switch
12. Wall socket

I can give you more precise details if it's useful.

Only nodes 1, 5, 11 and 12 have been interviewed correctly, despite triggering the three sensors.  11 interviewed after a button was pressed.

[winston.log](https://github.com/AlCalzone/node-zwave-js/files/3946814/winston.log)
[f943537f.zip](https://github.com/AlCalzone/node-zwave-js/files/3946819/f943537f.zip)
[zw.zip](https://github.com/AlCalzone/node-zwave-js/files/3946827/zw.zip)

Thanks again for your help.

AlCalzone, 2019-12-10:
Ok, so what I can see at a first glance:
* Nodes 2, 3 are asleep, but never send a wake up notification, despite reporting changes. Can you try waking them up manually a couple of times?
* Nodes 4 and 6 seem to be missing completely (unknown to the controller), but I guess you meant that by that symbol.
* Node 7 never got the ping because it was dropped by the controller --> I should add some fallback for this case to https://github.com/AlCalzone/node-zwave-js/issues/403

CliffS, 2019-12-11:
Nodes 2, 3 and 7 never respond as interviewed, even when triggered multiple times during the start-up sequence.  Nodes 4, 6 and 10 were removed from the controller which does not appear to reuse these slots.

Nodes 8 and 9 also need deleting.  But I have found no way to delete them without the original devices.

Incidentally, rerunning this script multiple times, Node 11 still only interviews when I press it, even though it is shown as `"interviewStage": "Complete"` in the cache.

Any ideas how to move forward?

AlCalzone, 2019-12-11:
Let me explain the interview procedure and the concept of sleeping nodes first, maybe my answers become clearer then.

Battery powered nodes have a small window where they respond to controller messages (status *awake*). This window is announced by the node using a WakeUp message. After 10s without communication or after actively sending the node to sleep, the node won't respond (status *asleep*).
Even if a node sends us messages (sensor readings), that does **not** mean that it will respond to requests.
Mains powered nodes always respond, except if they are not reachable (status *dead*).

1. The controller "pings" every node to test if it is reachable/alive. Several things can now happen:
    1. The node is not battery powered and responds.
    2. The node is battery powered and responds. The messages to this node are prioritized so the node can go back to sleep sooner.
    3. The node is battery powered and does not respond. The node gets marked as asleep and the controller waits until the node wakes up.
    4. The node is not battery powered and does not respond. The controller marks it as dead.
2. The controller immediately interviews all nodes that are awake or not battery powered.
    1. If a node was previously interviewed, some parts of the interview process are repeated (i.e. requesting current values).
    2. Once the complete or partial interview has finished, the `interview completed` event is emitted.
3. As soon as a battery-powered node wakes up, step 2 is performed for that node.

> Nodes 2, 3 and 7 never respond as interviewed, even when triggered multiple times during the start-up sequence

I'm not sure what the issue with nodes 2 and 3 is. As I wrote, they do send some messages, but they don't send the wake up notification. This means they are never marked as awake and the interview is never performed. Can you tell me which devices they are? Maybe a startup log from OpenZWave would also help.

The problem with node 7 (at least in the log you sent me) is that the controller timed while sending the ping, so the node never got a chance to respond. This is something I need to fix in this library.

CliffS, 2019-12-11:
This is the ozwcache file from the same controller which I think will give you all the information you need (and a lot that you don't).

When openzwave-shared starts, it takes some time (around 50 seconds) but eventually everything is recognised.  FYI, I am using a version from Github:  `github:OpenZWave/node-openzwave-shared#7558a309879169ea8787405c82028ccd6e1aba26`.

[ozwcache_0xf943537f.zip](https://github.com/AlCalzone/node-zwave-js/files/3950630/ozwcache_0xf943537f.zip)

Thanks again for all your work on this (and coap/tradfri).

AlCalzone, 2019-12-11:
Thanks for the cache file, I'll have a look later. I'm also interested in the `OZW_log.txt` to see which messages are being exchanged by the nodes in question. Could you send me those too please?

CliffS, 2019-12-11:
Oops.  I had turned logging off.  But here is a short log.

[OZW_Log.zip](https://github.com/AlCalzone/node-zwave-js/files/3950923/OZW_Log.zip)

AlCalzone, 2019-12-11:
From reading the log, OZW behaves exactly like this library. Both Node 2 and 3 fail to respond to the ping, so they are marked asleep and are not interviewed (at least not during the logged period).

Looking at the [manual](https://manuals.fibaro.com/content/manuals/de/FGMS-001/FGMS-001-DE-A-v1.00.pdf), node 2 has a default wake up interval of 2 hours and must be triggered using the B key to send a node information frame (which the driver interprets as "Hi, I'm awake!"). I guess you did that when you "triggered" it. If that doesn't work, simply waiting 2 hours should do it.

CliffS, 2019-12-12:
I had not triggered node 2 with the B key, simply ensured it had seen movement.  As you predicted, running the test code for 2 hours, node 2 was successfully interviewed.

I still haven't seen the door sensors: nodes 3 and 7.  In the cache file, their `interviewStage` is still `ProtocolInfo`.

Maybe I'm worrying too much about the `interview completed`?  I was working on the basis that I needed to wait for `interview completed` on the driver before continuing.  Should I just carry on regardless?  On `openzwave-shared`, I wait for `scan complete` which takes about a minute.

AlCalzone, 2019-12-12:
> Maybe I'm worrying too much about the interview completed?

Yes and no. If the driver has interviewed the node before, it knows enough so you can work with it right away. However some values in the value DB might not be up to date yet.
If the interview stage is still `ProtocolInfo`, the node hasn't been interviewed yet. 

Maybe I should add an event `node ready` similar to `driver ready` to let library users know its safe to perform operations on the node? That event would be triggered either immediately if the node was already interviewed or after the first interview is completed. However in the case of sleeping nodes, all commands will stay in queue until the node wakes up - so I'm not sure its very useful.

Your nodes 3 and 7 have a wakeup interval of 86400s:
```xml
<Value type="int" index="0" value="86400" max="2147483647" min="-2147483648" poll_intensity="0" verify_changes="false" write_only="false" read_only="false" units="Seconds" label="Wake-up Interval" instance="1" genre="system"/>
```
so I guess you need a lot of patience for the interview to complete if you don't want to wake it up manually.

CliffS, 2019-12-12:
I think a `node ready` would be a great idea.  Also an `all nodes ready` on the driver would be useful.  Then the interview becomes far less important.  I really don't want to have to wait 24 hours after restarting the server. :)

Is there any way I can remove nodes 8 & 9 from the controller without the actual devices?

AlCalzone, 2019-12-22:
@CliffS I created issues for the events (will be in the next release) and removing the failed nodes. Is the interview working for you now?

CliffS, 2019-12-22:
@AlCalzone Yes, if I leave it long enough, it interviews.  So I think the `node ready` will solve my problems and allow me to dump OpenZwave.

Actually, I have one more problem.  I'll open a separate issue shortly.

AlCalzone, 2019-12-22:
Alright, closing this issue then :)

closed 2019-12-22 by AlCalzone