zwave-js/zwave-js #516: The command class Scene Activation is not implemented

CliffS, 2019-12-23:
The full message is `Dropping message because it could not be deserialized: The command class Scene Activation is not implemented`.

The item is a Nodon Wallswitch CWS-3-1-00

Is there any possibility that `Scene Activation`will be supported in the foreseeable future?

AlCalzone, 2019-12-29:
I can do that next 👍

AlCalzone, 2019-12-30:
Reading the specs, I don't quite understand why this CC is supposed to be supported in the controller node.

> The Scene Activation Command Class used for launching scenes in a number of actuator nodes e.g.
another scene-controlling unit, a multilevel switch, a binary switch etc.

What exactly is the intended usage? Do you want to react to the activation of a given scene ID?

CliffS, 2019-12-30:
At the server, yes, I expect to receive a code per scene and to be able to act upon it.

CliffS, 2019-12-31:
In case you're interested, my current code for the switch (after compiling to js) is:

```javascript
  WallSwitch = class WallSwitch extends Zdevice {
    constructor() {
      super(...arguments);
      this.node.on('scene set', function(scene) {
        var button, press;
        button = (function() {
          switch (Math.floor(scene / 10)) {
            case 4:
              return 'left up';
            case 3:
              return 'right up';
            case 2:
              return 'left down';
            case 1:
              return 'right down';
            default:
              throw new Error(`Unknown scene: ${scene}`);
          }
        })();
        press = (function() {
          switch (scene % 10) {
            case 0:
              return 'press';
            case 2:
              return 'long press';
            case 1:
              return 'release';
            case 3:
              return 'double';
            default:
              throw new Error(`Unknown scene: ${scene}`);
          }
        })();
        this.emit('switch', {
          button: button,
          type: press
        });
        return log.debug(`${this.name} ${button} ${press}`);
      });
    }
  };
```

PR #529 (2020-01-01): Implement Scene Activation CC
Fixes: #516
closed 2020-01-01 by AlCalzone