zwave-js/zwave-js #530: Further events

CliffS, 2020-01-01:
Currently the following events appear still to need handling and to translate into events on the `Node` object:

* `SceneActivationCCSet`: please see my notes on PR https://github.com/AlCalzone/node-zwave-js/pull/529
* `BinarySensorCCReport`: this is needed to detect movement on the Multilevel Sensor.  I am currently getting `value updated` events for temperature, illuminance and seismic intensity but not for `binary sensor`.

AlCalzone, 2020-01-01:
> BinarySensorCCReport [...] but not for binary sensor

Do you have an example of a message that does not trigger the value updated event? You can create a logfile by setting the `LOGTOFILE` environment variable to something truthy (e.g. "true").
I'm only interested in the timespan around the expected event (a couple of seconds).

CliffS, 2020-01-02:
I am now getting `value updated` for `Binary Sensor` events.  Thanks.

It seems I was running an older version of your code for this test.  Something to do with naming a git remote as `upsteam`. /sigh

Also `SceneActivationCCSet` seems to be working fine too.

I'm closing this issue.  Many thanks.

Cliff.

closed 2020-01-02 by CliffS