zwave-js/zwave-js #546: Check if we need to do additional cleanup after removeFailedNode

AlCalzone, 2020-01-06:
e.g. deleting the node, raising the all nodes ready event, ...

AlCalzone, 2020-01-07:
Fixed by #549

@CliffS I did some more changes regarding the `all nodes ready` logic (see also changelog). By default, nodes that are known to be dead (did not respond and cannot be asleep) are now ignored in the check. As soon as the status of all nodes is known and all alive nodes have been interviewed at least once, `all nodes ready` is emitted.

I did not include the check for `isFailed` yet because there's still an ordering issue for controller messages when a node is asleep

closed 2020-01-07 by AlCalzone

CliffS, 2020-01-08:
@AlCalzone This looks perfect.  It means that the `all nodes ready` event will be reliable after the initial creation of the cache.  Thanks again for all the effort you're putting into this.