zwave-js/zwave-js #552: Rework the definition process for value IDs

AlCalzone, 2020-01-08:
Some thoughts:
* There should be a central property of each CC that provides named access to all defined value IDs (or functions that return a value ID) as well as metadata 
* If possible, this property should be used in the decorators (think about this)
* It should be possible to define if this value is always accessible (e.g. writeonly values #532) and/or private
* This should replace or improve `registerValue`.

robertsLando, 2020-10-23:
What about using [msg-pack-lite](https://www.npmjs.com/package/msgpack-lite) to decode the valueId json and encode it back to valueId? We could use it's encoding as unique Id. Also we could provide  a public `getId(ValueID valueId) : String` method to calcolate a the valueId unique id

robertsLando, 2020-10-29:
Based on our slack conversation:

> Is  there a file where I can find all properties/ propertiesKeys of all command classes?

We could create a file with all properties/properties keys that actually are hardcoded in `protected [SET_VALUE]: ` 

Like https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts#L249

raman325, 2021-06-17:
As discussed in Discord, this could be used to auto generate CC documentation. Once you get there, please include whether the CC has stateless values

PR #4704 (2022-06-21): feat: implement discoverable and transparently-typed CC value definitions
This PR revamps the definition for all value IDs. Instead of having separate `getXYZValueId` functions floating around, all value definitions for all CCs are now (will be...) collected on a single object per CC, which has transparent type definitions that include the value ID definition:
[image]
For example, `BasicCCValues.currentValue.id` equals to the following value ID:
```js
{
    commandClass: CommandClasses.Basic,
    property: "currentValue",
}
```

and `BasicCCValues.currentValue.endpoint(1)` returns the followin [... cut]
closed 2022-07-20 by AlCalzone