zwave-js/zwave-js #675: Multicast?

lukescott, 2020-02-18:
From what I can tell no Z-Wave library supports multicast. They all support single-cast and possibly broadcast for the SwitchAll command class.

The SwitchAll command is useful for things like turning off all the lights in the house. You configure each device in how it responds to SwitchAll On or SwitchAll Off. You could configure your lights to respond to both On and Off but have your ceiling fans only respond to Off.

SwitchAll is not useful in other contexts, like turning off the lights downstairs vs upstairs because devices have to be individually preconfigured.

However, broadcast does support other command classes. It's just dangerous to send a blanket broadcast. For example, my Garage Door opener uses a relay that uses the BinarySwitch command class. If I were to send "BinarySwitch On" as a broadcast not only would my lights and fans turn on, but my Garage Door would also open. Not good.

But from what I can gather you can add a multicast filter to the header that limits which devices respond to the message. The message is still sent to all devices and is un-routed, but you can specify which devices actually respond.

I'm currently working on a personal project that ties my Z-Wave network to HomeKit. From what I can gather when you issues a command like "Hey Siri, Turn Off all the lights downstairs" (zone) HomeKit will send set characteristic data for each device in one HTTP command to the responsible Accessory. The HomeKit libraries currently parse that and send it to each Service making it difficult to know if it was a match command or individual... But it is possible to modify it to batch the changes.

What I want to do is if a batch is detected, send a multi-cast message instead to turn on/off the particular devices and then follow up on status after the fact.

AlCalzone, 2020-02-18:
I just took a look into the documentation that I have (and that you can find in the `specs` directory in this repo).
You want to use the `SendDataMulti` command, which can be found [here](https://github.com/AlCalzone/node-zwave-js/blob/master/specs/INS13954-Instruction-Z-Wave-500-Series-Appl-Programmers-Guide-v6_81_0x.pdf) on page 111. The difference to a normal `SendData` command is that the node ID is replaced with a variable-length bitmask (I would assume the LSB is node 1).

lukescott, 2020-02-18:
What's the difference between `SendDataMulti` and `SendDataMulti_Bridge`?

Is there an actual library that can be used that matches these headers?

AlCalzone, 2020-02-18:
I haven't concerned myself with the Bridge API yet and I'm not sure what the prerequisites for that are. As far as I understand, a Bridge Controller connects two networks and acts as a repeater, hence the additional *source node id* in the commands.

All this library does (and what I can test) is talk to a static controller (e.g. USB stick) using the serial API, so I've disregarded all bridge/portable/slave APIs.

> Is there an actual library that can be used that matches these headers?

Do you mean if there is an implementation which uses this method? I can add support for `SendDataMulti` here - it doesn't seem to require too much work.

lukescott, 2020-02-18:
> As far as I understand, a Bridge Controller connects two networks and acts as a repeater, hence the additional source node id in the commands.

Ah, sorry I think I'm getting terminology confused. In HomeKit a Bridge is a common device that controls multiple devices. I was thinking about the controller in terms of that.

> Do you mean if there is an implementation which uses this method? 

Yeah does Silicon Labs have an actual implementation? I've seen header files that seem to indicate an implementation, but I've never actually seen it.

>  I can add support for SendDataMulti here - it doesn't seem to require too much work.

That would be amazing! I'm in the process of updating the HomeKit library to allow batch updating.

AlCalzone, 2020-02-18:
> Yeah does Silicon Labs have an actual implementation? 

I think so, at least the specification PDFs let it seem like it. I don't have that though - I started by looking at OZW and then googling all the specification files I've collected. It might be that the implementation comes with the developer hardware kits.

lukescott, 2020-02-18:
https://www.silabs.com/products/development-tools/wireless/mesh-networking/z-wave

I'm assuming this is it? What's the difference between 500 and 700? I thought I remembered seeing a post somewhere about the 700 kit not including the SDK. So I'm guessing the 500 controller kit would be most relevant?

Source code would be amazing, but even having an official binary would be useful for testing I/O. I also wonder what the terms are in regards to open source projects.

If we had to get the controller kit ($1000!?) we could probably split the cost up. Although we'd have to be sure it was a for-sure benefit and we weren't violating any terms.

AlCalzone, 2020-02-18:
I'm not interested in buying controller kits. So far I've been doing fine with the specs - most headaches came from the external devices which sometimes have... let's call it "interesting" behavior.

AlCalzone, 2020-02-20:
@lukescott The PR for multicast support is up - I'll merge pretty soon. The changes were a bit larger than expected, but mainly because I used the opportunity to do some refactoring I've wanted to do for a while.

PR #677 (2020-02-20): Add support for multicast messages
With this PR we support sending CCs to a list of nodes (multicast) using the `SendDataMulticast` message. Therefore the `nodeId` property of the `CommandClass` class has been extended to accept a numeric array. When this is the case, the `Driver.sendCommand` method automatically sends the command to multiple nodes using `SendDataMulticast`. Multicast CCs always use the implemented version for serialization as opposed to singlecast CCs which check the target node's version.

As a side-effect, serializing the node id and payload length has been moved from `Comm [... cut]
closed 2020-02-20 by AlCalzone