zwave-js/zwave-js #682: ZigBee USB ID 0658:0200 Sigma Designs, Inc. wird niccht vollständig inkludiert

olli0815, 2020-02-23:
hi,

mein usb (ID 0658:0200 Sigma Designs, Inc. ) wird nicht sauber inkludiert
[
[image]
](url)

die schnittstelle schon
[image]

olli0815, 2020-02-24:
Vermissen nicht direkt, da ich den stick ja eh nicht anspreche.
Früher mit dem alten Adapter hingegen sah es so aus:
[image]

aber wie gesagt an der Stelle wirklich "nice to have"

AlCalzone, 2020-02-24:
Naja. Sinn macht das nicht wirklich beim Controller:

- Ready-Status gibt es
- Alive ist er immer
- Schlafen tut er nie
- Den events-Datenpunkt gibt es in meinem Adapter nicht
- Und Kommandoklassen (Basic) kann man ihm auch nicht senden

Die Bezeichnung könnte ich aber zumindest anzeigen lassen.

AlCalzone, 2020-03-07:
Ich gehe davon aus, dass die nächste Version des Adapters dieses Wochenende kommt. Dann sollte das funktionieren.

PR #715 (2020-03-07): Load device config for the controller node 
fixes: #682
closed 2020-03-07 by AlCalzone