zwave-js/zwave-js #683: Keine erweiterte Eigenschaften Roller Shutter (FGR222)

olli0815, 2020-02-23:
Hi,

trotz libopenzwave habe ich keine erweiterten Datenpunkte zum Steueren des Neigungswinkles der Roller shutter.

AlCalzone, 2020-02-23:
Der Adapter ist komplett unabhängig von OpenZWave, das ist der Hauptgrund, warum er überhaupt existiert.
Die Fibaro-Klasse ist eigentlich implementiert, aber ungetestet. Ich bräuchte einen Log vom Adapterstart bis zum Abschluss des Interviews des betreffenden Nodes. Am besten vorher Cache löschen.

olli0815, 2020-02-24:
Kannst du damit etwas anfeangen? sonst muss ich das Log.file nochmal vom server fummeln, da ich es über den ioBroker nicht (mehr) downloaden kann..

[Zwave2 log](https://github.com/AlCalzone/node-zwave-js/files/4243473/Zwave2.log)

AlCalzone, 2020-02-24:
Ne, ich meinte das Debug-Log des Adapters. Ist im Test-Thread beschrieben wie du da dran kommst.

olli0815, 2020-02-24:
[iobroker.2020-02-24.log](https://github.com/AlCalzone/node-zwave-js/files/4243610/iobroker.2020-02-24.log)

AlCalzone, 2020-02-24:
Jup, schaue ich bei Gelegenheit genauer an.

AlCalzone, 2020-02-24:
Kannst du mir noch verraten, welcher Node das ist?

olli0815, 2020-02-24:
z.b 4,5,6,7,10 & 11 das sind die roller shutter
da fehlt e"Venetian blind tilt position" via MANUFACTURER_PROPRIETARY

AlCalzone, 2020-02-24:
Ah, ich sehe was los ist. Die Interview-Prozedur für die Kommandoklasse fehlt noch. Muss mal schauen, wie das funktioniert.

AlCalzone, 2020-02-24:
Probably depends on https://github.com/AlCalzone/node-zwave-js/issues/552

olli0815, 2020-02-24:
perferkt.
Tausend Dank schonmal und weiter so!!
Wenn ich was testen kann, gerne

AlCalzone, 2020-02-24:
Das wird noch etwas dauern. Sage Bescheid.

AlCalzone, 2020-02-27:
In der nächsten Version sollte es hoffentlich funktionieren. Würde mich dann über Feedback freuen.

closed 2020-02-27 by AlCalzone

olli0815, 2020-03-01:
In der nächsten, oder ist es schon im git?

AlCalzone, 2020-03-01:
Nächste Adapter-Version. Selbst wenn es auf github ist, hilft es nicht, da ich diese Library zuerst in einer neuen Version veröffentlichen muss, bevor es mit dem Adapter funktioniert.

PR #698 (2020-02-27): Support configuring which ManufacturerProprietary commands a device supports and make FibaroCC usable
TODO:
- [x] make Fibaro CC expect responses
- [x] implement interview
- [x] add tests for expected responses

fixes: #683