zwave-js/zwave-js #685: DoorLockCC

lukescott, 2020-02-23:
I currently have Schage BE469. I think it might require a network key to work.

The features I currently use are the DoorLockCC and the NotificationCC. With the NotificationCC I can tell if a door was unlocked with a code or with a key. If with a code, it can tell me which user code was used.

In OpenZWave 1.4 (or 1.5?) there was a bug that would cause it to crash. So I have a forked version of it.  I also modified OpenZWave to not clear out the notification status after an interval. That was a part of the spec I felt they deviated from. For example, if I got a notification that the door was jammed, it would clear it after 1000ms even though the door was still jammed.

(The door would report jammed if half-way locked/unlocked).

If you need help I'm more than willing to assist. I just not sure about the network key part.

This is the last device I need to support before I can migrate everything away from OpenZWave. Is there a way for me to use both, or does each library need exclusive access? I'm currently using a Razzbery HAT on my Raspberry Pi.

AlCalzone, 2020-02-23:
> Seems like configuration and notification/alert are missing?

I can only say for sure with a logfile. You can create that by setting the env variable `LOGTOFILE` to something truthy, e.g. `"true"`. If the above info is correct, the device might need security to work correctly (which makes sense for a door lock). That is still on my roadmap, but requires quite a bit of work and testing to support.

> Is there a way for me to use both, or does each library need exclusive access?

Both need exclusive access to the serial port because the internal state of the application and the controller need to be somewhat in sync. A workaround would be using two controllers, e.g. a GPIO board and a USB stick. OZW would control one, this library the other.

lukescott, 2020-02-23:
Log file: [zwave-2260.log](https://github.com/AlCalzone/node-zwave-js/files/4242388/zwave-2260.log)

> Both need exclusive access to the serial port because the internal state of the application and the controller need to be somewhat in sync. A workaround would be using two controllers, e.g. a GPIO board and a USB stick. OZW would control one, this library the other.

Oh cool, so just get a USB controller and put it as a secondary controller. I'll look into that.

AlCalzone, 2020-02-23:
I'm not sure though if they can both work with the same network. It might be that you have to put the secure node in a different network so it can still communicate over OZW.

closed 2020-06-04 by AlCalzone

lukescott, 2020-06-08:
@AlCalzone do you have an example on how to use this? Particularly how to load with a network key? I would like to dive into this next weekend and try it out with my Schlage lock.

AlCalzone, 2020-06-08:
You can provide a network key (16 bytes) to the driver as a Buffer:
```
const driver = new Driver("COM4", {
	networkKey: Buffer.from([
		1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16,
	])
})
```
The rest should work as usual.

How you use the door lock depends on your use case. You can configure the lock with `setConfiguration(config)` and operate it with the `set(mode)` command. For details I suggest you dive into the type declarations.

PR #838 (2020-06-04): Add support for Door Lock CC
Fixes: #685