zwave-js/zwave-js #729: Command Class Wishlist

AlCalzone, 2020-03-09:
**All users of this library:** here's your chance to influence which CCs I will implement next.

The rules are simple:
1. Pick one of the [un-implemented CCs](https://github.com/AlCalzone/node-zwave-js/issues/6)
2. Vote for it:
    a. Search this issue. If there's a comment for your CC, vote for it using the thumbs up button.
    b. If there's no comment yet, please add one.

Unrelated comments and comments that don't follow the rules will be ignored and deleted.

---

**These CCs are next:**
1. Improved support for the Association CC (#272): Additional methods on the `Controller` class to manage node and endpoint associations
2. Security S0
3. Security S2
4. _this could be your CC!_

RoboPhred, 2020-03-26:
I want to get a vote in for Color.  I could try to implement it too, but I haven't been able to find the location of the spec docs.

RoboPhred, 2020-03-26:
Color Switch

I would like to try implementing it, but so far I have been unable to find the spec docs.

AlCalzone, 2020-03-27:
That's the Color Switch CC, to be found in https://github.com/AlCalzone/node-zwave-js/blob/master/specs/SDS13781-Z-Wave-Application-Command-Class-Specification.pdf on page 148ff. If you would like to make a PR, go ahead.

ChristianGrieger, 2020-06-16:
"Alarm Sensor CC"

16:08:05.075 CNTRLR « [Node 013] TODO: no handler for application command
16:09:17.290 DRIVER   Dropping message because it could not be deserialized: The command class Alarm Sensor is not implemented

Thank you!

larstobi, 2020-11-19:
Association Command Configuration

larstobi, 2020-11-19:
Scene Actuator Configuration

larstobi, 2020-11-19:
Scene Controller Configuration

robertsLando, 2020-11-20:
Thermostat Fan Mode

robertsLando, 2020-11-25:
Thermostat fan state

billiaz, 2020-12-01:
I am interested in Schedule one, to be used with new devices supporting scheduling
Schedule Comman Class. I am also happy help implementing it

kpine, 2020-12-04:
Barrier Operator

MYeager1967, 2020-12-06:
I'd love to finally be able to move my scene controller, an Enerwave ZWN-SC7, off of the Vera controller that it resides on. This device does not support the Central Scene Command class (91 or 0x5B). Instead, it uses Z-Wave Group Associations along with the Scene Controller Configuration Command Class (45 or 0x2D). This is described in the user manual. You use this command class to a map an association group, which is mapped to a physical button, to a scene ID. For example, for this device it looks like Group 1 is mapped to “Button 1”. The scene ID can be any number 1-255. I'm certain to be moving to zwavejs as I use Zwave2mqtt and it's moving to zwavejs.

AlCalzone, 2020-12-06:
@MYeager1967 please upvote https://github.com/zwave-js/node-zwave-js/issues/729#issuecomment-730310951 to vote for `Scene Controller Configuration`

jtscott, 2020-12-12:
Entry Control, literally no DIY keypad solution on the market at the moment!

LordMike, 2021-01-10:
Application Status

I'm seeing some reports of it being dropped due to not being supported. Has something to do with "device being busy". Sounds simple :)

maivorbim, 2021-02-10:
Irrigation, given that summer is coming :)

AlCalzone, 2021-02-10:
> summer is coming

I'll use that as the release name

bepsoccer, 2021-02-10:
Sound Switch, please.  I have a couple of the Zooz MultiSirens that I use with the legacy HA Zwave implementation that use parameter 3 via a service call, Sound Switch would make life easier!

Ikcelaks, 2021-02-15:
I'm currently working on Scene Controller Configuration and will do Scene Actuator Configuration afterwards.

MYeager1967, 2021-02-15:
Being able to move my last z-wave device off the old Vera controller would be awesome!!!!

AlCalzone, 2021-02-15:
Reminder:
> Unrelated comments and comments that don't follow the rules will be ignored and deleted.

maivorbim, 2021-02-15:
Powerlevel would be very useful to debug networks

WizKid, 2021-02-20:
Started implementing Entry Control in https://github.com/WizKid/node-zwave-js/commit/7648f2de5669df450976023870a529d71e6cd8ac . Still more to do which I'm planning to do. But thought I post in case someone else is working on it.

Especially handling the notifications and doing something useful with them. But I'm testing it with Ring Keypad and overall it seems to work. Pushing in a code and clicking Arm or any of the other buttons works. In the sense that the messages are parsed and I can see in the log that we get the right data.

Ikcelaks, 2021-02-20:
> Powerlevel CC would be very useful to debug networks and would integrate nicely with the new node graph.

I'm implementing PowerlevelCC now.

WizKid, 2021-02-20:
Put up a pull request for Entry Control at #1838 .

arcticgreen, 2021-02-26:
I’m voting for `MultiCommandCCCommandEncapsulation` since I have three Philio PST02-1A that are ignored when they send motion and door-open data... (temperature and illuminance are OK):
```
14:38:55.358 DRIVER « [Node 024] [REQ] [ApplicationCommand]
                       └─[MultiCommandCCCommandEncapsulation]
14:38:55.365 DRIVER   Received a command that contains multiple CommandClasses. 
                      This is not supported yet! Discarding the message...
14:38:55.367 CNTRLR « [Node 024] TODO: no handler for application command
```

https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/driver/Driver.ts#L2006

gbuktenica, 2021-04-08:
Energy Production

Hedda, 2021-05-20:
Window Covering

jprates, 2021-05-27:
Hi,

Thanks for asking the masses what we need.

I'm voting for Transport Service as I've loads of messages on my log stating "Dropping message because it could not be deserialized: The command class Transport Service is not implemented".

I think they are being generated while trying to communicate with my Aeotec Water Sensor 7, so maybe this will also help pushing for that device to be supported soon. ;)

danielbrunt57, 2021-08-09:
Door Lock Logging

EiNSTeiN-, 2022-01-01:
Humidity Control Mode/Operating State/Setpoint

pmeulen, 2022-01-09:
`Meter Table Monitor`

I've an "Eneco - Meteradapter" by "Prodrive B.V." PN: 6599-1500-0201 that sends this command. It is connected to the P1 port of my electricity meter.
Z-Wave JS logs show two of these every 10 seconds:
```
SERIAL « 0x0122000400081c600d06003d0d000111000007e601090d1d0060006fbc9d67000 (36 bytes)
         0013b0c
DRIVER   Dropping message because it could not be deserialized: The command class Meter
          Table Monitor is not implemented (ZW0303)
SERIAL » [ACK]
```

I expect that this contains the kWh counter(s) of the meter.

jmgiaever, 2022-06-08:
**Schedule Entry Lock V3**

E.g to be able to set temporary/time limited user codes on locks. 

I'm aware it's deprecated, but guessing many locks that exist today use this CC.

mikemcguire513, 2022-10-04:
I would like to see support for the Screen Meta Data Command class.  This would allow for setup of Button Labels on the Nexia NX1000 scene controller.

danielbrunt57, 2022-10-04:
Command Class: Door_Lock_Logging

AlCalzone, 2023-02-01:
@danielbrunt57 you keep requesting that, but it is supported already.

danielbrunt57, 2023-02-02:
> @danielbrunt57 you keep requesting that, but it is supported already.

I have had this Kwikset 910 since about 2008 and it supports these command classes:

- Antitheft V2
- Association Grp Info
- Association V2
- Battery
- Configuration
- Device Reset Locally
- Door Lock Logging
- Door Lock V2
- Firmware Update Md V3
- Manufacturer Specific V2
- Notification V4
- Powerlevel
- Schedule Entry Lock V2
- Security
- Time Parameters
- User Code
- Version V2
- Zwaveplus Info V2

From 2000 to 2021 I used HomeSeer and had logging in 2008 when I acquired the lock (that's 15 years ago btw!)
yet I see nothing in Z-WaveJS resembling logging...

[image]

am I missing or not understanding something???

AlCalzone, 2023-02-02:
Not all CCs are exposed via the value API. Some are just too complex for that, some don't make tons of sense. 
https://zwave-js.github.io/node-zwave-js/#/api/CCs/DoorLockLogging

AFAIK this CC also doesn't operate in push-style but would have to be polled sporadically/on demand.

You can open an issue in https://github.com/zwave-js/zwave-js-ui and request support for a dedicated UI for Door Lock Logging.

tamorgen, 2023-04-12:
I've been in contact with iBlinds tech support, and they are telling me that many of the issues I have with iBlinds are due to the command class that ZWave JS is using (Multi-level switch).  They told me that there is a newer command class for Window Coverings, that isn't being implemented by many hubs, that is more appropriate for Horizontal blinds.  Is this command class implementation on the roadmap any time soon? 

If it is, what would need to be done to change the device integration?

WizKid, 2023-04-12:
@tamorgen : Maybe I'm misunderstanding the specification. But according to page 159 of the specification https://raw.githubusercontent.com/zwave-js/specs/d6c115ec1c46dd1db9f3dca3fc2106df577fac9c/Z-Wave%20Device%20Class%20Specification.pdf it says:

> It is not RECOMMENDED to use this generic device class for new devices. Instead, it is
> RECOMMENDED to base devices of this category on the Multi-level switch generic device class with
> the Multi-position Motor specific device class

To me that sounds like ZwaveJS is doing the right thing.

tamorgen, 2023-04-12:
> @tamorgen : Maybe I'm misunderstanding the specification. But according to page 159 of the specification [...]

No, it should not.  The problem here is that while the generic multi-level command class works fine for roller shades or curtains, where Open is at one end of the spectrum, and close is at the other, Horizontal blinds have an open at 50% (in the middle).

To quote their CFO, with whom i've been interacting:

> Horizonal blinds are relatively new to the smart home environment. Z-Wave just recently implemented a specific command class for Window Coverings which includes horizontal blinds. However, to date, there are no hubs that have implemented the new command class. Instead, all hubs are still using the Z-Wave Multilevel Switch command class to support window coverings. While this works well for roller shades and curtains, it is not ideal for horizontal blinds where 50% is fully open and as you are aware can cause problems for our implementation.

tamorgen, 2023-04-12:
> @tamorgen : Maybe I'm misunderstanding the specification. But according to page 159 of the specification [...]

Also, the link you provided does not have a date on it.  I googled the same document, and the one's I found have specific dates listed on the first page of the PDF.

https://sdomembers.z-wavealliance.org/document/dl/638

tamorgen, 2023-04-12:
> > @tamorgen : Maybe I'm misunderstanding the specification. But according to page 159 of the specification [zwave-js/specs@`d6c115e`/Z-Wave%20Device%20Class%20Specification.pdf (raw)](https://raw.githubusercontent.com/zwave-js/specs/d6c115ec1c46dd1db9f3dca3fc2106df577fac9c/Z-Wave%20Device%20Class%20Specification.pdf) it says:
> 
> You are mixing up _Device Types_ vs. _Command Classes_. The Window Covering Generic Device Class is discouraged, not the command class. The Window Covering Command Class is alive and well, a requirement for Window Covering Device Type for the Device Type V2 Specification.
> 
> Anyways, this is a wish list issue. The WC class is already specifically called out as requested. There's been no statement it would not be implemented. Hit the thumbs up in the comment and wait, or even better, implement it!
> 
> [image]

Thank you for your feedback.  I've voted for it.

AlCalzone, 2023-04-18:
Receiving MultiCommandCCCommandEncapsulation is supported since v10.10.0

AlCalzone, 2023-04-18:
Energy Production CC support is added in #5677

EkielZan, 2024-04-16:
Hi, I've a Quby / Engie Meter Adapter 
See here : https://devices.zwave-js.io/?jumpTo=0x0130:0x0002:0x0001:0.0
or
https://github.com/openhab/org.openhab.binding.zwave/blob/main/doc/quby/en00_0_0.md

And I receive this error when trying to get data from it:

The command class Meter Table Monitor is not implemented (ZW0303)