zwave-js/zwave-js #743: Removing a device from the network.

RoboPhred, 2020-03-25:
I have a failed switch I need to remove from the network, as well as several I am decommissioning.  I have unpaired the devices, and all now show up as a dead status when listing nodes.

I read through the docs but could not find any mention of removing a device from the network, just exclusion mode which I already ran using the Aeon stick gen5 in independent mode.

Is there an api call to remove a node from the network?

AlCalzone, 2020-03-25:
Yes! The docs are a bit outdated, but you're looking for this:
https://github.com/AlCalzone/node-zwave-js/blob/0b1338c31305b14199e03fa20a151458dd5dc97a/src/lib/controller/Controller.ts#L1364

RoboPhred, 2020-03-28:
Thanks, this is what I need.

closed 2020-03-28 by RoboPhred