zwave-js/zwave-js #745: Cannot access CCs: returns "does not support the Command Class Multi Channel Association" for most CCs

RoboPhred, 2020-03-28:
Started experiencing this after updating to the newest version of zwave-js.

When asking for a supported CC, I get an error indicating it cannot find the unrelated CC "Multi Channel Association"

The device is a Aeotec Nano Dimmer with a WallSwipe attached.

```
>zwave.controller.nodes.get(17).deviceClass.generic.name
'Multilevel Switch'
>zwave.controller.nodes.get(17).supportsCC(CommandClasses["Multilevel Switch"])
true
>zwave.controller.nodes.get(17).commandClasses[CommandClasses["Multilevel Switch"]]
Error: Node 17 does not support the Command Class Multi Channel Association!
    at formatValue (util.js:609:18)
    at formatObject (util.js:647:17)
    at formatProperty (util.js:831:11)
    at formatValue (util.js:609:18)
    at formatObject (util.js:647:17)
    at formatProperty (util.js:831:11)
    at formatValue (util.js:609:18)
    at formatMap (util.js:788:7)
    at formatValue (util.js:423:37)
    at Object.get (C:\dev\homectrl-server\packages\homectrl-plugin-zwave\node_modules\zwave-js\src\lib\node\Endpoint.ts:244:12)
```

When asking for a CC that is not supported, I get a proper error message:
```
>zwave.controller.nodes.get(17).commandClasses[CommandClasses["Battery"]]
Error: Node 17 does not support the Command Class Battery!
    at emitOne (events.js:116:13)
    at REPLServer.onLine (repl.js:468:10)
    at REPLServer.runBound [as eval] (domain.js:314:12)
    at bound (domain.js:301:14)
    at REPLServer.defaultEval (repl.js:279:5)
    at finish (repl.js:512:38)
    at REPLServer.self.writer (repl.js:367:19)
    at Object.inspect (util.js:324:10)
    at formatValue (util.js:423:37)
    at Object.get (C:\dev\homectrl-server\packages\homectrl-plugin-zwave\node_modules\zwave-js\src\lib\node\Endpoint.ts:244:12)
```

Device cache:
```
"17": {
            "id": 17,
            "interviewStage": "Complete",
            "deviceClass": {
                "basic": 4,
                "generic": 17,
                "specific": 1
            },
            "isListening": true,
            "isFrequentListening": false,
            "isRouting": true,
            "maxBaudRate": 40000,
            "isSecure": false,
            "isBeaming": true,
            "version": 4,
            "commandClasses": {
                "0x25": {
                    "name": "Binary Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "version": 1
                        }
                    },
                    "values": [
                        {
[... 1873 lines cut ...]
                            "property": "libraryType",
                            "value": 3
                        },
                        {
                            "endpoint": 0,
                            "property": "protocolVersion",
                            "value": "4.54"
                        },
                        {
                            "endpoint": 0,
                            "property": "firmwareVersions",
                            "value": [
                                "2.3"
                            ]
                        },
                        {
                            "endpoint": 0,
                            "property": "interviewComplete",
                            "value": true
                        }
                    ]
                }
            }
        }
```

RoboPhred, 2020-03-28:
After power cycling the switch, I can now call .set and .get on it properly.  However, I still get the error when asking for the command class directly.  Not sure what to make of the issue.

AlCalzone, 2020-03-28:
> zwave.controller.nodes.get(17).commandClasses[CommandClasses["Multilevel Switch"]]

That is wrong. Do it this way and it should work:
```ts
zwave.controller.nodes.get(17).commandClasses["Multilevel Switch"]
```

RoboPhred, 2020-03-28:
I am getting the same behavior:

```
>zwave.controller.nodes.get(18).supportsCC(CommandClasses["Multilevel Switch"])
true
>zwave.controller.nodes.get(18).commandClasses["Multilevel Switch"]
Error: Node 18 does not support the Command Class Multi Channel Association!
    at formatValue (util.js:609:18)
    at formatObject (util.js:647:17)
    at formatProperty (util.js:831:11)
    at formatValue (util.js:609:18)
    at formatObject (util.js:647:17)
    at formatProperty (util.js:831:11)
    at formatValue (util.js:609:18)
    at formatMap (util.js:788:7)
    at formatValue (util.js:423:37)
    at Object.get (C:\dev\homectrl-server\packages\homectrl-plugin-zwave\node_modules\zwave-js\src\lib\node\Endpoint.ts:244:12)
>
```

AlCalzone, 2020-03-29:
> the repl is trying to serialize all properties that the command class returns,

Yeah, that must be it. I'm not sure if I can filter out the method calls from `util.inspect` without looking at the stack trace.

AlCalzone, 2020-03-31:
Closing in favor of #756

closed 2020-03-31 by AlCalzone