zwave-js/zwave-js #746: Receiving value change events from Lifeline supporting devices.

RoboPhred, 2020-03-28:
I have a Aeotech Nano Dimmer with a Wall Swipe attached.  Looking at the device cache, the device seems to support the Lifeline extension of sending updates to the hub.

However, toggling the switch sends no feedback at all to zwave-js.  I do not get any events when the switch state changes.

Is there some other setup work I have to do to enable events on switch state changes?

Device cache:
```
"17": {
            "id": 17,
            "interviewStage": "Complete",
            "deviceClass": {
                "basic": 4,
                "generic": 17,
                "specific": 1
            },
            "isListening": true,
            "isFrequentListening": false,
            "isRouting": true,
            "maxBaudRate": 40000,
            "isSecure": false,
            "isBeaming": true,
            "version": 4,
            "commandClasses": {
                "0x25": {
                    "name": "Binary Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "version": 1
                        }
                    },
                    "values": [
                        {
                            "endpoint": 0,
                            "property": "currentValue",
                            "value": false
                        },
                        {
                            "endpoint": 0,
                            "property": "interviewComplete",
                            "value": true
                        }
                    ]
                },
                "0x26": {
                    "name": "Multilevel Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "version": 3
                        }
                    },
                    "values": [
                        {
                            "endpoint": 0,
                            "property": "switchType",
                            "value": 2
                        },
                        {
                            "endpoint": 0,
                            "property": "currentValue",
                            "value": 0
                        },
                        {
                            "endpoint": 0,
                            "property": "interviewComplete",
                            "value": true
[... 1838 lines cut ...]
                            "property": "libraryType",
                            "value": 3
                        },
                        {
                            "endpoint": 0,
                            "property": "protocolVersion",
                            "value": "4.54"
                        },
                        {
                            "endpoint": 0,
                            "property": "firmwareVersions",
                            "value": [
                                "2.3"
                            ]
                        },
                        {
                            "endpoint": 0,
                            "property": "interviewComplete",
                            "value": true
                        }
                    ]
                }
            }
        }
```

AlCalzone, 2020-03-28:
Normally, it should not be necessary to do anything special, if the device is configured properly. You could create a logfile (env variable `LOGTOFILE=true`) and check if anything is received but ignored.

RoboPhred, 2020-04-09:
I have now wired up a few other ZWavePlus devices, and they all return value changes as expected.  I suspect the NanoDimmer is not implemented properly.  Closing as this is a device specific fluke.

I suppose this could be added to the "needs polling" tag, and we probably also need a flag in the device config to indicate zwaveplus nodes that need to be polled despite their lifeline.

closed 2020-04-09 by RoboPhred