zwave-js/zwave-js #747: [Support] Determining const values from Command Class Control Specification

RoboPhred, 2020-03-28:
I am looking into implementing the Color Switch command class.  I think I can puzzle out how command classes are added, but I am up against a blocker regarding the documentation.

I am looking at the SDS14223-Z-Wave-Command-Class-Control-Specification document, and it specifies constants as named, eg `COMMAND_CLASS_SWITCH_COLOR` and `SWITCH_COLOR_GET`.  I have looked through the specs folder and I cannot find where these are defined.  I know COMMAND_CLASS_SWITCH_COLOR is 0x33, and I can probably guess that SWITCH_COLOR_GET is 0x01, but it would feel safer to be working from an official list of these values.

RoboPhred, 2020-03-29:
To avoid polluting the issue queue, I uploaded my current progress as a draft MR
- https://github.com/AlCalzone/node-zwave-js/pull/748
I would appreciate it if you could help me find my way in this project.

AlCalzone, 2020-03-29:
You'll find these constants in https://github.com/AlCalzone/node-zwave-js/blob/master/specs/SDS13548-List-of-defined-Z-Wave-Command-Classes.xlsx

closed 2020-03-31 by AlCalzone