zwave-js/zwave-js #749: "Driver is not defined" when trying to run current master branch.

RoboPhred, 2020-03-29:
Due to the mixing of a typed import with typescript's `emitDecoratorMetadata` , the current master branch cannot run.

The problem is probably found in other Command Classes, but the one I am running into is in `MultiChannelCC.ts`.
The use of a typed import
`import type { Driver } from "../driver/Driver";`
does not import Driver class into this file.  However, the MultiChannelCC constructor is trying to use it:
```js
public constructor(driver: Driver, options: CommandClassOptions) 
```

Because it is a type passed to the constructor, `emitDecoratorMetadata` tries to store that type in metadata.  The compiled output is:
```js
MultiChannelCC = __decorate([
    CommandClass_1.commandClass(CommandClasses_1.CommandClasses["Multi Channel"]),
    CommandClass_1.implementedVersion(4),
    __metadata("design:paramtypes", [Driver, Object])
], MultiChannelCC);
```

Driver is attempted to be used here, but the import is a type import, so there is no Driver defined.  This causes a crash when loading the library.

The solution would be to import all constructor arguments as standard imports.

This might be an issue we should bring up with the Typescript devs, as it is a bit silly that it allows you to do this.

RoboPhred, 2020-03-29:
I was able to resolve this locally by shutting off `emitDecoratorMetadata`.  As far as I can tell, zwave-js does not use this design time metadata for anything, so shutting it off should be safe.

AlCalzone, 2020-03-29:
Ooooohhh, you've found the root case for another issue. `ts-node` only runs in `transpileOnly` mode since TS 3.8.

AlCalzone, 2020-03-29:
I think its safe to remove that option. I had it on from the start when I was planning to do more decorator magic. 

Raised https://github.com/microsoft/TypeScript/issues/37672 to make the TypeScript team aware of it.

AlCalzone, 2020-03-29:
fixed by #751

closed 2020-03-29 by AlCalzone