zwave-js/zwave-js #760: Network healing and device associations

RoboPhred, 2020-04-09:
Currently, I constantly move nodes around, and my main controller is often offline as I have not prepared the Pi to host it yet.  Nonetheless, I am already beginning to use some of the devices in association mode.

While most associated devices work fine without the hub, I have one that seems intent on routing its association through the hub at the other end of the house, despite being in the same room as the devices it is controlling.

I tried running healNetwork on it several times, but the problem persists and its associations do not function when the hub is offline.

Doing some research into heal, I found [this post](https://www.silabs.com/community/wireless/z-wave/forum.topic.html/best_way_to_healaz-MNsb), which seems to indicate that in the process of healing, you should route associations to each other.  I suspect that this would solve my issue,

- Does zwave-js currently assign association routes on heal?
- Is there a way to manually assign routes from the zwave-js api?

AlCalzone, 2020-04-12:
You need to distinguish between associations (nodes control some command classes directly in other nodes) and routes (used for routing packets through the network).

I believe the post is talking about the controller commands AssignReturnRoute (funcId 0x46) and DeleteReturnRoute (0x47). Both are not implemented yet.

AlCalzone, 2020-05-05:
I've read up on this, seems you were onto something with the direct routes. OpenHAB does it the following way:
1. update neighbors
2. delete return routes
3. add return routes for the lifeline association (controller)
4. add return routes for other associations between nodes (up to 3, if any are configured)

I'll probably implement this soon (should be in the upcoming v3.0 release).

PR #791 (2020-05-08): Refresh return routes during network heal
fixes: #760
closed 2020-05-08 by AlCalzone