zwave-js/zwave-js #765: Diagnosing failed interviews

RoboPhred, 2020-04-14:
I am experiencing an issue where one of my nodes never calls "interview completed".

I have many GE Enbrighten Gen 2 switches, the majority of which interview fine.  However, one of them never raises interview completed.  Despite this, the node's status is marked as `2` (Awake), its cache is fully formed, and I can interact with it just fine from the controller.  

I am seeing sporatic failure messages for it during the interview process, and several messages about ignoring malformed data, but I have so many devices its impossible to track what messages are for what node.  Is there any filter I can place on the log messages to cut down on excess messages or filter messages to this specific node?

RoboPhred, 2020-04-14:
I discovered that I can manually call .interview on nodes.  It looks like this particular switch is making it past most of the interview, then reliably failing during `ConfigurationCCInfoReport`

```
>zwave.controller.nodes.get(23).interview().then(console.log)
17:59:25.592 CNTRLR   [Node 023] Beginning interview - last completed stage: NodeInfo
17:59:25.599 CNTRLR   [Node 023] ManufacturerSpecificCC: doing a partial interview...
17:59:25.603 CNTRLR   [Node 023] [~] [Manufacturer Specific] interviewComple [Endpoint 0] [internal]
                      te: true => true
Promise { <pending> }
>17:59:25.671 CNTRLR   [Node 023] VersionCC: doing a partial interview...
17:59:25.674 CNTRLR   [Node 023] [~] [Version] interviewComplete: true => tr [Endpoint 0] [internal]
                      ue
17:59:25.678 CNTRLR   [Node 023] trying to load device config
17:59:25.687 CNTRLR   [Node 023] device config loaded
17:59:25.705 CNTRLR   [Node 023] MultilevelSwitchCC: doing a partial interview...
17:59:25.709 CNTRLR » [Node 023] requesting current switch state...
17:59:25.712 DRIVER » [Node 023] [REQ] [SendData]                                        [P: Normal]
                      │ transmitOptions: 0x25
                      │ callbackId:      94
                      └─[MultilevelSwitchCCGet]
17:59:25.750 SERIAL » 0x0109001317022602255eaf                                            (11 bytes)
17:59:25.773 SERIAL « [ACK]                                                                   (0x06)
17:59:25.777 DRIVER   ACK received from controller for current transaction
17:59:25.780 SERIAL   The receive buffer is empty, waiting for the next chunk...
17:59:25.805 SERIAL « 0x0104011301e8                                                       (6 bytes)
17:59:25.810 SERIAL   Buffer := 0x0104011301e8                                             (6 bytes)
17:59:25.813 SERIAL » [ACK]                                                                   (0x06)
17:59:25.834 DRIVER « [RES] [SendData]                                                [confirmation]
                        wasSent: true
17:59:25.841 SERIAL   The receive buffer is empty, waiting for the next chunk...
17:59:25.845 SERIAL « 0x010700135e000005b0                                                 (9 bytes)
17:59:25.848 SERIAL   Buffer := 0x010700135e000005b0                                       (9 bytes)
17:59:25.871 SERIAL » [ACK]                                                                   (0x06)
17:59:25.875 DRIVER « [REQ] [SendData]                                                [confirmation]
                        callbackId:     94
                        transmitStatus: OK
17:59:25.879 DRIVER   ACK received from node for current transaction. RTT = 167.25 ms
17:59:25.902 SERIAL   The receive buffer is empty, waiting for the next chunk...
17:59:25.909 SERIAL « 0x010b00040017052603636300c7                                        (13 bytes)
17:59:25.912 SERIAL   Buffer := 0x010b00040017052603636300c7                              (13 bytes)
17:59:25.915 CNTRLR   [Node 023] [~] [Multilevel Switch] targetValue: 99 => 99          [Endpoint 0]
17:59:25.937 CNTRLR   [Node 023] [~] [Multilevel Switch] duration: [Duration: 0seconds] [Endpoint 0]
                       => [Duration: 0seconds]
17:59:25.943 CNTRLR   [Node 023] [~] [Multilevel Switch] currentValue: 99 => 99         [Endpoint 0]
17:59:25.945 SERIAL » [ACK]                                                                   (0x06)
17:59:25.967 DRIVER « [Node 023] [REQ] [ApplicationCommand]                                  [final]
                      └─[MultilevelSwitchCCReport]
                          payload: 0x636300
17:59:25.987 DRIVER   ACK already received, resolving transaction
17:59:25.990 DRIVER   resuming send queue
17:59:25.993 SERIAL   The receive buffer is empty, waiting for the next chunk...
17:59:25.997 CNTRLR   [Node 023] [~] [Multilevel Switch] interviewComplete:  [Endpoint 0] [internal]
                      true => true
17:59:26.016 CNTRLR   [Node 023] ZWavePlusCC: doing a partial interview...
17:59:26.036 CNTRLR   [Node 023] [~] [Z-Wave Plus Info] interviewComplete: t [Endpoint 0] [internal]
                      rue => true
17:59:26.042 CNTRLR   [Node 023] ConfigurationCC: Loading configuration parameters from device confi
                      g
17:59:26.045 CNTRLR   [Node 023] [~] [Configuration] isParamInformationFromConfig: true = [internal]
                      > false
17:59:26.066 CNTRLR   [Node 023] [Configuration] 3: metadata updated
17:59:26.089 CNTRLR   [Node 023] [Configuration] 4: metadata updated
17:59:26.093 CNTRLR   [Node 023] [~] [Configuration] isParamInformationFromConfig: false  [internal]
                      => true
17:59:26.097 CNTRLR   [Node 023] ConfigurationCC: doing a complete interview...
17:59:26.114 CNTRLR » [Node 023] finding first configuration parameter...
17:59:26.139 DRIVER » [Node 023] [REQ] [SendData]                                        [P: Normal]
                      │ transmitOptions: 0x25
                      │ callbackId:      95
                      └─[ConfigurationCCPropertiesGet]
[... 142 lines cut ...]
17:59:31.143 SERIAL   Buffer := 0x0104011301e8                                             (6 bytes)
17:59:31.164 SERIAL » [ACK]                                                                   (0x06)
17:59:31.169 DRIVER « [RES] [SendData]                                                [confirmation]
[... cut ...]
17:59:31.302 SERIAL » [ACK]                                                                   (0x06)
17:59:31.306 DRIVER « [Node 023] [REQ] [ApplicationCommand]                                [partial]
                      └─[ConfigurationCCInfoReport]
                          payload: 0x00030153657420746865204c4544206c696768
17:59:31.332 SERIAL   The receive buffer is empty, waiting for the next chunk...
17:59:32.564 DRIVER   resuming send queue
false
17:59:32.605 DRIVER   The send queue is empty
```

AlCalzone, 2020-04-14:
> * Is there any filter I can place on the log messages to cut down on excess messages or filter messages to this specific node?

No, but that would be a nice feature

> * Is there an event raised when the interview process for a node fails?

Not yet.

---

Regarding your log: The node does answer, but the repsonse doesn't seem to be interpreted as the response:
```
17:59:26.780 DRIVER » [Node 023] [REQ] [SendData]                                        [P: Normal]
                      │ transmitOptions: 0x25
                      │ callbackId:      97
                      └─[ConfigurationCCInfoGet]
                          payload: 0x0003
...
17:59:27.122 DRIVER « [Node 023] [REQ] [ApplicationCommand]                                [partial]
                      └─[ConfigurationCCInfoReport]
                          payload: 0x00030153657420746865204c4544206c696768
```

I bet the `@expectedCCResponse` is wrong from copy & pasting.

AlCalzone, 2020-04-14:
Ok that is not the issue. The InfoReport is a multipart message. You only receive the first of two messages, then nothing.
I'll need to investigate if the receipt must be confirmed somehow.

AlCalzone, 2020-04-14:
I haven't found anything about needing to confirm the receipt. Did I understand correctly that you have multiple copies of that device and node 23 is the only with interview issues? Then it might be a connectivity problem.

RoboPhred, 2020-04-14:
Yes, my house is full of these switches and that was the only node having trouble.  This morning I tried excluding it then including it again, and now the interview is working as expected, so I think the switch got into a bad state.  It's resolved now.

PR #769 (2020-04-15): feat(node): add "interview failed" event
Fixes: #765 (although that is closed) :)

Nodes now emit a `"interview failed"` event when the interview fails with some additional info why.
closed 2020-04-14 by RoboPhred