zwave-js/zwave-js #767: Use string value enums for public enumerated values where possible.

RoboPhred, 2020-04-14:
For things like node status or interview status, it would be helpful if string enums were used instead of numerical.  This would make the api more self-documenting and easier to discover when dumping objects with a repl.

```
export enum NodeStatus {
	Unknown = "unknown",
	Asleep = "asleep",
	Awake = "awake",
	Dead = "dead",
}

...

// Currently might show `2`, with this change will show `awake`
console.log(controller.nodes.get(23).status); 
```

This makes sense for NodeStatus which is unordered.  InterviewStage seems to be ordered, but an ordinal list could be made mapping the stages to numbers to assist with finding the relative ordering of the interview stages.

I can make a PR for this later on, if this is something you would like to adopt.

AlCalzone, 2020-04-15:
Turning the numeric values into their names is as easy as `Enum[enumValue]`.

Or is this issue just about inspecting the values? That could be more intuitive, that's true.
I'm not sure if I want to use string enums and additional lookup tables everywhere, especially for the values that are derived from the Z-Wave specs.
Maybe something sneaky could be done with custom inspect functions? 
```js
"use strict";
const { inspect } = require("util");

class Foo {
	/** @type {number} */
	get foo() {
		return /** @type {any} */ ({
			valueOf() {
				return 1;
			},
			[inspect.custom](options) {
				return "bar";
			},
		});
	}
}

const f = new Foo();
console.log(f.foo + 1); // prints 2
console.log(f.foo); // prints "bar"
```
These properties could "magically" be replaced by decorators when e.g. the env variable NODE_ENV is set to "development".

AlCalzone, 2020-04-15:
I've found this example on how to do instance-level decorators with TypeScript:
https://romkevandermeulen.nl/2018/01/24/typescript-property-decorators.html

However, we need to use WeakMaps:
> Do note that since each object that this decorator is used on is added to the Map, these objects won’t go out of memory and won’t be garbage collected.

RoboPhred, 2020-04-15:
The fact that these are constants defined by z-wave changes my thoughts on this, I had thought they were local to zwave-js to describe its current state.

Let's keep them as undecorated raw values; having them change form with environment variables will just add to the confusion and bring about works-on-my-machine bugs.

AlCalzone, 2020-04-16:
> constants defined by z-wave

Well, not all of them. For example, NodeStatus /  InterviewStage aren't. But many of the CC-related enums are.

> Having properties change values based on dev mode adds too much risk

I haven't tested but the above example might break if someone does checks on the return type. Maybe a custom inspect function on the containing classes would make more sense. So if you printed a node, you would get something like this:
```
ZWaveNode {
    ...
    status: NodeStatus.Awake (2),
}
```

AlCalzone, 2020-10-10:
I've decided that I'm not going to do this.

closed 2020-10-10 by AlCalzone