zwave-js/zwave-js #806: Color CC: Add RGBW value ID

AlCalzone, 2020-05-17:


robertsLando, 2020-11-30:
What do we need for this?

AlCalzone, 2020-11-30:
We need to which format the hex string should have. OZW concatenates a bunch of HEX digits:
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/Color.cpp#L210

I'm not sure if its worth going past RRGGBB with this, since it is standard and everything else is missing discoverability.

Basically what needs to be done is:
* Probably during the interview, metadata for this value ID should be created
* When a color report is received, the new value of this string must be figured out (from existing values and the new value) and also written
* When this value is set (via the SET_VALUE API) it must be split into its parts to call the set method.

robertsLando, 2020-11-30:
> When a color report is received, the new value of this string must be figured out (from existing values and the new value) and also written

And reported to the user by an event :)

AlCalzone, 2020-11-30:
> And reported to the user by an event :)

That happens when you set a value in the value DB

PR #1388 (2021-01-14): feat(cc): add hexColor value to Color Switch CC
This PR adds a new value `hexColor` to the Color Switch CC to simplify setting and displaying HEX colors. Support is limited to `"RRGGBB"` for now.

fixes: #806
closed 2021-01-14 by AlCalzone