zwave-js/zwave-js #829: test is not running

inteldon, 2020-05-31:
I'm trying the first test with import {Driver} from .... and then the next line when I try to use "const driver = new Driver(...)" I get Driver is not a constructor. Can anyone help? 

Thanks

AlCalzone, 2020-05-31:
Can you show specific code please? And which version?

inteldon, 2020-05-31:
I'm using node v10.19.0

Quickly tried to test the import functionality and it wasn't working. 

When I try: 
`const Driver = require("zwave-js");
const driver = new Driver("/dev/ttyAMA0")
`
I get that Driver is not a constructor

When I try: 
`const Driver = require("zwave-js");
const driver = Driver("/dev/ttyAMA0")
`
I get that Driver is not a function 

When I try: 
`import {Driver} from "zwave-js";`

I get Unexpected token { 

I tried installing babel and enabling es6 but nothing is working. 

Thanks

AlCalzone, 2020-05-31:
> import {Driver} from "zwave-js";

is ES module syntax (or TypeScript). It translates to 

```
const { Driver } = require("zwave-js");
```
in plain old JS (notice the brackets around Driver).

inteldon, 2020-05-31:
Thank you. At least I don't see this error anymore. I'll get started and will ask if I have questions.

closed 2020-05-31 by AlCalzone