zwave-js/zwave-js #830: Can't initialize driver

inteldon, 2020-05-31:
Hi. I'm using two different ZWave modules. First is the Razberry module (/dev/ttyAMA0) and the other is the UZB stick (/dev/ttyUSB0). Both are very popular for ZWave development. 

I tried running the code: 
`
const { Driver} = require("zwave-js");
const p1 = "/dev/ttyAMA0";
const p2 = "/dev/ttyUSB0";

const driver = new Driver(p1)

driver.once("driver ready", () => {
    console.log("ready!");
	driver.start();
	driver.controller.nodes.forEach(
	    console.log("Detected a node")
	);
	driver.destroy();
})
`

But in response I get `DRIVER   destroying driver instance...
`

Then I tried to use 'driver.start()' outside of 'driver.once' and I got a more elaborate error message. 

18:10:12.511 DRIVER   version 3.5.4
18:10:12.514 DRIVER
18:10:12.516 DRIVER   starting driver...
18:10:12.518 DRIVER   **opening serial port /dev/ttyAMA0**
18:10:12.544 DRIVER   **serial port opened**
18:10:12.557 SERIAL » [NAK]                                                                   (0x15)
18:10:14.080 DRIVER   loading configuration...
18:10:16.161 DRIVER   beginning interview...
18:10:16.167 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
18:10:16.172 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
18:10:16.191 CNTRLR   beginning interview...
18:10:16.193 CNTRLR   querying version info...
18:10:16.217 DRIVER » [REQ] [GetControllerVersion]                                   [P: Controller]
18:10:16.223 SERIAL » 0x01030015e9                                                         (5 bytes)
18:10:17.829 DRIVER   Timeout occured waiting for ACK - scheduling transmission attempt (2/3) in 100 ms...
18:10:17.929 DRIVER » [REQ] [GetControllerVersion]  [attempt 2/3]
18:10:17.931 SERIAL » 0x01030015e9                                                        (5 bytes)
18:10:19.534 DRIVER   Timeout occured waiting for ACK - scheduling transmission attempt (3/3) in 110   0 ms...
18:10:20.634 DRIVER » [REQ] [GetControllerVersion]                                     [attempt 3/3]
18:10:20.636 SERIAL » 0x01030015e9                                                         (5 bytes)
18:10:22.238 DRIVER   Timeout occured waiting for ACK received - maximum transmission attempts for the current transaction reached, dropping it...
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
Error: Failed to send the message after 3 attempts
    at Driver.handleUnsuccessfulTransmission (/home/pi/testCode/node_modules/zwave-js/build/lib/driver/Driver.js:1119:47)
    at Timeout._onTimeout (/home/pi/testCode/node_modules/zwave-js/build/lib/driver/Driver.js:1387:18)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
18:10:22.807 DRIVER   destroying driver instance...
`

At least this showed me that my serial devices are working, but I don't know what I need to change to make this work. 

Thank you

AlCalzone, 2020-05-31:
First of all, you need to start the driver outside any handlers, like it is shown in the [docs](https://alcalzone.github.io/node-zwave-js/#/getting-started/quickstart)

Second, your devices are not responding at all. Are you sure that they are not in use by another application?

inteldon, 2020-05-31:
In the code I supplied you can see me attempt various methods, one of them was according to the docs, which made the system print the logs (also mentioned above). 

`
18:10:12.518 DRIVER opening serial port /dev/ttyAMA0
18:10:12.544 DRIVER serial port opened
18:10:12.557 SERIAL » [NAK] (0x15)
18:10:14.080 DRIVER loading configuration...
18:10:16.161 DRIVER beginning interview...
`

I'm seeing the same response even when I use the serial usb port '/dev/ttyUSB0'. Please note that these ports work when I use them with openzwave-shared and with Domoticz and HomeAssistant. 

Thanks

inteldon, 2020-05-31:
Update - I found the problem. When I reinstalled the raspberry pi it started to recognize my device as '/dev/ttyACM0' instead. I was still using the '/dev/ttyUSB0'. I don't know why this happened, but it's fixed at least I don't see failures like before. 

Thanks

closed 2020-06-01 by AlCalzone