zwave-js/zwave-js #870: NetworkKey feature request

Durban-Designer, 2020-06-20:
Hello AlCalzone,

Loving the library but was curious if there was a timeline on the NetworkKey feature? I would love to assist if you need a bit of developer muscle, considering this library for an IOT project at my startup. If all goes well this code would be the core of a few key integrations. I sponsored you as a first step towards hopefully a long partnership.

Sincerely,

Royce

AlCalzone, 2020-06-20:
S0 security is already implemented, just not documented. The `options` argument of the driver constructor has a property `networkKey`, which accepts a 16-byte buffer. Example:
```js
const driver = new Driver("COM4", {
	networkKey: Buffer.from([
		// You probably shouldn't use this key:
		1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16,
	])
})
```

The `beginInclusion` method in the controller accepts a boolean parameter to indicate whether the inclusion should be performed non-secure. By default, if a network key is set, the driver tries to include the device securely:
```js
// secure:
const inclusionStarted = await driver.controller.beginInclusion(/* or false */);
// non-secure:
const inclusionStarted = await driver.controller.beginInclusion(true);
```
Note that some devices require a different activation for secure inclusion. E.g. Aeotec devices only include the Security CC in their NIF if the button was pressed twice in a short time.

> I sponsored you as a first step towards hopefully a long partnership.

Thanks for that :)

Durban-Designer, 2020-06-20:
Thank you so much for the rapid response!! I will be tinkering with this over the weekend and waking up to such detailed explanations really made my day!!!

AlCalzone, 2020-06-22:
I'm going to close this thread as it seems to be a non-issue. Feel free to reopen if necessary.

closed 2020-06-22 by AlCalzone