zwave-js/zwave-js #872: unable to set multilevel sensor values

inteldon, 2020-06-22:
Hi

Thanks for your work on this project. I'm trying a few tests to add/remove and control sensors and wrote this basic code: 

'
const { Driver } = require('zwave-js');
const driver = new Driver("/dev/ttyUSB1", {
	// prettier-ignore
	networkKey: Buffer.from([ 
		1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16,
	])
})
	.on("error", console.error)
	.once("driver ready", async () => {

                // await driver.controller.beginInclusion(); // demo inclusion code is on a separate file 
                const wallPlug = driver.controller.nodes.get(17);
		await wallPlug.commandClasses.Basic.set(255);
}); driver.start(); 
'

I tried the ."..Basic.set()" with several values from boolean to 0-100 or 0-255 and there was no result. I don't know what I"m doing wrong although I can see that the sensors get included/excluded in the network successfully. 

I know the sensor was added correctly because I got this message when I ran the inclusion code: 
[image]

AlCalzone, 2020-06-22:
This library hides the Basic CC values in favor of more appropriate CCs like Multilevel Switch. It might be that this behavior interferes with the API you're using. Since Multilevel Switch is in the list of supported CCs, you should probably use this:
```
await wallPlug.commandClasses["Multilevel Switch"].set(99);
```

FYI, the multilevel value range in Z-Wave is 0-99 (not 100) and the single value 255 which just means "on".

And one more thing: 
Before trying to control a node, you should wait for the Node's [ready event](https://alcalzone.github.io/node-zwave-js/#/api/node?id=quotreadyquot).

closed 2020-06-22 by AlCalzone

inteldon, 2020-06-23:
Thank you. 

I tried your suggestion and modified my code to the following: 
```
const wallPlug = driver.controller.nodes.get(18);
wallPlug.on("ready", async()=>{
	await wallPlug.commandClasses["Multilevel Switch"].set(99);		
});
```
The result was that nothing happened. If I turned on the plug from the button it would automatically turn off after a few seconds. Here are some of the messages printed on the console: 

I'm assuming node 1 is the controller

[image]

and node 18 below is my wall plug 

[image]

AlCalzone, 2020-06-23:
It would be easier if you posted the screenshot as text.

A Z-Wave network has to be interviewed before it can be used in a meaningful way. This takes a lot of bandwidth and time (several minutes up to hours if you have a lot of battery-powered nodes). Only after this is done, you should attempt communication with nodes. The next time you start the driver, most of the information is cached, so the communication may be started earlier. 

Both situations are indicated through the "ready" event, which has not yet fired in your case.

inteldon, 2020-06-25:
Thanks I'll give it a try. I have a couple of wall plugs and some sensors that were previously added and now removed, leaving me with node 18 only, which is a mains powered wall plug. 

I'm wondering why the controller was trying to interview nodes 2, 5, 6, 11, 12, 13, 14 and 15 as I don't have them in my network. I started thinking of this when you mentioned the controller might take a while to interview the nodes, so this could be that this information is cached and needs to be cleared so the controller needs to only interview the ones that are current (node 18 in my case), should I try running network healing? or deleting those other nodes? 

I'll run some tests and share results. 

Thanks again

AlCalzone, 2020-06-25:
I guess you didn't remove them from the network. The node IDs are stored in the controller hardware, so even clearing the cache files won't help. You should probably remove them from the controller with `removeFailedNode(nodeId)`:
https://github.com/AlCalzone/node-zwave-js/blob/f1f7b4291054a02a57a598bcc650ddf85668f2f2/packages/zwave-js/src/lib/controller/Controller.ts#L1706

commit 1caeadd7 (2020-06-22): fix: assume that the BasicCC API is always supported

fixes: #872