zwave-js/zwave-js #888: Hangs on CRC16 Encapsulation Command and NotificationCCReport

slangstrom, 2020-06-27:
So, I've experimented with node-zwave-js over the weekend and I'm having a problem with FGMS-001 Z-wave+ (Fibaro Motion Sensor), this is the only sensor I've tried right now. I can include it and everything goes fine, until the device tries to report "motion".

I see this in the log:

23:09:13.356 SERIAL « 0x0113000400080d56017105000000ff070800e8895f                        (21 bytes)
23:09:13.704 SERIAL » [ACK] (0x06)
23:09:13.716 DRIVER « [Node 008] [REQ] [ApplicationCommand]
                      └─[CRC16CCCommandEncapsulation]
                        │ payload: 0x7105000000ff070800e889
                        └─[NotificationCCReport]
                            payload: 0x000000ff070800

After this, nothing. Completely quiet. I have to restart the node process to get any new data.

AlCalzone, 2020-06-28:
I would have to see a full log to know whats going on. I've never had the case that the driver freezes after receiving something and I know there are other users with the same sensor.

slangstrom, 2020-06-29:
I have been digging some more about this specific issue and payload, and it looks like [this](https://github.com/AlCalzone/node-zwave-js/blob/ec070c71ecc774ca943036aa74ab536ccebf7521/packages/zwave-js/src/lib/driver/Driver.ts#L1946) while()-loop runs indefinitely.

I also included a AEON Labs Motion Sensor 6 and this works great, although that one is not using CRC16 Encap.

AlCalzone, 2020-06-29:
Nice work! I like reports like this :)
I'll release a patch in the next minutes.

commit a932fb8e (2020-06-29): fix(cc): actually do something when unwrapping CRC16

fixes: #888
closed 2020-06-29 by AlCalzone