zwave-js/zwave-js #895: When the interview reports no targetValue, should it be set to currentValue?

AlCalzone, 2020-07-02:
Relevant for all Switch type CCs

AlCalzone, 2020-10-05:
Solved in the adapter via an option

robertsLando, 2020-11-22:
IMO the correct behavior is to set the target to current when it doesn't automatically report it

darkbasic, 2020-11-22:
But how do we know if it doesn't automatically report it?

AlCalzone, 2020-11-22:
Depending on the CC version the device supports

darkbasic, 2020-11-22:
I though that my Qubino had the new CC version but yet it still didn't report the target.

AlCalzone, 2020-11-22:
I didn't take a look at your logs yet but it is entirely possible that it only reports the current value although it supports reporting both.

robertsLando, 2020-11-23:
@AlCalzone How does ozw handles this? Anyway I'm +1 for the solution you proposed here

AlCalzone, 2020-11-23:
They update the target value if it is included in the report:
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/SwitchBinary.cpp#L116-L124

I have a feeling that exposing `targetValue` as `undefined` isn't even what I intended to happen. I'll have to check it again.

robertsLando, 2020-11-23:
> I have a feeling that exposing targetValue as undefined isn't even what I intended to happen. I'll have to check it again.

I think exposing it as undefined is never a good option, when it's undefined it should be set to the current always

AlCalzone, 2020-11-23:
The current behavior is not working as intended (was broken by https://github.com/zwave-js/node-zwave-js/pull/1103)

darkbasic, 2020-11-27:
@AlCalzone did #1139 end up in 5.5.0?

AlCalzone, 2020-11-27:
It did

kpine, 2021-01-12:
I did not follow the entire conversion, but the current state of this is confusing, could you help clarify?

I have a Multilevel V2 Switch, which has no target value support. Here's what I've observed:

- A targetValue with read/write access is exposed to the application. The targetValue is used for a Set, but has no relevance for reporting values
- A currentValue with read access is exposed to the application. This is the only value I can currently use for the dimmer state.

How should an application (namely HA), which should support all versions of switches, deal with these values? Currently HA uses the targetValue to report brightness. My switch currently shows a brightness of 100% because it's converting 255 to max brightness. In reality, the 255 was the toggle command from the Set.

It cannot simply check for the existence of targetValue, because it seems to always be defined, but invalid for V2 switches.

Should it check the version of the CC to determine if it should look at targetValue? Applications should be insulated from knowing these details.

Can the targetValue be set to write-only for < V4 switches? An application could check if it's readable, if not use the currentValue.

Should it, for now, always use currentValue? Ideally, the application would want to set targetValue if it's available to avoid incorrect state.

Some of these questions would be answered based on how this bug is ultimately fixed. But for now, what would be the right guidance?

AlCalzone, 2021-01-12:
If you want to report a state, use `currentValue`. `targetValue` is used to control the device.
If a device has support for it, `targetValue` *can* be used to advertise which target a device is transitioning to, e.g. after physically switching a roller shutter.

closed 2021-02-03 by AlCalzone