zwave-js/zwave-js #938: Support devices that send a MultiChannelCCCommandEncapsulation (0x0D) but with a V1 CCCommand (0x06)

apla2016, 2020-09-04:
I am switching from zwave to zwave2. Several devices which were recognized in zwave are not fully recognized in zwave. All devices and parameters are available on Object page, but some are missing on Device Tab of Instances page.

Elexa Consumer Products Inc. (Dome) On/Off Plug-In Switch
manufacturid: 543
productid: 135
productType: 3

2GIG Technologies CT101 Thermostat (Iris)
manufacturid: 152
productid: 12
productType: 25857

closed 2020-09-04 by AlCalzone

AlCalzone, 2020-09-04:
This should be fixed in the next adapter version

apla2016, 2020-09-04:
Thank you! Beer is on me...

Anatoly

apla2016, 2020-09-05:
Thank you, works fine now!

Anatoly

AlCalzone, 2020-09-07:
> Thank you! Beer is on me... Anatoly

Much appreciated :)

apla2016, 2020-09-22:
Hello,

Sorry to bother you again. I have an issue with zWave device that you 
fixed for me - Radio Thermostat Company of America (RTC) CT101. It seems 
to be operational in general, but I have some issues:

- If I change setpoint on the actual thermostat, I see changes in 
corresponding Object value

- If I change setpoint in Object value (via VIS gadget), I don't see 
changes applied to the thermostat setpoint

- The room temperature displayed on the actual thermostat does not get 
updated into Object Value

Could it be related to latest fix that you implemented? Or, since your 
version of zWave provides more parameters for devices than the old one, 
I did not set one of the device configuration parameter?

Thank you,

Anatoly

AlCalzone, 2020-09-22:
Please open a new issue with that description and attach a log file so I don't have to guess.

PR #939 (2020-09-04): Accept misformatted V2+ Multi Channel Command Encapsulation with V1 header
fixes: #938