zwave-js/zwave-js #974: Inclusion of Aeotec Wallmote (Quad) hangs when interviewing configuration params

slangstrom, 2020-09-15:
I have tried to include Aeotec Wallmote a few times and it always hangs when doing the same thing. I have included the whole logfile from start of inclusion to end. When reading the manual, there are only parameter 1-4, not 5 or 39.

First it tries to interview configuration parameter #5, which fails.

_10:06:43.662 CNTRLR » [Node 041] querying parameter #5 value...
[...]
10:06:50.176 CNTRLR   [Node 041] The node did not respond after 3 attempts.
                      It is probably asleep, moving its messages to the wakeup queue.
10:06:50.184 CNTRLR   [Node 041] The node is now asleep.
10:06:50.221 CNTRLR « [Node 041] received no value for parameter #5_

And after that it tries parameter #39, which wont work since the node is asleep.

_10:06:50.223 CNTRLR » [Node 041] querying parameter #39 value..._

[wallmote.txt](https://github.com/AlCalzone/node-zwave-js/files/5223431/wallmote.txt)

AlCalzone, 2020-09-15:
Which version of the library are you using? This should be fixed in the current alpha (https://github.com/AlCalzone/node-zwave-js/commit/00718ee4216f3eb8976091e4e4003ca1fe06ab3a), but I wouldn't recommend using that in production yet, since it has issues with some secure devices.
This shouldn't be a critical problem in 4.x though, except that the interview will take a few breaks while the node is assumed asleep.

From what I've read, parameter #39 should already be there in the old firmware revisions, even though it is not documented. #5 was added in a firmware revision after 1.5, but I'm not sure which one, so I assumed 1.6. My 1.8 device does have it - which version do you have?
Once I've corrected the config files, your issue should go away.

slangstrom, 2020-09-15:
Oh, great. I actually run alpha.1 and maybe that's the issue here. I'll try to upgrade and see if the problem disappears. I've got alot of Z-Wave devices (around 50) and there were some problems (retransmission, etc) with 4.x that has been fixed in alpha.x, so it's more stable for me.

I think this is version 2.3:

10:06:31.571 CNTRLR   [Node 041] [+] [Version] firmwareVersions: 2.3                    [Endpoint 0]

AlCalzone, 2020-09-15:
Huh, I didn't know there is a 2.x firmware. Maybe that's a different hardware. If that doesn't have parameter #39 either, please let me know and will remove them for the 2.x firmware line.

slangstrom, 2020-09-15:
Hang on, I did link to the wrong product in the first post. The correct one is [here](https://aeotec.freshdesk.com/support/solutions/articles/6000162392-wallmote-quad-user-guide-) although there's no parameter #5/#39 there either.

Either way, I tried the inclusion again in alpha.3 and it worked great. Thank you!

AlCalzone, 2020-09-15:
Does the log or values.jsonl contain any reference to these parameters?

slangstrom, 2020-09-15:
From log:

11:34:36.280 CNTRLR   [Node 044] ConfigurationCC: Loading configuration parameters from device config
11:34:36.311 CNTRLR   [Node 044] [+] [Configuration] isParamInformationFromConfig: false  [internal]
11:34:36.329 CNTRLR   [Node 044] [Configuration] 1: metadata updated
11:34:36.366 CNTRLR   [Node 044] [Configuration] 2: metadata updated
11:34:36.372 CNTRLR   [Node 044] [Configuration] 3: metadata updated
11:34:36.399 CNTRLR   [Node 044] [Configuration] 4: metadata updated
11:34:36.404 CNTRLR   [Node 044] [Configuration] 5: metadata updated
11:34:36.439 CNTRLR   [Node 044] [Configuration] 39: metadata updated
11:34:36.495 CNTRLR   [Node 044] [Configuration] 255: metadata updated
11:34:36.500 CNTRLR   [Node 044] [~] [Configuration] isParamInformationFromConfig: false  [internal] => true

values.jsonl:

{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":\"isParamInformationFromConfig\"}","v":true}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":1}","v":1}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":2}","v":1}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":3}","v":1}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":4}","v":1}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":39}","v":20}
{"k":"{\"nodeId\":44,\"commandClass\":112,\"endpoint\":0,\"property\":\"interviewComplete\"}","v":true}
{"k":"{\"nodeId\":44,\"commandClass\":133,\"endpoint\":0,\"property\":\"groupCount\"}","v":9}

metadata.jsonl: [metadata.jsonl.txt](https://github.com/AlCalzone/node-zwave-js/files/5224007/metadata.jsonl.txt)

slangstrom, 2020-09-15:
Here's a manual which includes references to paramter 5 so it does look like it exists: https://cdn.shopify.com/s/files/1/0066/8149/3559/files/24_wallmote_quad_-_es.pdf

To configure which color will be displayed when the button is pressed.
Value1 = Red.
Value2 = Green.
Value3 = Blue.
Value4 = Reserved.

AlCalzone, 2020-09-15:
Ok that looks like your device (or firmware) does have #39, but not #5. Thanks for confirming.

slangstrom, 2020-09-15:
No problem. This is a **great** library, thanks for all the work.

AlCalzone, 2020-09-15:
This should be fixed in v4.2.2, I'll port the changes to master next.

PR #975 (2020-09-15): Backport 00718ee - Don't change node status if they don't respond to ConfigurationCC::Get
fixes: #974
closed 2020-09-15 by AlCalzone