zwave-js/zwave-js #980: Reading Config from Fibaro FGBS001 fails

ivykittycat, 2020-09-21:
Starting the ZWave2 Instance fails to complete the interview for device FGBS001 from Fibaro named Node 13 leaving the Node unusable. All other devices  of type FGR222, FGD212 & FGS223 work as expected. 

Test was done with ZWave2 Version 1.5.0 and 1.7.0 alpha 2 with slightly different behaviour in the log file:

V1.5.0.  Node 13 Interview started.  ( But no interview completed) and no config-data accessible through objects.
V1.7.0a2 :  Node 13 has returned from the dead followed 6 sec later by Node 13 is dead. 

no other message in the logfile. 

Versions in use:
iobroker.admin 4.0.10
node.js: 12.18.3
npm: 6.14.6

System in use:
ZWave Stick Z-Wave.Me HS1DS via USB connected to /dev/ttyACM0
Raspberry with Raspbian

In contrast operation succeeds when executed with traditional instance ZWave Version 2.0.1.

Please advise

ivykittycat, 2020-09-21:
An idea from far away: Do I have a chance to track the behaviour of the old ZWave instance, because it somehow manages to work around that issue? 
And in principle, would it make sense to have in the object tree for a device a leaf with "failed features" where all features with implementation errors are listed as a string by name. In this case "Alarm sensor CC" to gracefully catch the interview stall.

AlCalzone, 2020-09-21:
AFAIK, the old Z-Wave adapter just skips every CC interview step that fails. While that is preferable in many cases, it would not allow to detect non-responsive devices during the interview the way the new adapter is written.
I have an exception in place for multiple CCs (e.g. Binary Sensor), which I should really extend to all sensor type CCs (which are prone to not answering).

AlCalzone, 2020-09-22:
The fix might take a few more days to be released, but it will be in the next adapter version

AlCalzone, 2020-09-22:
I'll be adding your idea to an issue I've been planning for a while now, seems like a good addition.

closed 2020-09-22 by AlCalzone

ivykittycat, 2020-09-28:
1.7.0 final fixed the problem. Thanks for your support, excellent job.

PR #984 (2020-09-22): Skip interview of sensor-type CCs when a timeout occurs
fixes: #980