zwave-js/zwave-js #985: Add statistics

AlCalzone, 2020-09-22:
Things that should be considered (total and per node):

- [ ] Transmitted messages
- [ ] Received messages
- [ ] Dropped messages
- [ ] Controller errors (CAN, NAK, timeout)
- [ ] Node errors (NoACK, timeout, ...)
- [ ] Response time
- [ ] Failures during interview

LordMike, 2021-01-10:
> Bonus to having events per message, rather than just counters, is also being able to see how long it actually took to deliver that "turn on" message. Would help immensely in tracking down where delays are occurring, when lights appear slow.

https://github.com/zwave-js/node-zwave-js/issues/1270#issuecomment-757505434

nicoh88, 2021-01-21:
+1 for `last seen`

jaimevisser, 2021-04-13:
Especially on bigger z-wave networks this would be a big help in tuning sensor data output.

Masterz69, 2021-11-06:
Missing ```last seen``` property on Home Assistant for ZWave devices.

Just today had an situation: FGBS-222 was powered off for ~25 minutes.
Device is configured to report every 5 minutes, couple of messages not received by controller.
ZWAVEJS2MQTT device ```node status``` still ```alive``` :( 
- as no configuration on device level: when to declare device  to not ```alive``` when no messages received from.

IMHO I can't relay on ```node status``` to determine actual device status.
```last seen``` will help to react on devices sending no info long time.

AlCalzone, 2023-07-12:
`lastSeen` will be added in https://github.com/zwave-js/node-zwave-js/pull/6008