zwave-js/zwave-js #1034: Unsupported notification for Fibaro FGS-101 when triggering water detected

slangstrom, 2020-10-10:
Currently testing an Fibaro Flood Sensor (FGS-101). When triggering the water detected (by just putting it on a piece of wet paper) I get the following message in the log:

13:11:34.528 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[NotificationCCReport]
                          V1 alarm type:  5
                          V1 alarm level: 2
13:11:34.529 CNTRLR « [Node 002] received unsupported notification {
                          "nodeId": 2,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 5,
                          "alarmLevel": 2
                      }

I have attached both "values" and "metadata" files.

[f1e28f04.metadata.txt](https://github.com/AlCalzone/node-zwave-js/files/5358883/f1e28f04.metadata.txt)
[f1e28f04.values.txt](https://github.com/AlCalzone/node-zwave-js/files/5358884/f1e28f04.values.txt)

AlCalzone, 2020-10-10:
This is a V1 Alarm CC which was renamed to Notification CC starting with V3. The handler for the notification CC does translation of the values, automatic reset to idle, etc. This is not yet implemented for the legacy versions because they are not standardized.

slangstrom, 2020-10-10:
Okay, I see, but don't you think these events (unsupported notification) should be emitted on the node somehow? Then it's possible to handle these edge cases anyway.

AlCalzone, 2020-10-10:
Yeah, they probably should just be stored as a simple value ID (`alarm_<alarmType>`) with value `alarmLevel`.

slangstrom, 2020-10-10:
Yes, sounds good! 👍

slangstrom, 2020-10-10:
Altough, if you check the "metadata"-file you can see that it has the following:

{"k":"{\"nodeId\":2,\"commandClass\":113,\"endpoint\":0,\"property\":\"Water Alarm\",\"propertyKey\":\"Sensor status\"}","v":{"type":"number","readable":true,"writeable":false,"min":0,"max":255,"label":"Sensor status","states":{"0":"idle","2":"Water leak detected"}}}

So it has detected that "Water Alarm" is a supported notification type

AlCalzone, 2020-10-10:
Interesting, could you show me the cache files and create a log where you force a water alarm?

slangstrom, 2020-10-10:
I don't have access to the device right now, but here's a log from earlier today. Do you want the <homeid>.jsonl too? metadata and values are attached in my first post.

Isn't this related to https://github.com/AlCalzone/node-zwave-js/blob/aed8b4dde771bc3f2e7bac98a8d2b773e71bb3bd/packages/zwave-js/src/lib/commandclass/AlarmSensorCC.ts#L40? Or is that Alarm CC V2 and above?

**Triggering water detected:**

10:47:33.531 SERIAL « 0x0123000400661d9881e336ba910ff2e09cb3da8d2450193751e635529bc03cba5 (37 bytes)
                      001b78919
10:47:33.540 SERIAL » [ACK]                                                                   (0x06)
10:47:33.553 DRIVER « [Node 102] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[NotificationCCReport]
                            V1 alarm type:  5
                            V1 alarm level: 2
10:47:33.573 CNTRLR « [Node 102] received unsupported notification {
                          "nodeId": 102,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 5,
                          "alarmLevel": 2
                      }
10:47:35.507 SERIAL « 0x0108000400660298404f                                              (10 bytes)

**And when no water detected:**

10:47:35.763 SERIAL « 0x0123000400661d988119092d5a151b7716c33f758cea6efd24954bf9c9c0c4841 (37 bytes)
                      fec24ffac
10:47:35.770 SERIAL » [ACK]                                                                   (0x06)
10:47:35.790 DRIVER « [Node 102] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[NotificationCCReport]
                            V1 alarm type:  5
                            V1 alarm level: 0
10:47:35.813 CNTRLR « [Node 102] received unsupported notification {
                          "nodeId": 102,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 5,
                          "alarmLevel": 0
                      }

AlCalzone, 2020-10-10:
Yes I need the json and values.jsonl file.

Alarm CC is part of Notification CC. Alarm Sensor CC is something different.

slangstrom, 2020-10-10:
Oh, of course. Here's the file. Nevermind the different nodeIds, it's two different computers.

[main.txt](https://github.com/AlCalzone/node-zwave-js/files/5359632/main.txt)

.

AlCalzone, 2020-10-10:
Interesting... that device supports Notification CC V5, but chooses to send V1/V2 alarm frames. 
I think in that case the driver could check whether the type and level translates to a supported notification type and event. If yes, it could be handled like one.

closed 2020-10-10 by AlCalzone

AlCalzone, 2020-10-10:
This should be fixed in `v5.3.0` which is available in a couple of minutes.

PR #1036 (2020-10-10): feat: treat Alarm V1 frames as Notification Reports if possible
fixes: #1034