zwave-js/zwave-js #1037: Inclusion of Remotec ZRC-90 fails

slangstrom, 2020-10-12:
So I tried to include my Remotec ZRC-90, "Scene master", but the inclusion fails pretty quickly. Based on the log file attached I don't even think it's being interviewed. I attached my cache-files too, there's nothing in them though.

[remoteczrc90_log.txt](https://github.com/AlCalzone/node-zwave-js/files/5362993/remoteczrc90.txt)
[eabbf325.txt](https://github.com/AlCalzone/node-zwave-js/files/5362996/eabbf325.txt)
[eabbf325.metadata.txt](https://github.com/AlCalzone/node-zwave-js/files/5362997/eabbf325.metadata.txt)
[eabbf325.values.txt](https://github.com/AlCalzone/node-zwave-js/files/5362998/eabbf325.values.txt)

slangstrom, 2020-10-12:
I also tried to include Everspring ST812, the inclusion seems to work but got "the node is unknown or not initialized yet, discarding..." there too. Do you want me to open a separate issue for that one?

AlCalzone, 2020-10-12:
Since both issues seem to affect the node inclusion, you can leave it here.

AlCalzone, 2020-10-12:
> but the inclusion fails pretty quickly. Based on the log file attached I don't even think it's being interviewed.

It seems that the inclusion went through but the interview was not started afterwards although it should be. You can probably work around it by restarting your application until I've found the source of the problem.

slangstrom, 2020-10-13:
Here's the files for Everspring ST812. When I restart and then manually wake up Remotec, the interview process gets completed. 

[e81a9a19.metadata.txt](https://github.com/AlCalzone/node-zwave-js/files/5369019/e81a9a19.metadata.txt)
[e81a9a19.txt](https://github.com/AlCalzone/node-zwave-js/files/5369020/e81a9a19.txt)
[e81a9a19.values.txt](https://github.com/AlCalzone/node-zwave-js/files/5369021/e81a9a19.values.txt)
[everspring_st812_inclusion.txt](https://github.com/AlCalzone/node-zwave-js/files/5369022/everspring_st812_inclusion.txt)

slangstrom, 2020-10-15:
Issue with inclusion of ID Lock too, don't know if it's related to the above though. Attached log files.

[idlock_inclusion.txt](https://github.com/AlCalzone/node-zwave-js/files/5383834/idlock_inclusion.txt)

AlCalzone, 2020-10-15:
That seems to be something else. The device is included and the interview starts immediately, but then it fails to respond to some messages. First it doesn't respond to `SecurityCCCommandsSupportedGet` which causes the driver to think it is not secure (which is wrong because it responsed to nonce requests). Then it doesn't respond to Version requests (because the `VersionCC` is probably only supported securely).

However I'm not seeing the security bootstrapping process in the log. Could it be that the device was previously included securely but you did an insecure inclusion? In that case you might have to factory-reset the device.

slangstrom, 2020-10-15:
Alright! This was actually the first inclusion attempt after I opened the box, but let me try to reset it and try again.

slangstrom, 2020-10-15:
So I did a hard reset of my PC Z-Wave module and then did a hard reset of the Z-Wave module on the ID Lock. Still the same, attaching the inclusion log and all the cache-files.

[c3ed057b.metadata.txt](https://github.com/AlCalzone/node-zwave-js/files/5384651/c3ed057b.metadata.txt)
[c3ed057b.txt](https://github.com/AlCalzone/node-zwave-js/files/5384652/c3ed057b.txt)
[c3ed057b.values.txt](https://github.com/AlCalzone/node-zwave-js/files/5384653/c3ed057b.values.txt)
[idlock_inclusion.txt](https://github.com/AlCalzone/node-zwave-js/files/5384654/idlock_inclusion.txt)

AlCalzone, 2020-10-15:
This time I'm seeing the bootstrapping process succeed. But the device does something weird as soon as the driver tries to communicate securely. The normal flow is
```
--> Get Nonce
<-- Nonce Report
--> Encrypted Request
<-- Nonce Request
--> Nonce Report
<-- Encrypted Response
```

What I'm seeing is:
```
--> Get Nonce
<-- Nonce Report
--> Encrypted Request
... nothing for one second ...
<-- Encrypted Response (unknown nonce)
```

I'm not sure whats going on and where the nonce is coming from that the device uses to encrypt the response. The encrypted response is discarded because the driver did not create any nonce for that exchange.

slangstrom, 2020-10-16:
Is there anything I can do to debug more? I tried to include it in a friends Fibaro Home Center and it worked fine.

AlCalzone, 2020-10-16:
There's no way to see the communication logs in Home Center, is it?

slangstrom, 2020-10-16:
Nope, they have a basic "Inclusion started", "Interviewing device", etc but nothing more... I guess it would be possible to try open-zwave, could that help somehow?

AlCalzone, 2020-10-16:
The OZW log would probably help, yeah!

slangstrom, 2020-10-17:
Here's the log file from OpenZWave, hope it helps!

[idlock_openzwave.txt](https://github.com/zwave-js/node-zwave-js/files/5395760/idlock_openzwave.txt)

AlCalzone, 2020-10-17:
In this log, the node asks the controller for a nonce. Very weird. Wonder why it didn't in the other log.
Maybe try re-interviewing the node in zwave-js (or delete the cache)?

slangstrom, 2020-10-18:
So I ran zwave-js and here's the output from the interview process. I used the same security network key in both zwave-js and OpenZWave.

[idlock_interview.txt](https://github.com/zwave-js/node-zwave-js/files/5397361/idlock_interview.txt)

AlCalzone, 2020-10-18:
Hm that looks better - maybe a temporary hiccup?
The interview mostly goes through, except for the User Code CC which does something it shouldn't.
```
12:35:12.501 CNTRLR   [Node 002] UserCodeCC: doing a complete interview...
12:35:12.502 CNTRLR » [Node 002] querying capabilities...
12:35:12.504 CNTRLR   [Node 002] Error during node interview: Node #2 does not support the command C
                      apabilitiesGet!
```

slangstrom, 2020-10-18:
Well, yes, the only difference is that I didn't _include_ it in zwave-js, I did that in openzwave.

AlCalzone, 2020-10-18:
Could you try doing the same again in zwave-js now? Wonder if it has to do with the inclusion process.

slangstrom, 2020-10-18:
You mean include it with zwave-js? Reset the lock, reset the z-wave stick and try the inclusion?

AlCalzone, 2020-10-18:
yup

slangstrom, 2020-10-19:
Still the same problem, tried multiple

[idlock_inclusion_new.txt](https://github.com/zwave-js/node-zwave-js/files/5401097/idlock_inclusion_new.txt)
 times.

AlCalzone, 2020-10-19:
Did you do a complete reset? The device does not respond to the NetworkKeySet command, after which the driver must treat it as a non-secure device. The weird thing is that it asks for a nonce to send a reply, but that reply never arrives.

I'll take a look at what OZW does differently.

slangstrom, 2020-10-19:
Yes, did a reset of both the lock and the stick. The only difference I can tell without looking at the code is that in openzwave, you pass true as an argument when doing the inclusion to enable the security inclusion process while in zwave-js it's managed automatically.

AlCalzone, 2020-10-19:
I'll do a 1:1 comparison of the data exchange later to see if there is any difference.

AlCalzone, 2020-10-19:
So, I've analyzed the first few messages in both inclusion logs (OZW and the last zwave-js):

**zwave-js**

1. Inclusion:
    1. NodeFound
    1. AddingSlave
    1. ProtocolDone
    1. Done
2. Security Bootstrap
```
 --> SchemeGet (0)
<--  SchemeReport (0)
 --> NonceGet (for NetworkKeySet)
<--  NonceReport
 --> CommandEncapsulation > NetworkKeySet
<--  NonceGet (for the response to NetworkKeySet!)
 --> NonceReport
... nothing ...
```

**OZW**

1. Inclusion:
    1. NodeFound
    1. AddingSlave
    1. ProtocolDone
    1. Done
2. Security Bootstrap
```
 --> SchemeGet (0)
<--  SchemeReport (0)
 --> NonceGet (for NetworkKeySet)
<--  NonceReport
 --> CommandEncapsulation > NetworkKeySet
<--  NonceGet (for the response to NetworkKeySet!)
 --> NonceReport
<--  A message that OZW does not understand.
     zwave-js parses it as NetworkKeyVerify with the configured key (the expected response to NetworkKeySet)
```

---

The only difference I see is that OZW requires an acknowledgement for the NonceReport and does not use the "Explorer Frame" option. I'm not entirely sure why that would make a difference, but we can try that with a test build I can prepare later tonight.

AlCalzone, 2020-10-19:
Version `5.3.2-alpha.0` will be available in a couple of minutes. Please try the inclusion with that version again and create a log. The nonce reports are now transmitted using the same options that OZW uses, so there should be no more difference in communication.

slangstrom, 2020-10-19:
I just tried and still the same issue... I think. Did a full reset of the stick and lock and also had the lock 20-30 cm from the z-wave stick when doing the inclusion just to be sure that there were no problems with range. Attached the log.

[idlock_inclusion_alpha0.txt](https://github.com/zwave-js/node-zwave-js/files/5404077/idlock_inclusion_alpha0.txt)

AlCalzone, 2020-10-19:
The security bootstrap process succeeds now:
```
20:48:47.236 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[SecurityCCNetworkKeyVerify]
```

But we're back to this problem https://github.com/zwave-js/node-zwave-js/issues/1037#issuecomment-709296495 now.
Looking at the OZW log it seems that it also receives an invalid response to that command, but chooses to ignore it.
I'm working on a PR to make the interview less strict in many cases, this should be one of them.

AlCalzone, 2020-10-21:
This should be fixed in `v5.3.2`

slangstrom, 2020-10-22:
Alright, tried all three devices in 5.3.2. Remotec, Everspring and ID-Lock. I was not able to include neither of them. Both Remotec and Everspring did same as last time. No interview for Remotec at all, and for Everspring the interview is being made but still says "unknown node" after it's done. I did a reset of everything before I began.

[everspring_st812_inclusion.txt](https://github.com/zwave-js/node-zwave-js/files/5421005/everspring_st812_inclusion.txt)
[idlock_inclusion.txt](https://github.com/zwave-js/node-zwave-js/files/5421006/idlock_inclusion.txt)
[remotec_inclusion.txt](https://github.com/zwave-js/node-zwave-js/files/5421007/remotec_inclusion.txt)

AlCalzone, 2020-10-22:
Ohhh, I think I found whats going wrong with the remotec. It gets added as a controller, not as a slave (like all the devices I've seen so far). As a result, a variable does not get set. I'm tracking that part in https://github.com/zwave-js/node-zwave-js/issues/1069 now.

ID-Lock now gets past the inclusion process but does the same it did earlier during the inclusion. It responds without requesting a new nonce, so the driver does not know the nonce it used for encryption. I'll have to check if it is reusing previous nonces or something.

The everspring device has me confused. Directly after the inclusion, the node should be added to `controller.nodes` with the information you see in the log:
```
09:26:09.795 CNTRLR   finished adding node 3:
                        basic device class:    Static Controller
                        generic device class:  Alarm Sensor
                        specific device class: Routing Alarm Sensor
                        supported CCs: 
                        · Basic (0x20)
                        [...]
```
But the message "the node is unknown or not initialized yet" implies that the nodeid is not found in `controller.nodes`. The only location where one can be deleted is after **exclusion**.
Can you check if the `"node removed"` is emitted some time during the attempted inclusion?

slangstrom, 2020-10-22:
1. Alright, worth noting about Remotec is that if I first include it, restart zwave-js and then waking it up, it gets interviewed and works like it should (Central Scene commands gets handled by zwave-js)
2. Ok, let me know if there's something I can do.

3. Everspring

   1. "node removed" is **not** emitted.
   2. The node is present in the cache-files
   3. For debug I put a console.log on driver.controller.nodes during the interview-process and the node is present there too, during and after the interview.
   4. I did notice it always fails on the same interview step:

`20:46:10.190 CNTRLR » [Node 003] querying current value for Water Leak...
20:46:10.229 SERIAL » 0x010a001303039c0105250853                                          (12 bytes)
20:46:10.231 DRIVER » [Node 003] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      8
                      └─[AlarmSensorCCGet]
                          sensor type: Water Leak
20:46:10.236 SERIAL « [ACK]                                                                   (0x06)
20:46:10.241 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:46:10.242 SERIAL » [ACK]                                                                   (0x06)
20:46:10.244 DRIVER « [RES] [SendData]
                        was sent: true
20:46:10.258 SERIAL « 0x0107001308000002e1                                                 (9 bytes)
20:46:10.259 SERIAL » [ACK]                                                                   (0x06)
20:46:10.262 DRIVER « [REQ] [SendData]
                        callback id:     8
                        transmit status: OK
20:46:11.874 CNTRLR   [Node 003] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
20:46:12.389 SERIAL » 0x010a001303039c0105250853                                          (12 bytes)
20:46:12.392 DRIVER » [Node 003] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      8
                      └─[AlarmSensorCCGet]
                          sensor type: Water Leak
20:46:12.398 SERIAL « [ACK]                                                                   (0x06)
20:46:12.405 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:46:12.407 SERIAL » [ACK]                                                                   (0x06)
20:46:12.410 DRIVER « [RES] [SendData]
                        was sent: true
20:46:12.420 SERIAL « 0x0107001308000002e1                                                 (9 bytes)
20:46:12.422 SERIAL » [ACK]                                                                   (0x06)
20:46:12.425 DRIVER « [REQ] [SendData]
                        callback id:     8
                        transmit status: OK
20:46:12.743 SERIAL « 0x010d00040003079c02000500000069                                    (15 bytes)
20:46:12.749 SERIAL » [ACK]                                                                   (0x06)
20:46:12.754 DRIVER « [REQ] [ApplicationCommand]
                      └─[AlarmSensorCCReport]
                          sensor type: Water Leak
                          alarm state: false
20:46:12.765 DRIVER     the node is unknown or not initialized yet, discarding...
`

AlCalzone, 2020-10-23:
Remotec:
Then it has probably two modes of inclusion: Secondary controller and/or slave - could that be?

Regarding the everspring device:
Could you set a breakpoint [here](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/driver/Driver.ts#L1607) and inspect `controller.nodes` as well as the detected node ID?
I'm also a bit stumped by the fact that this code is even hit during the interview. The `AlarmSensorCCReport` should be matched to the previous `AlarmSensorCCGet`.

slangstrom, 2020-10-23:
Remotec: No, I don't think so. According to the manual this is the way to do it. I did try a few different ways but none worked.

Everspring: attached two print screens from the debugger output

[image]
[image]

AlCalzone, 2020-10-23:
Huh, `nodeId: 0`...
That seems like a bug in `AlarmSensorCCReport` (or the node sends corrupted Alarm Sensor commands).

AlCalzone, 2020-10-23:
Yup, that's a bug in the CC implementation (or I misunderstood the specs)

AlCalzone, 2020-10-23:
Tracking that bug in #1072, so the only thing left here is the ID Lock and its strange nonce usage.

AlCalzone, 2020-10-25:
So, I've taken a closer look at the idlock log. The device **does** reuse nonces, which is clearly forbidden by the Z-Wave specs:

1. The driver sends a nonce (id 0x7f)
```
09:44:18.789 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      10
                      └─[SecurityCCNonceReport]
                          nonce: 0x7f216b115b3055b3
```

2. The device sends a message using that nonce
```
09:44:18.845 SERIAL « 0x01260004000220988126fbb14a2e48ea38e65073649a6585730fe5a7e7617f451 (40 bytes)
                                                                                    ^^
                      445eff88623eb25
09:44:18.845 SERIAL » [ACK]                                                                   (0x06)
09:44:18.846 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[SecurityCCCommandsSupportedReport]
                            reportsToFollow: 0
                            supportedCCs:
                            · Version
                            · Door Lock
                            · User Code
                            · Association
                            · Association Group Information
                            · Notification
                            · Firmware Update Meta Data
                            · Battery
                            controlledCCs:
```

3. a bunch of unencrypted communication

4. The driver does an encrypted request
```
09:44:18.954 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      13
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 47
                        └─[VersionCCGet]
```

5. The device responds, but reuses the same nonce it has used above
```
09:44:19.835 SERIAL « 0x0125000404021f9881a22080f5d642ed82adc5ac6ad6606740027d0f197f62985 (39 bytes)
                                                                                  ^^
                      ea9a71dbc5cac
09:44:19.836 DRIVER   Message with invalid data received. Dropping it
```

AlCalzone, 2020-10-25:
Seems that the ID lock supports firmware updates - unfortunately, I can't see the current version in either OZW or here, since in both cases the VersionReport cannot be decoded. The [changelog](https://idlock.no/kundesenter/idl150updater-version/) is pretty vague regarding the Z-Wave fixes.
Could you ask the support if reusing S0 nonces is something that was fixed in an update?

slangstrom, 2020-10-25:
Yeah, the lock supports firmware and the version on my lock is 1.5.6, which is the current version. I can try to contact them tomorrow. I did try to include the lock in a Homey and that worked like it should. I don't know if the log files can help but here's from the inclusion in my Homey.

[idlock_homey_inclusion.txt](https://github.com/zwave-js/node-zwave-js/files/5435897/idlock_homey_inclusion.txt)

AlCalzone, 2020-10-26:
I'm not too familiar with the Homey log format, but I think that the same is happening there too - Homey just doesn't care like OZW.
If I'm correct, the device seems to reuse nonces after an unencrypted message exchange (lines 433ff):
```log
# Node requests a new nonce
[2020-10-25T20:17:44.752Z] Node[3]: Received application command for COMMAND_CLASS_SECURITY, data: 0x40
[2020-10-25T20:17:44.752Z] Node[3]: [COMMAND_CLASS_SECURITY] {}
[2020-10-25T20:17:44.754Z] Node[3]: [COMMAND_CLASS_SECURITY] [SECURITY_NONCE_REPORT] {"type":"Buffer","data":[17,220,166,43,149,123,179,127]}
[2020-10-25T20:17:44.755Z] Node[3]: sendData to COMMAND_CLASS_SECURITY, params 0x8011dca62b957bb37f
[2020-10-25T20:17:44.756Z] Command[56] start: sendData
[2020-10-25T20:17:44.757Z] ProcessSendData[55]: To node: 3 with data: 0x988011dca62b957bb37f and txOptions: AUTO_ROUTE,EXPLORE
[2020-10-25T20:17:44.783Z] Command[56] end: sendData

# Node responds with that nonce (ID 17)
[2020-10-25T20:17:44.814Z] Node[3]: Received application command for COMMAND_CLASS_SECURITY, data: 0x814adb0c27e5605f8ae809754ccf115d812be98a9bdd7d
[2020-10-25T20:17:44.814Z] Node[3]: [COMMAND_CLASS_SECURITY] {"Initialization Vector byte":{"type":"Buffer","data":[74,219,12,39,229,96,95,138]},"Properties1 (Raw)":{"type":"Buffer","data":[232]},"Properties1":{"Sequence Counter":8,"Sequenced":false,"Second Frame":true},"Command byte":{"type":"Buffer","data":[9,117,76,207]},"Receivers nonce Identifier (Raw)":{"type":"Buffer","data":[17]},"Receivers nonce Identifier":17,"Message Authentication Code byte":{"type":"Buffer","data":[93,129,43,233,138,155,221,125]}}
[2020-10-25T20:17:44.836Z] Node[3]: Decapsulated frame from COMMAND_CLASS_SECURITY to COMMAND_CLASS_VERSION, data 0x148001
[2020-10-25T20:17:44.836Z] Node[3]: [COMMAND_CLASS_VERSION] {"Requested Command Class (Raw)":{"type":"Buffer","data":[128]},"Requested Command Class":128,"Command Class Version (Raw)":{"type":"Buffer","data":[1]},"Command Class Version":1}
[2020-10-25T20:17:44.840Z] Command[4] status: addNode, status: INTERVIEW_CC_15_OF_15

# Controller sends an unencrypted CC
[2020-10-25T20:17:45.357Z] Command[4] status: addNode, status: INTERVIEW_MANUFACTURER_SPECIFIC
[2020-10-25T20:17:45.404Z] Node[3]: sendData to COMMAND_CLASS_MANUFACTURER_SPECIFIC, params 0x04
[2020-10-25T20:17:45.405Z] Command[57] start: sendData
[2020-10-25T20:17:45.406Z] ProcessSendData[56]: To node: 3 with data: 0x7204 and txOptions: AUTO_ROUTE,ACK,EXPLORE
[2020-10-25T20:17:45.460Z] Command[57] end: sendData

# Node replies unencrypted
[2020-10-25T20:17:45.478Z] Node[3]: Received application command for COMMAND_CLASS_MANUFACTURER_SPECIFIC, data: 0x05037300030001
[2020-10-25T20:17:45.498Z] Node[3]: [COMMAND_CLASS_MANUFACTURER_SPECIFIC] {"Manufacturer ID":{"value":883},"Product Type ID":{"value":3},"Product ID":{"value":1}}
[2020-10-25T20:17:45.503Z] Command[4] status: addNode, status: INTERVIEW_FIRMWARE_INFORMATION

# Controller requests a new nonce for the upcoming secure request
[2020-10-25T20:17:45.504Z] Node[3]: sendData to COMMAND_CLASS_VERSION, params 0x11
[2020-10-25T20:17:45.505Z] Node[3]: [COMMAND_CLASS_SECURITY] [SECURITY_NONCE_GET] {"type":"Buffer","data":[]}
[2020-10-25T20:17:45.506Z] Node[3]: sendData to COMMAND_CLASS_SECURITY, params 0x40
[2020-10-25T20:17:45.507Z] Command[58] start: sendData
[2020-10-25T20:17:45.509Z] ProcessSendData[57]: To node: 3 with data: 0x9840 and txOptions: AUTO_ROUTE,ACK,EXPLORE
[2020-10-25T20:17:45.544Z] Command[58] end: sendData
[2020-10-25T20:17:45.562Z] Node[3]: Received application command for COMMAND_CLASS_SECURITY, data: 0x80d7fbddf6da7d4dfd
[2020-10-25T20:17:45.562Z] Node[3]: [COMMAND_CLASS_SECURITY] {"Nonce byte":{"type":"Buffer","data":[215,251,221,246,218,125,77,253]}}

# Controller does an encrypted request
[2020-10-25T20:17:45.585Z] Node[3]: [COMMAND_CLASS_SECURITY] [SECURITY_MESSAGE_ENCAPSULATION] {"type":"Buffer","data":[45,89,69,38,210,34,22,220,195,191,202,215,144,172,220,243,21,218,144,71]}
[2020-10-25T20:17:45.586Z] Node[3]: sendData to COMMAND_CLASS_SECURITY, params 0x812d594526d22216dcc3bfcad790acdcf315da9047
[2020-10-25T20:17:45.586Z] Command[59] start: sendData
[2020-10-25T20:17:45.587Z] ProcessSendData[58]: To node: 3 with data: 0x98812d594526d22216dcc3bfcad790acdcf315da9047 and txOptions: AUTO_ROUTE,ACK,EXPLORE
[2020-10-25T20:17:45.628Z] Command[59] end: sendData

# Node responds encrypted (twice [!], with the already-used nonce 17)
[2020-10-25T20:17:46.471Z] Node[3]: Received application command for COMMAND_CLASS_SECURITY, data: 0x8180e207f5acd7d18eb445cc4453360a527268ec54113df90e5cc7dc5503
[2020-10-25T20:17:46.471Z] Node[3]: [COMMAND_CLASS_SECURITY] {"Initialization Vector byte":{"type":"Buffer","data":[128,226,7,245,172,215,209,142]},"Properties1 (Raw)":{"type":"Buffer","data":[180]},"Properties1":{"Sequence Counter":4,"Sequenced":true,"Second Frame":true},"Command byte":{"type":"Buffer","data":[69,204,68,83,54,10,82,114,104,236,84]},"Receivers nonce Identifier (Raw)":{"type":"Buffer","data":[17]},"Receivers nonce Identifier":17,"Message Authentication Code byte":{"type":"Buffer","data":[61,249,14,92,199,220,85,3]}}
[2020-10-25T20:17:46.714Z] Node[3]: Received application command for COMMAND_CLASS_SECURITY, data: 0x813b50bc404902fee7a071a64417b1ae82a68ba41d11d310d9bb7e119897
[... cut ...] ta":[17]},"Receivers nonce Identifier":17,"Message Authentication Code byte":{"type":"Buffer","data":[211,16,217,187,126,17,152,151]}}
```

I'm certain that this is a bug in the device. Other drivers just don't care, but reusing nonces is a security risk (putting aside that S0 is less secure than S2 anyways).
Please let me know if the manufacturer is going to do something about this. If not, I can implement a flag to disable nonce expiration for this device only, so that you can at least use it.

slangstrom, 2020-10-29:
Okay, so I tried to contact ID Lock support 3 days ago (monday) and still no reply... could we do a workaround to disable the nonce expiration until they reply? I did read somewhere that even if the lock is updateable, the Z-Wave module is not, so it's possible that this is the only solution either way.

AlCalzone, 2020-11-04:
@slangstrom The linked PR **should** fix your issue. Would you like an alpha release to test?

roflmao, 2020-11-26:
@slangstrom The zwave module is possible to firmwareupgrade: https://idlock.no/wp-content/uploads/2019/11/OTA-guide-v1.4.pdf
I have upgraded both my idlock 150 to 1.6 firmware.

AlCalzone, 2020-11-26:
> The zwave module is possible to firmwareupgrade:

**Note:** you don't need any extra software, you don't have to unpair the devices. Just upgrade from zwave-js

roflmao, 2020-11-26:
@AlCalzone That is pure awesomeness! I will test it for sure on other devices.

robertsLando, 2020-11-26:
@roflmao Also zwavejs2mqtt allows to do that using the UI :)

roflmao, 2020-12-14:
@AlCalzone Trying to add my idlock 150 with the latest 1.6.0 firmware (Node 028)

[zwave-1.log](https://github.com/zwave-js/node-zwave-js/files/5691948/zwave-1.log)

AlCalzone, 2020-12-15:
Uh that doe

AlCalzone, 2020-12-16:
@roflmao can you try the `fix-id-lock` branch?

roflmao, 2020-12-21:
@AlCalzone Tried latest master: 
[zwave-1.log](https://github.com/zwave-js/node-zwave-js/files/5726847/zwave-1.log)

AlCalzone, 2020-12-21:
Hmm this issue is like whack-a-mole:
```
23:30:45.451 CNTRLR   [Node 028] trying to load device config
23:30:45.452 CNTRLR   [Node 028] no device config loaded
```

roflmao, 2020-12-21:
@AlCalzone I dont seem to find any config in the zwave-js repository either..

AlCalzone, 2020-12-21:
There is one for specific firmware versions and #1212 adds a fallback one so the version CC interview can be completed:
https://github.com/zwave-js/node-zwave-js/pull/1212/files#diff-30921b89cc20aec16480118e42f425219ba2c86b65e1866bde169bd3965aba58

roflmao, 2020-12-21:
Right.

How about a log from a complete OZW interview?

AlCalzone, 2020-12-21:
Not necessary. Loading the device config file must succeed, so the rest of the interview can continue. I can't reproduce locally though - my test passes 🤔

AlCalzone, 2020-12-21:
Can you please check if you have the file `config\devices\0x0373\id-150_compat.json` in the `@zwave-js/config` module? And if `id-150_compat.json` appears in the `config/devices/index.json`?

roflmao, 2020-12-21:
Oh, i just broke index.json because of https://github.com/zwave-js/node-zwave-js/pull/1236.

AlCalzone, 2020-12-21:
If the files is there, running `yarn run config -i` should include it in the index - so all good.

[15 more replies]

PR #1055 (2020-10-20): Ignore several non-critical timeouts during CC interview
fixes: #1050
fixes: #1037 (assuming #1053 is merged)
PR #1095 (2020-11-11): Improve nonce handling in Security CC and device compatibility
With this PR, we now store both the `issuer` of a nonce (previously `nodeId`) and the `receiver` (the node for which a nonce was generated). This allows us to be more spec-compliant, since all nonces for a receiver should be invalidated once it has replied with one of them.

This also enables us to selectively **not** invalidate nonces to improve compatibility with devices that reuse nonces like ID Lock.

fixes: #1037
PR #1212 (2020-12-20): feat(config): support config files without firmware version, load them after Manufacturer Info is known
With this PR, we add support for config files which explicitly have no firmware version (`firmwareVersion: false`). This allows us to look up those fallback files directly after the `Manufacturer Specific CC` interview, before the `Version CC` is interviewed.

The change is required to support devices like the ID lock, which require compat flags to even complete the Version CC interview.

fixes: #1037
PR #1262 (2021-01-01): fix: delete "keep until next" nonces AFTER the next one was received
Prior to this PR we would expire nonces before sending a new one - which is fine for most devices. However the ID Locks keep using the previous nonce until they have confirmed that they received a new nonce.

Now we only expire the old one after sending the new one was successful.

fixes: #1037 (I hope)
closed 2021-01-01 by AlCalzone