zwave-js/zwave-js #1039: Collaboration

robertsLando, 2020-10-15:
Hi @AlCalzone, I'm Daniel the author of [Zwave2Mqtt](https://github.com/OpenZWave/Zwave2Mqtt). 

Me and @chrisns would be really excited to move z2m to use your zwave-js implementation, actually it's based on OZW but for many reasons we would prefer to have more control/support on the lower zwave side. I really love your js implementation and I could also help you here with the code if you are interested. I'm sure things can happen :) 

Also @chrisns sent you a more detailed private email some days ago about this. Let us know :)

AlCalzone, 2020-10-15:
Hey guys, thanks for your interest in this library. I'm sure some help here and there would be nice. 

@chrisns I checked both my inbox and spam folders but I found nothing. Mind sending the email again?

chrisns, 2020-10-15:
pasted:
Hi!

Daniel professionally and I recently with a bit of my occasional hobbyist time recently pitching in maintain [zwave2mqtt](https://github.com/OpenZWave/Zwave2Mqtt).

I happened to google land upon your repo and was wondering if you'd be interested in a brief chat about your project, the vision/roadmap, we're really interested+excited in porting over and removing the openzwave dependency for a host of reasons, looking at the code it might not even be that hard to do, and make zwave2mqtt blazing fast too!

Maybe grab a quick 30 minute google meet chat? (beers/coffee optional depending on time zones), and/or join the [zwave2mqtt slack](https://join.slack.com/t/zwave2mqtt/shared_invite/enQtNjc4NjgyNjc3NDI2LTc3OGQzYmJlZDIzZTJhMzUzZWQ3M2Q3NThmMjY5MGY1MTc4NjFiOWZhZWE5YjNmNGE0OWRjZjJiMjliZGQyYmU) - because I'm sure you needed another slack!

Let me know if you're up for it and timezone and I'll set something up. 

Cheers

Chris (chrisns) & Daniel (robertsLando)

chrisns, 2020-10-15:
we also kinda maintain (if you can call our little attention to it that) https://github.com/OpenZWave/node-openzwave-shared

Ideally we'd like to explore to swapping out the underline ozw component and shim in yours providing some abstraction to maintain the api so that app devs can then migrate to using your lib overtime in a strangler pattern. running both your lib and the ozw at the same time on a physical device is probably a bit hard/unstable?

AlCalzone, 2020-10-15:
I'd be open for a chat / messaging, probably easier to squeeze that in than a fixed appointment. So slack is probably the way to go? Haven't used it yet.

Running both libraries at the same time won't work, since both need to have exclusive data exchange with the Z-Wave stick in order to function properly and maintain cache integrity. You can switch between them without resetting the device or anything, but only one can run at a given time.

robertsLando, 2020-10-15:
@AlCalzone join slack here: https://join.slack.com/t/zwave2mqtt/shared_invite/zt-8ns655f6-d407vtI~KjU~1z11jyaQ9Q

closed 2020-10-16 by AlCalzone