zwave-js/zwave-js #1040: zwave-js throws build errors when used due to leaked internal types

RoboPhred, 2020-10-16:
I am upgrading a project from zwave-js@2 to zwave-js@5.3.1

I have fixed all build errors due to changed types, but zwave-js itself is throwing errors due to unknown types that seem to only be used internally.

Here are the errors produced:
```
node_modules/@alcalzone/jsonl-db/build/lib/db.d.ts:1:21 - error TS7016: Could not find a declaration file for module 'fs-extra'. 'D:/dev/git/RoboPhred/wutwot/packages/wutwot-zwave/node_modules/fs-extra/lib/index.js' implicitly has an 'any' type.
  Try `npm install @types/fs-extra` if it exists or add a new declaration (.d.ts) file containing `declare module 'fs-extra';`

1 import * as fs from "fs-extra";
                      ~~~~~~~~~~

node_modules/@zwave-js/core/build/test/SpyTransport.d.ts:1:23 - error TS2688: Cannot find type definition file for 'jest'.

1 /// <reference types="jest" />
                        ~~~~

node_modules/@zwave-js/core/build/test/SpyTransport.d.ts:8:16 - error TS2503: Cannot find namespace 'jest'.

8     get spy(): jest.Mock;
                 ~~~~

```

It seems that in order to install zwave-js, I either need to turn on `skipLibCheck`, or install @types/fs-extra and @types/jest

I am not sure how the compiler is finding a reference to these files.  The files under build/test are especially problematic, as I would not think anything I am touching should be trying to load test files.

AlCalzone, 2020-10-16:
Could you try using the `--traceModuleResolution` flag https://www.typescriptlang.org/docs/handbook/module-resolution.html#tracing-module-resolution and send me the output so I can have an idea where to start investigating?

RoboPhred, 2020-10-16:
Here is a module trace.
[build-trace.log](https://github.com/zwave-js/node-zwave-js/files/5392541/build-trace.log)

SpyTransport seems to come directly from the index
```
======== Resolving module './test/SpyTransport' from 'D:/dev/git/RoboPhred/wutwot/packages/wutwot-zwave/node_modules/@zwave-js/core/build/index.d.ts'. ========
```

jsonl-db seems to be pulled in by ValueDB
```
======== Resolving module '@alcalzone/jsonl-db' from 'D:/dev/git/RoboPhred/wutwot/packages/wutwot-zwave/node_modules/@zwave-js/core/build/values/ValueDB.d.ts'. ========
```

AlCalzone, 2020-10-16:
> export * from "./test/assertZWaveError";

is not critical, since it is used internally in multiple modules and only imports ZWaveError and ZWaveErrorCodes.

> export * from "./test/SpyTransport";

really does not make sense at runtime however. I'll have to investigate whether it makes sense splitting these methods into another dev-only package.

> It takes a typing import against jsonl-db, which follows a chain of imports and eventually demands fs-extra.

The problem here is that one of the public methods takes an argument that uses fs-extra's typings. I'll need to rework how that is done. ==> https://github.com/AlCalzone/jsonl-db/issues/53

closed 2020-10-16 by AlCalzone

AlCalzone, 2020-10-21:
fixed in `v5.3.2`

PR #1047 (2020-10-16): Update jsonl-db to elimitate exported references to fs-extra
After this change, `--traceResolution` shows no references to jest or fs-extra

fixes: #1040