zwave-js/zwave-js #1041: Missing type exports

RoboPhred, 2020-10-16:
After upgrading to zwave-js@5, many typescript types have become available from the top level zwave-js export.  However, there are a few I use in my project that still are not exported from the top-level, and I believe they are good candidates:

- ZWaveController [from `zwave-js/build/lib/controller/Controller`]
- Endpoint [from `zwave-js/build/lib/node/Endpoint`]
- InterviewStage [from `zwave-js/build/lib/node/Types`.  Both the type and enum object should be exported]

I believe these three are missing, as I was unable to trace their export back to the index.d.ts file.

AlCalzone, 2020-10-16:
Are you sure about `ZWaveController`? `index.d.ts` contains `export * from "./Controller"`, which contains a type-export `export type { HealNodeStatus, ZWaveController, } from "./build/lib/controller/Controller";`.
In a consuming app, I can navigate from the root-level import all the way to the implementation.

closed 2020-10-16 by AlCalzone

RoboPhred, 2020-10-16:
You are right, ZWaveController is properly exported.  I must have missed that.

AlCalzone, 2020-10-21:
fixed in `v5.3.2`

PR #1043 (2020-10-16): fix(build): add/fix missing and wrong exports
fixes: #1041