zwave-js/zwave-js #1042: [5.3.1] Crash on encountering non-secure node (uncaught promise)

RoboPhred, 2020-10-16:
I recently upgraded zwave-js from 2 to 5, and am now encountering a hard crash during the interview process.

[zwave-9800.log](https://github.com/AlCalzone/node-zwave-js/files/5388547/zwave-9800.log)

Interestingly, with LOGTOFILE set, the error I am getting does not appear in the log, but seems to be printed to stderr by nodejz itself:

Here is the entire console output from the invocation of my program:
```
Logging to file:
D:\dev\git\RoboPhred\wutwot\packages\wutwot-server\lib\zwave-9800.log
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
Error: Node 024 supports Security, but no network key was configured. Continuing interview non-securely.
    at ZWaveNode.interviewCCs (D:\dev\git\RoboPhred\wutwot\packages\wutwot-zwave\node_modules\zwave-js\src\lib\node\Node.ts:1205:7)
    at ZWaveNode.interview (D:\dev\git\RoboPhred\wutwot\packages\wutwot-zwave\node_modules\zwave-js\src\lib\node\Node.ts:901:19)
    at Driver.interviewNode (D:\dev\git\RoboPhred\wutwot\packages\wutwot-zwave\node_modules\zwave-js\src\lib\driver\Driver.ts:783:21)
    at Driver.initializeControllerAndNodes (D:\dev\git\RoboPhred\wutwot\packages\wutwot-zwave\node_modules\zwave-js\src\lib\driver\Driver.ts:753:15)
    at Immediate.<anonymous> (D:\dev\git\RoboPhred\wutwot\packages\wutwot-zwave\node_modules\zwave-js\src\lib\driver\Driver.ts:622:5)
```

AlCalzone, 2020-10-16:
Can you share how you initialize the driver in your code? This "error" should only be emitted as an `"error"` event on the driver, which can cause crashes if no handler is attached. 
However, this should be enforced since 5.0.0, but that doesn't seem to be working in your case.

RoboPhred, 2020-10-16:
Alright, having this error emitted by an event is new to me.  I did have an error handler set up, but [it is only used in the initialization phase](https://github.com/RoboPhred/wutwot/blob/master/packages/wutwot-zwave/src/impl/ZWaveProviderImpl.ts#L36), and I remove the listener because I wanted the errors to be thrown globally if something critical went wrong.

I guess this means non-fatal errors are being emitted through the error handler then.  How can I tell the difference between fatal errors that stop the zwave module from operating, and security errors like this that the driver will recover from automatically?

AlCalzone, 2020-10-16:
I really should document this better. There is a common base class `ZWaveError`, which has a `code` property to identify the kind of error independent from the erroor message. So you could do:

```ts
driver.on("error", error => {
  if (error instanceof ZWaveError && error.code === ZWaveErrorCodes.Driver_Failed) {
    // fatal error
  } else {
    // non-fatal error
  }
});
```

RoboPhred, 2020-10-16:
I had a look through the ZWaveErrorCodes, and I would rather capture and dismiss known 'good' errors rather than only look for known fatal errors and ignore the rest.

To that end, it seems the error I am encountering is not a subclass of ZWaveError:

[image]

Could you introduce a status code for this particular error, so I can dismiss this particular one while still bailing for all other unknown errors?

RoboPhred, 2020-10-16:
The core resolution of this issue was that I need to always have a handler for driver errors, so this issue is completed.

I opened another issue for giving ZWaveError codes to all known errors that the driver can emit: https://github.com/zwave-js/node-zwave-js/issues/1044

closed 2020-10-16 by RoboPhred