zwave-js/zwave-js #1044: Ensure thrown errors are ZWaveError with appropriate code

RoboPhred, 2020-10-16:
Continuing from #1042

I want my application to selectively dismiss known recoverable errors, and shut down the program for all unknown errors encountered.

To this end, all known errors that can be recovered from should contain an appropriate ZWaveErrorCode.

I searched the code base for "new Error", and found several points where generic errors are being thrown without error codes:

- [LanguageCCSet.language](https://github.com/zwave-js/node-zwave-js/blob/582f52df2654fca991d79822ebfa6e48639d0bf1/packages/zwave-js/src/lib/commandclass/LanguageCC.ts#L145)
`ZWaveErrorCodes.Argument_Invalid`
- [LanguageCCSet.country](https://github.com/zwave-js/node-zwave-js/blob/582f52df2654fca991d79822ebfa6e48639d0bf1/packages/zwave-js/src/lib/commandclass/LanguageCC.ts#L161)
`ZWaveErrorCodes.Argument_Invalid`
- [Driver.restoreNetworkStructureFromCache](https://github.com/zwave-js/node-zwave-js/blob/6a7f0f92b881abd82744529a6a2f09dc591dc57e/packages/zwave-js/src/lib/driver/Driver.ts#L2286)
`ZWaveErrorCodes.Driver_InvalidCache`
- [Secure node without a security manager](https://github.com/zwave-js/node-zwave-js/blob/7b8146edc88f9c14e6700fbccebdef480a9028ed/packages/zwave-js/src/lib/node/Node.ts#L1205)
Not sure about this one, it probably deserves a new error code, maybe `Node_SecurityNotConfigured`

AlCalzone, 2020-10-16:
Thanks for the investigation! I've gone ahead and created a PR already

closed 2020-10-16 by AlCalzone

AlCalzone, 2020-10-21:
fixed in `v5.3.2`

PR #1046 (2020-10-16): fix: consistently use ZWaveErrors instead of plain Errors
fixes: #1044