zwave-js/zwave-js #1062: Configure logging through driver options

AlCalzone, 2020-10-21:
Ideally, this should have preference over env variables. Things to consider:

- [ ] minimum loglevel (currently `LOGLEVEL` env variable) 
- [ ] ability to turn logging off (loglevel: off), should also be supported through the `LOGLEVEL` env variable
- [ ] ability to provide a custom transport for logs (compatible with `winston`)

AlCalzone, 2020-11-24:
I'm open for suggestions how to elegantly pass these options through to the logger module in `packages\core\src\log\shared.ts`

robertsLando, 2020-11-24:
> I'm open for suggestions how to elegantly pass these options through to the logger module in packages\core\src\log\shared.ts

I would make the logger a singleton instance and add a method to configure those properties

AlCalzone, 2020-11-24:
PRs welcome :)

robertsLando, 2020-11-24:
I dunno how singletons are managed on Typescript but I could try :)

AlCalzone, 2020-11-24:
Just like JS :D

PR #1167 (2020-12-23): Prepare Release v6.0.0
This draft PR is to collect the work that goes into version 6.

fixes #1062
fixes: #1135
fixes: #1223
closed 2020-12-23 by AlCalzone