zwave-js/zwave-js #1067: Openzwave shared migration

robertsLando, 2020-10-22:
This issue will collect some questions about the migration from OZW shared and this library

robertsLando, 2020-10-22:
Lifecycle: 

The lifecycle of a node in zwavejs is: 

`node added`
`node  alive/awake`
`node ready`
`interview completed`

A node is `ready` once we know it is not `dead` and we have all the necessary information. That is:

* After the initial interview (the event happens at the same time as interview completed)
* When starting the driver after the node was already interviewed completely AND the node is (probably) asleep, alive or awake

While a node is being interviewed, a bunch of queries are made and you'll either see `value added` or `value updated` as well as some `metadata updated` events.
This interview starts once a node is either `awake` or `alive`  and after it was included.

robertsLando, 2020-10-22:
In openzwave-shared each valueId was identified by `<nodeId>-<cmdClass>-<instance>-<index>` Ex: `1-112-0-0` 

In zwavejs a valueid is identified by (https://zwave-js.github.io/node-zwave-js/#/api/valueid):

```ts
interface ValueID {
    commandClass: CommandClasses;
    endpoint?: number;
    property: number | string;
    propertyKey?: number | string;
}
```

AlCalzone, 2021-01-02:
@robertsLando do we still need this?

robertsLando, 2021-01-04:
Nope

closed 2021-01-04 by robertsLando