zwave-js/zwave-js #1070: Support controller replication etc.

AlCalzone, 2020-10-22:
According to INS13954-7, several more steps and different timeouts are involved when including a controller. We currently do the bare minimum, but should check out what to do when including a controller (and what that SIS/SUC stuff means 😅 )

robertsLando, 2020-11-02:
http://www.dtvrc.com/2019/02/06/difference-between-suc-and-sis-controller/

AlCalzone, 2020-11-02:
Well, that is better than the sad attempt at an explanation that can be found in the specs.

AlCalzone, 2021-05-11:
Nope and honestly it's not at the top of my to do list.

eliasisrael, 2021-05-11:
I understand. Lots to do and many requests. I'm interested to help if I can, but my hours available for coding are limited these days. Assuming I could do something, would that be a help?

eliasisrael, 2021-05-11:
OK, let me try to learn the code base and see if I can plausibly spend enough time to be useful. After that, we can talk about things.

kizza42, 2021-08-25:
Just chiming in on this issue as well. 
I have a different [Z-Wave hub](https://products.z-wavealliance.org/products/2720?selectedFrequencyId=-1) (I'm on the team developing it) as my primary controller and added a Aeotec USB Stick as a secondary controller, for the most part, things work well together with HA/ZWaveJS. I'm a bit jealous because you guys do few things a bit better than our implementation!! 
But I noticed the other day that new nodes added to my primary controller aren't being passed through to HA/ZWaveJS.
If I add items via HA/ZWaveJS they do pass back to my primary controller however...