zwave-js/zwave-js #1082: [feat] Replace failed node api

robertsLando, 2020-10-30:


robertsLando, 2020-11-10:
@AlCalzone Just for information, does this require a big effort to be implemented?

AlCalzone, 2020-11-10:
Probably not. Should be an API call, followed by the inclusion process.

robertsLando, 2020-11-10:
I'm looking at the code to try to implement this as I think it's an important feature but I have no idea where to start from :rofl:

AlCalzone, 2020-11-10:
I'll take a look at the serial api specs tomorrow and give you some pointers

AlCalzone, 2020-11-11:
1. Read INS13954 (in the specs directory), chapter *4.4.21 ZW_ReplaceFailedNode*. Some of it is probably duplicated with the add and remove node commands

1. You want to make sure that the API call's function type is included in
https://github.com/zwave-js/node-zwave-js/blob/f568e5ad205a713b0319504ac5af7b9e19f13b84/packages/zwave-js/src/lib/message/Constants.ts#L47
(it should be) and has a nice name, not `ZW_FUNC_ID_...`.

1. Make sure you have the project snippet extension installed for VSCode. This allows you to type the snippet name in a TypeScript file and autocomplete a bunch of boilerplate code.

1. Create a new file for the message classes in `packages\zwave-js\src\lib\controller` and create a boilerplate with the VSCode snippet `zwmsgcb` (if the message uses a callback) or `zwmsg` (if it doesn't). You might have to delete all imports and recreate them with auto imports - I didn't touch these snippets since the refactor.

robertsLando, 2020-11-11:
Ok so the function code seems to be `0x63`. I will try to implement it once the other pr is  merged

robertsLando, 2020-11-11:
> Make sure you have the project snippet extension installed for VSCode

Is this extension named `project snippet` ?

AlCalzone, 2020-11-11:
This is the one

PR #1117 (2020-11-18): feat(controller): implement replaceFailedNode API
Fixes #1082 

@AlCalzone Im't stucked now on the controller function :thinking: 

How do I get the node? How can I specify if I want to replace the node in secure mode or not? Is the inclusion started automatically?
closed 2020-11-18 by AlCalzone