zwave-js/zwave-js #1084: [bug] Wrong releases body

robertsLando, 2020-11-02:
The body of releases is wrong. I think the problem is [here](https://github.com/zwave-js/node-zwave-js/blob/master/.github/workflows/test-and-release.yml#L160-L164)

AlCalzone, 2020-11-02:
Thanks I didn't notice. Seems that the new GITHUB_ENV solution no longer supports the multiline encoding.

AlCalzone, 2020-11-02:
Solution is similar to
```
          echo 'ENV_VAR_1<<EOF' >> $GITHUB_ENV
          echo 'foo' >> $GITHUB_ENV
          echo 'bar' >> $GITHUB_ENV
          echo 'do not eval `this`' >> $GITHUB_ENV
          echo 'EOF' >> $GITHUB_ENV
          echo 'ENV_VAR_2=single line' >> $GITHUB_ENV
```

robertsLando, 2020-11-02:
@chrisns Does this bug remind something to you? :laughing:

robertsLando, 2020-11-02:
@AlCalzone What about add it to steps output?

chrisns, 2020-11-02:
Urgh. Giving me flashbacks

AlCalzone, 2020-11-02:
@chrisns Do you have a preferred solution for this? I'm just rolling with any one that works :)

AlCalzone, 2020-11-02:
But yeah, the steps output seems somewhat clean. I think that wasn't available when I first wrote that workflow.

chrisns, 2020-11-02:
On holiday will have a look when I'm back tomorrow

closed 2020-11-02 by AlCalzone

AlCalzone, 2020-11-03:
steps.output works :)

robertsLando, 2020-11-03:
@AlCalzone Saw it, good! :)

PR #1086 (2020-11-02): Use steps output to propagate version and release body
fixes: #1084