zwave-js/zwave-js #1089: Securely included node doesn't show up as secure

darkbasic, 2020-11-03:
[image]

I tried to include a Qubino Flush Dimmer in zwavejs2mqtt but unfortunately it shows up as non-secure. What's wrong?

[zwave-284642.log](https://github.com/zwave-js/node-zwave-js/files/5480684/zwave-284642.log)

```
$ npm start

> zwavejs2mqtt@0.0.0 start /home/niko/devel/zwavejs2mqtt
> node bin/www

  z2m:Store scenes.json not found +0ms
  z2m:Store nodes.json not found +2ms
  z2m:App zwavejs2mqtt version: 0.0.0 +0ms
  z2m:App Application path:/home/niko/devel/zwavejs2mqtt +1ms
  z2m:Mqtt MQTT is disabled +0ms
  z2m:Zwave Connecting to /dev/ttyACM0 +0ms
Logging to file:
/home/niko/devel/zwavejs2mqtt/bin/zwave-284642.log
  z2m:App Listening on port 8091 +0ms
  z2m:App New connection ohv-AeAq0ZkOcAPRAAAA +1s
  z2m:Zwave Zwave driver is ready +2s
  z2m:Zwave Driver ready +1ms
  z2m:Zwave Node added 1 +1ms
  z2m:Zwave Scanning network with homeid: 0xc54055a2 +0ms
  z2m:Zwave Node 1 is alive +59ms
  z2m:Zwave Node 1 doesn't support groups associations +3ms
  z2m:Zwave Node 1: value added 1-32-0-currentValue => undefined +3ms
  z2m:Zwave Node 1: value added 1-32-0-targetValue => undefined +1ms
  z2m:Zwave Node 1 ready: AEON Labs - ZW090 (Z‐Stick Gen5 USB Controller) +1ms
  z2m:Zwave Scan completed +0ms
  z2m:Zwave Network scan complete. Found: 1 nodes +1ms
  z2m:Zwave Node 1: interview completed, all values are updated +1ms
  z2m:App User disconnected ohv-AeAq0ZkOcAPRAAAA +8s
GET / 304 8.727 ms - -
GET /static/css/app.ac70a60bcc407fe11707.css 304 4.830 ms - -
GET /static/js/app.a4eeee65f64587f9e920.js 304 0.763 ms - -
  z2m:App New connection -xi4jW9V5OFHT317AAAB +396ms
GET /static/logo.png 304 0.690 ms - -
  z2m:App Zwave api call: startInclusion [ true ] +22s
  z2m:Zwave Secure inclusion started +29s
  z2m:Zwave Success zwave api call startInclusion true +1ms
  z2m:Zwave Node 2: added +13s
  z2m:Zwave Node added 2 +1ms
  z2m:Zwave Inclusion stopped +5ms
  z2m:Zwave Node 2 is alive +104ms
  z2m:Zwave Node 2: value added: 114-0-manufacturerId => 345 +52ms
  z2m:Zwave Node 2: value added: 114-0-productType => 1 +1ms
  z2m:Zwave Node 2: value added: 114-0-productId => 81 +0ms
  z2m:Zwave Node 2: value added: 134-0-libraryType => 3 +57ms
  z2m:Zwave Node 2: value added: 134-0-protocolVersion => 4.5 +0ms
  z2m:Zwave Node 2: value added: 134-0-firmwareVersions => 3.7 +0ms
  z2m:Zwave Node 2: value added: 94-0-zwavePlusVersion => 1 +490ms
  z2m:Zwave Node 2: value added: 94-0-nodeType => 0 +0ms
  z2m:Zwave Node 2: value added: 94-0-roleType => 5 +1ms
  z2m:Zwave Node 2: value added: 94-0-installerIcon => 7168 +0ms
  z2m:Zwave Node 2: value added: 94-0-userIcon => 7168 +1ms
  z2m:Zwave Node 2: metadata updated: 38-0-Up +45ms
  z2m:Zwave Node 2: metadata updated: 38-0-Down +1ms
  z2m:Zwave Node 2: value added: 38-0-currentValue => 0 +61ms
  z2m:Zwave Node 2: value added: 37-0-currentValue => false +65ms
  z2m:Zwave Node 2: metadata updated: 50-0-reset +50ms
  z2m:Zwave Node 2: metadata updated: 50-0-value-65537 +92ms
  z2m:Zwave Node 2: metadata updated: 50-0-previousValue-65537 +0ms
  z2m:Zwave Node 2: metadata updated: 50-0-deltaTime-65537 +1ms
  z2m:Zwave Node 2: value added: 50-0-value-65537 => 64 +0ms
  z2m:Zwave Node 2: value added: 50-0-deltaTime-65537 => 0 +0ms
  z2m:Zwave Node 2: metadata updated: 50-0-value-66049 +91ms
  z2m:Zwave Node 2: metadata updated: 50-0-previousValue-66049 +1ms
[... 153 lines cut ...]
  z2m:Zwave Node 2: value added 2-112-0-2 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-3 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-4 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-10 => 255 +1ms
  z2m:Zwave Node 2: value added 2-112-0-11 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-12 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-20 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-21 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-30 => 0 +1ms
  z2m:Zwave Node 2: value added 2-112-0-40 => 10 +0ms
  z2m:Zwave Node 2: value added 2-112-0-42 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-60 => 10 +0ms
  z2m:Zwave Node 2: value added 2-112-0-61 => 99 +0ms
  z2m:Zwave Node 2: value added 2-112-0-65 => 100 +1ms
  z2m:Zwave Node 2: value added 2-112-0-66 => 3 +0ms
  z2m:Zwave Node 2: value added 2-112-0-67 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-68 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-100 => 0 +0ms
  z2m:Zwave Node 2: value added 2-112-0-101 => 0 +1ms
  z2m:Zwave Node 2: value added 2-112-0-110 => 32536 +0ms
  z2m:Zwave Node 2: value added 2-112-0-120 => 5 +0ms
  z2m:Zwave Node 2: value added 2-112-0-250 => 0 +0ms
  z2m:Zwave Node 2 ready: Qubino - ZMNHDD (Flush Dimmer Plus) +2ms
  z2m:Zwave Node 2: interview completed, all values are updated +2ms
```

AlCalzone, 2020-11-03:
```
[Node 002] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Version
                      · Device Reset Locally
                      · Manufacturer Specific
                      · Powerlevel
                      · All Switch
                      · Binary Switch
                      · Multilevel Switch
                      · Meter
                      · Notification
                      · Association
                      · Multi Channel Association
                      · Association Group Information
                      · Configuration
                      controlled CCs:
                      · Multilevel Switch
```

Either the device does not support security (then a "secure" inclusion will just be a normal inclusion) or you need to do a special activation sequence during inclusion. Aeotec devices for example require a double press of the button. Not sure about Qubino

darkbasic, 2020-11-03:
The device does support security (but only S0 and not S2 according to Qubino) and there is no special inclusion procedure documented in the manual to enable it.

AlCalzone, 2020-11-03:
Ok could you re-include it and send me a log of that?

darkbasic, 2020-11-03:
The log I provided should already contain the inclusion part, doesn't it?
If you look at the console output the inclusion is the first thing I did after starting zwavejs2mqtt.

AlCalzone, 2020-11-03:
Oh yeah, you're right. The node information frame it sends definitely does not include the `Security CC`. It might be that you have to factory-reset the device in order to re-include it securely.
Normally the list of supported (unencrypted) CCs is pretty short - this is not the case here.

darkbasic, 2020-11-03:
I did factory reset it before inclusion...

AlCalzone, 2020-11-03:
Then I don't know what to do. If the device does not tell the controller that it supports Security, it cannot be included securely.

darkbasic, 2020-11-03:
This is weird, because if I recall correctly OZW 1.4 used to show my devices as secure, while OZW 1.6 has the same issue as zwave-js.
Also I have proof that it's somehow possible to show it as secure:
[image]

This is a screenshow of latest openzwave2mqtt, but unfortunately I don't remember what I did differently to show such specific device as secure :(

AlCalzone, 2020-11-03:
I actually believe it is a problem of your specific device. I remember a report (Qubino ZMNHSD1) where the device completely forgot that it supports `Security CC` but somehow got stuck in secure mode (thus listing no supported Command Classes except Z-Wave+ CC and another one).

I think it would be worth contacting the Qubino support with that problem. When you do, please include the following log excerpt, so they can parse the device information themselves from that serial message. Otherwise they'll probably tell you it is a problem with the controller software:
```
11:33:55.298 SERIAL « 0x0107004a02060200b4                                                 (9 bytes)
11:33:55.299 SERIAL » [ACK]                                                                   (0x06)
11:33:55.301 DRIVER   handling request AddNodeToNetwork (74)
11:33:55.302 CNTRLR   finished adding node 2:
                        basic device class:    Static Controller
                        generic device class:  Multilevel Switch
                        specific device class: Multilevel Power Switch
                        supported CCs: 
                        · Basic (0x20)
                        · Multilevel Switch (0x26)
                        · All Switch (0x27)
                        · Z-Wave Plus Info (0x5e)
                        · Version (0x86)
                        · Device Reset Locally (0x5a)
                        · Manufacturer Specific (0x72)
                        · Powerlevel (0x73)
                        · Binary Switch (0x25)
                        · Meter (0x32)
                        · Notification (0x71)
                        · Association (0x85)
                        · Multi Channel Association (0x8e)
                        · Association Group Information (0x59)
                        · Configuration (0x70)
                        controlled CCs: 
                        · Multilevel Switch (0x26)
```

petrovcicklemen, 2020-11-09:
@darkbasic There are a couple of options:
- you own an older revision of the device, that doesn't support S0
- you own a device, which has the [configurable parameter 250](https://products.z-wavealliance.org/ProductManual/File?folder=&filename=Manuals/2114/Qubino_Flush%20Dimmer%20PLUS%20extended%20manual_eng_1.10.1.pdf) for enabling/disabling s0 inclusion 
- a failed S0 inclusion interview, due to which the device is included in unsecure mode
- a missing S0 key, that is not configured in Zwave2Mqtt

darkbasic, 2020-11-10:
> you own an older revision of the device, that doesn't support S0

I've bought 25 of them on December 2018. The reseller didn't have any stock so they came directly from Qubino.

> you own a device, which has the configurable parameter 250 for enabling/disabling s0 inclusion

This is the first time I see this. In the copy of the manual I downloaded from your website when I first bought these devices there is no mention of parameter 250 at all.

> a failed S0 inclusion interview

Why could it fail?

> a missing S0 key, that is not configured in Zwave2Mqtt

I do have a network key, not sure if I need a separate one just for S0 but I can't configure more than one

Also please note that I experience the same issue with my Flush Shutters (I bought 16 of them on Dicemeber 2018) with a VERY recent firmware because I had to return all of them in 2020 to update the firmware due to bugs.

AlCalzone, 2020-11-10:
> Why could it fail?

Missing required messages, elapsed timeouts between subsequent messages, etc.. This shouldn't be the case since the S0 bootstrapping takes place **after** the `AddNodeToNetwork` is handled (which already doesn't show Security S0 in its list).

> I do have a network key, not sure if I need a separate one just for S0 but I can't configure more than one

zwave-js only supports S0 so far, so there is only one key.

> there is no mention of parameter 250

Can you check in the dashboard if it is there? The config file for ZMNHDD does have that parameter.

darkbasic, 2020-11-10:
> Can you check in the dashboard if it is there? The config file for ZMNHDD does have that parameter.

It's in the dashboard, but the config contains parameters for all different kind of firmwares.

> zwave-js only supports S0 so far, so there is only one key.

Really? That's a big problem because lots of devices don't support S0 at all because of bugs in the zwave SDK which lead to all sort of problems including packets loss. Some vendors have patched it, others only support S2.

darkbasic, 2020-11-10:
@petrovcicklemen the same problem happens with latest openzwave by the way

AlCalzone, 2020-11-10:
> Really?

I already have a S2 supporting device lying on my TODO heap. Implementing it is not trivial, especially without devices to test (which I didn't until someone lent me one recently). [OZW](https://github.com/OpenZWave/open-zwave/issues/1827) and [OpenHAB](https://community.openhab.org/t/z-wave-s2/68624/10) do not support S2 either, so if everything goes as planned, `zwave-js` is going to be one of the first that do.

petrovcicklemen, 2020-11-10:
> I've bought 25 of them on December 2018. The reseller didn't have any stock so they came directly from Qubino.
Yes, and they could have the mentioned revision, where the parameter is implemented. Especially, since you are observing this on multiple different gateways. So there are two options:
a) try enabling the mentioned parameter and reinclude the device
b) use the PC Controller app, to see if the device has the parameter and if it includes in S0 (so you test without the gateway's overhead).

> This is the first time I see this. In the copy of the manual I downloaded from your website when I first bought these devices there is no mention of parameter 250 at all.
The website manual is for the latest revision, which is also stated at the end of the manual:
[image]
You should have received the manual, for your revision, in the form of a small pamflet, that was packaged in the blisters.

> Also please note that I experience the same issue with my Flush Shutters (I bought 16 of them on Dicemeber 2018) with a VERY recent firmware because I had to return all of them in 2020 to update the firmware due to bugs.
I'm almost 99% those Shutters support S0, so I think there is something else going on here. If you have the option of obtaining the mentioned PC application, that would be of big help.

darkbasic, 2020-11-10:
> The website manual is for the latest revision, which is also stated at the end of the manual

@petrovcicklemen I'm not talking about the website manual available right now, I'm talking about the website manual I downloaded and printed when I initially bought my devices:

[image]

[image]

As you can see it says SW version S3 (the one I have) and it doesn't mention parameter 250 at all.

> You should have received the manual, for your revision, in the form of a small pamflet, that was packaged in the blisters

I did, but as you can see from the following images there is no mention of parameter 250 either:

[image]

[image]

The whole thing is completely undocumented.

> If you have the option of obtaining the mentioned PC application, that would be of big help

I would need to get a Windows PC in order to do so, because my sole OS is Linux.

> try enabling the mentioned parameter and reinclude the device

I tried parameter 250 and it works. It should really have been documented.

darkbasic, 2020-11-10:
@AlCalzone this is the log with parameter 250 enabled:

[zwave-2621.log](https://github.com/zwave-js/node-zwave-js/files/5519144/zwave-2621.log)

petrovcicklemen, 2020-11-10:
@darkbasic My apologies for the outdated manual. I also though the installation manual, provided in the blisters would list the supported command classes, since the last time I looked at them, I think they did. Noted.

AlCalzone, 2020-11-11:
> this is the log with parameter 250 enabled:

That looks good! I'm gonna go ahead and close this issue, since it was a configuration/device issue.

closed 2020-11-11 by AlCalzone