zwave-js/zwave-js #1090: [Qubino Flush Dimmer] Switch command class and dimmer command class don't update each other

darkbasic, 2020-11-03:
When I turn on the light using the switch command class I expect it to update the level of the dimmer command class accordingly and viceversa. Unfortunately that doesn't happen.

- Turning the light on with the switch command class (2-37-0-targetValue)
Before:
[image]
After:
[image]
As you can see the switch on 2-37-0 has been updated to true but the level (2-38-0) hasn't been updated to 99 accordingly.

- Turning the light on with the dimmer command class (setting the 2-38-0-targetValue level to 99)
Before:
[image]
After:
[image]
As you can see the level on 2-38-0 has been updated to 99 but the switch (2-37-0) hasn't been updated to true accordingly.

[zwave-284642.log](https://github.com/zwave-js/node-zwave-js/files/5480705/zwave-284642.log)

```
$ npm start

> zwavejs2mqtt@0.0.0 start /home/niko/devel/zwavejs2mqtt
> node bin/www

  z2m:Store scenes.json not found +0ms
  z2m:Store nodes.json not found +2ms
  z2m:App zwavejs2mqtt version: 0.0.0 +0ms
  z2m:App Application path:/home/niko/devel/zwavejs2mqtt +1ms
  z2m:Mqtt MQTT is disabled +0ms
  z2m:Zwave Connecting to /dev/ttyACM0 +0ms
Logging to file:
/home/niko/devel/zwavejs2mqtt/bin/zwave-284642.log
  z2m:App Listening on port 8091 +0ms
  z2m:App New connection ohv-AeAq0ZkOcAPRAAAA +1s
  z2m:Zwave Zwave driver is ready +2s
  z2m:Zwave Driver ready +1ms
  z2m:Zwave Node added 1 +1ms
  z2m:Zwave Scanning network with homeid: 0xc54055a2 +0ms
  z2m:Zwave Node 1 is alive +59ms
  z2m:Zwave Node 1 doesn't support groups associations +3ms
  z2m:Zwave Node 1: value added 1-32-0-currentValue => undefined +3ms
  z2m:Zwave Node 1: value added 1-32-0-targetValue => undefined +1ms
  z2m:Zwave Node 1 ready: AEON Labs - ZW090 (Z‐Stick Gen5 USB Controller) +1ms
  z2m:Zwave Scan completed +0ms
  z2m:Zwave Network scan complete. Found: 1 nodes +1ms
  z2m:Zwave Node 1: interview completed, all values are updated +1ms
  z2m:App User disconnected ohv-AeAq0ZkOcAPRAAAA +8s
GET / 304 8.727 ms - -
GET /static/css/app.ac70a60bcc407fe11707.css 304 4.830 ms - -
GET /static/js/app.a4eeee65f64587f9e920.js 304 0.763 ms - -
  z2m:App New connection -xi4jW9V5OFHT317AAAB +396ms
GET /static/logo.png 304 0.690 ms - -
  z2m:App Zwave api call: startInclusion [ true ] +22s
  z2m:Zwave Secure inclusion started +29s
  z2m:Zwave Success zwave api call startInclusion true +1ms
  z2m:Zwave Node 2: added +13s
  z2m:Zwave Node added 2 +1ms
  z2m:Zwave Inclusion stopped +5ms
  z2m:Zwave Node 2 is alive +104ms
  z2m:Zwave Node 2: value added: 114-0-manufacturerId => 345 +52ms
  z2m:Zwave Node 2: value added: 114-0-productType => 1 +1ms
  z2m:Zwave Node 2: value added: 114-0-productId => 81 +0ms
  z2m:Zwave Node 2: value added: 134-0-libraryType => 3 +57ms
  z2m:Zwave Node 2: value added: 134-0-protocolVersion => 4.5 +0ms
  z2m:Zwave Node 2: value added: 134-0-firmwareVersions => 3.7 +0ms
  z2m:Zwave Node 2: value added: 94-0-zwavePlusVersion => 1 +490ms
  z2m:Zwave Node 2: value added: 94-0-nodeType => 0 +0ms
  z2m:Zwave Node 2: value added: 94-0-roleType => 5 +1ms
  z2m:Zwave Node 2: value added: 94-0-installerIcon => 7168 +0ms
  z2m:Zwave Node 2: value added: 94-0-userIcon => 7168 +1ms
[... 299 lines cut ...]
  { nodeId: 2, commandClass: 37, endpoint: 0, property: 'targetValue' },
  false
] +43s
  z2m:Zwave Node 2: value updated: 37-0-currentValue true => false +40s
  z2m:Zwave Success zwave api call writeValue  +2ms
  z2m:Zwave Node 2: value updated: 50-0-value-66049 81.2 => 0 +2s
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-66049 0 => 0 +1ms
  z2m:Zwave Node 2: value updated: 50-0-value-66049 0 => 0 +994ms
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-66049 0 => 0 +0ms
  z2m:App Zwave api call: writeValue [
  { nodeId: 2, commandClass: 38, endpoint: 0, property: 'targetValue' },
  99
] +10s
  z2m:Zwave Node 2: value updated: 38-0-currentValue 0 => 0 +7s
  z2m:Zwave Success zwave api call writeValue  +1ms
  z2m:Zwave Node 2: value updated: 38-0-currentValue 0 => 99 +877ms
  z2m:Zwave Node 2: value updated: 50-0-value-66049 0 => 28.4 +1s
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-66049 0 => 0 +1ms
  z2m:Zwave Node 2: value updated: 50-0-value-66049 28.4 => 74.8 +996ms
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-66049 0 => 0 +0ms
  z2m:Zwave Node 2: value updated: 50-0-value-66049 74.8 => 81.5 +995ms
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-66049 0 => 0 +0ms
  z2m:Zwave Node 2: value updated: 50-0-value-65537 64 => 64.1 +2m
  z2m:Zwave Node 2: value updated: 50-0-deltaTime-65537 0 => 0 +0ms
```

AlCalzone, 2020-11-03:
The request does sound plausible, but I'm not sure if it is feasibly to do this automagically. At least I couldn't find anything in the specs I've read that these two CCs should be mapped to each other.

Did this work with the OZW backend? A quick search didn't show any code where that could be happening.

I'm a bit hesitant to do this, because I can already anticipate that there's at least one device out there where Binary Switch and Multilevel Switch do different things.

robertsLando, 2020-11-03:
Could be something configurable using the device config file? Maybe a special option

darkbasic, 2020-11-03:
I think this should be a matter of setting the right association groups or something similar.
Regarding OZW if I recall correctly it used to work with OZW 1.4 but it doesn't work anymore with 1.6. If I include the device with OZW 1.6 it doesn't even update the power (W) value when I turn on/off the light, while if I include it with zwave-js it does update the power (W) at least. So I guess it's something which gets set during inclusion.
Something similar happened to me with the Qubino Flush Shutter in venetian mode, which didn't report the tilt value whenever you changed it using the physical push button with OZW 1.4. The following procedure fixed it: https://community.home-assistant.io/t/integration-of-qubino-zmnhdd1-flush-dimmer/129487/13
With OZW 1.6 it works out of the box, so I guess it's setting those associations automatically.

Could it be something similar for the Flush Dimmer?

This is the manual: https://qubino.com/manuals/Flush_Dimmer.pdf

AlCalzone, 2020-11-03:
> The following procedure fixed it: https://community.home-assistant.io/t/integration-of-qubino-zmnhdd1-flush-dimmer/129487/13

Both OZW and this driver automatically configure lifeline associations during the device interview. Missing reports are almost always related to misconfigured associations.
`zwave-js` automatically uses Multi Channel Assocations (with a target endpoint) if possible, otherwise "normal" node assocations.
For some devices I had to force node associations, because I was getting reports that the reports wouldn't work otherwise. This includes `ZMNHDD`!

I'm not sure if the dashboard already allows you to configure associations - if it does, you could try to change group 1 (should include the controller) from a node association (node 1, NO endpoint) to a multi channel association (node 1, endpoint 0)

I see you opened a similar issue in the OZW repo, I'll watch this to see if the maintainers have any more information about the correct behavior.

darkbasic, 2020-11-03:
> I'm not sure if the dashboard already allows you to configure associations

@robertsLando  does it?

robertsLando, 2020-11-03:
Yes, you can

robertsLando, 2020-11-03:
Control Panel > Groups Tab > Select a Node Then the goups and the target node / endpoint and press add association

AlCalzone, 2020-11-03:
While you're at it, try if adding the controller to group 4 changes anything. This **might** at least give you multilevel switch updates when turning it on with binary switch.

darkbasic, 2020-11-03:
> try to change group 1 (should include the controller) from a node association (node 1, NO endpoint) to a multi channel association (node 1, endpoint 0)

[image]

Those are my associations for node 2, group 1. Unfortunately I cannot manage to remove the existing one, neither with "Remove" nor "Remove all".

```
  z2m:App Zwave api call: removeAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +3m
  z2m:Zwave Assocaitions: Removing associations from Node 2 Group 1: [object Object] +8m
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +350ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 1 => 1 +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1  =>  +0ms
  z2m:Zwave Success zwave api call removeAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +650ms
```

```
  z2m:App Zwave api call: removeAllAssociations [ 2 ] +1m
  z2m:Zwave Assocaitions: Removing all associations from Node 2 +1m
  z2m:Zwave Success zwave api call removeAllAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +1s
```

Any clue?

AlCalzone, 2020-11-03:
This should be a node association:
>  Success zwave api call getAssociations [ { nodeId: 1 } ] 

but the remove call tries to remove an endpoint association `{ nodeId: 1, endpoint: 0 }` (which does not exist)
@robertsLando

robertsLando, 2020-11-03:
I know but shouldn't the removeAllAssociations remove it?

AlCalzone, 2020-11-03:
Probably, yeah. I'd need to see the logfile 😅

darkbasic, 2020-11-03:
Here it is: 
[zwave-359840.log](https://github.com/zwave-js/node-zwave-js/files/5484110/zwave-359840.log)

In the log first I tried to remove it with "Remove", then "RemoveAll".

AlCalzone, 2020-11-04:
Strange, there's only the first remove visible in the log. @robertsLando I'm inclined to believe this is a bug in zwavejs2mqtt. The interview and the log from above
```
  z2m:App Zwave api call: removeAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +3m
  z2m:Zwave Assocaitions: Removing associations from Node 2 Group 1: [object Object] +8m
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +350ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 1 => 1 +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1  =>  +0ms
  z2m:Zwave Success zwave api call removeAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +650ms
```
shows that there is only a node association (last line), but zwavejs2mqtt tries to remove an endpoint association (first line).

robertsLando, 2020-11-04:
> shows that there is only a node association (last line), but zwavejs2mqtt tries to remove an endpoint association (first line).

Yes, you right. Actually when I call `removeAssociations` I let the user specify the association to remove 'manually' (he doesn't select it from the associations list). I will try to improve this in the ui, anyway like I said, shouldn't the removeAllAssociations remove it?

robertsLando, 2020-11-04:
@darkbasic Try to pull last commit https://github.com/zwave-js/zwavejs2mqtt/commit/3def708f8c86c638333ddcd0069a77d93951cb56

AlCalzone, 2020-11-04:
> removeAllAssociations

That is not part of zwave-js
https://zwave-js.github.io/node-zwave-js/#/api/controller?id=managing-associations

robertsLando, 2020-11-04:
> That is not part of zwave-js

:rofl: :rofl: Let me go take my :coffee: 

Is `removeNodeFromAllAssocations` the same?

robertsLando, 2020-11-04:
Or maybe it means to remove the node from the other associations, not to remove associations of that node. Tricky :confounded:

AlCalzone, 2020-11-04:
Yep, that has an entirely different meaning :)
When a node is removed from a network, it should be removed from all associations. This is what this method is for.

You could just call removeAssociations with `getAssociations(nodeId).get(groupId)`.

robertsLando, 2020-11-04:
@AlCalzone Fixed: 

https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js#L1158

https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js#L1184

darkbasic, 2020-11-04:
@AlCalzone   `z2m:Zwave Error while adding associations to 2: Node 2, group 1 does not support multi channel associations! +1ms`

[zwave-3679.log](https://github.com/zwave-js/node-zwave-js/files/5486588/zwave-3679.log)

```
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +12s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +37s
  z2m:App Zwave api call: removeAssociations [ 2, 1, [ { nodeId: 1 } ] ] +11s
  z2m:Zwave Assocaitions: Removing associations from Node 2 Group 1: [object Object] +11s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +389ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 1 =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1  =>  +1ms
  z2m:Zwave Success zwave api call removeAssociations  +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +75ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1  =>  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +535ms
  z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +1m
  z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +1m
  z2m:Zwave Error while adding associations to 2: Node 2, group 1 does not support multi channel associations! +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +1s
```

@robertsLando `z2m:Zwave Assocaitions: Removing associations from Node 2 Group 1: [object Object] +11s`
typo is Associations plus json needs to be serialized

darkbasic, 2020-11-04:
@robertsLando I also don't like the fact that once you've changed the placeholder channel ID you can't select the empty one anymore

darkbasic, 2020-11-04:
@AlCalzone I also can't add the controller to group 4

[zwave-3679.log](https://github.com/zwave-js/node-zwave-js/files/5486624/zwave-3679.log)

```
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +2m
  z2m:Zwave Success zwave api call getAssociations [] +2m
  z2m:App Zwave api call: addAssociations [ 2, 4, [ { nodeId: 1 } ] ] +5s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 4 of  Node 2 +5s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +387ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4  => 1 +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4  =>  +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +61ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4 1 =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4  =>  +0ms
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +550ms
```

robertsLando, 2020-11-04:
> I also don't like the fact that once you've changed the placeholder channel ID you can't select the empty one anymore

What?

robertsLando, 2020-11-04:
> z2m:Zwave Error while adding associations to 2: Node 2, group 1 does not support multi channel associations! +1ms

That's really strange as it is has multiChannel flag to true

darkbasic, 2020-11-04:
> What?

Nevermind, I saw the slider and I didn't notice you can actually delete the content of the channel ID field.

Anyway, wouldn't it be better to have a "Delete" button near each "Current associations" item instead of the current way?

robertsLando, 2020-11-04:
Anyway, wouldn't it be better to have a "Delete" button near each "Current associations" item instead of the current way?

Yes it would be more user friendly, I'm just lazy :rofl:

darkbasic, 2020-11-04:
Anyway I've just tried to add a multichannel node association to group 4 and now both the node association and the multichannel one appeared:

```
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +11s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +3m
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +8m
  z2m:Zwave Success zwave api call getAssociations [] +8m
  z2m:App Zwave api call: addAssociations [ 2, 4, [ { nodeId: 1, endpoint: 0 } ] ] +16s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 4 of  Node 2 +16s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +319ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4  => 1 +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4  => [object Object] +0ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 }, { nodeId: 1, endpoint: 0 } ] +684ms
```

[zwave-3679.log](https://github.com/zwave-js/node-zwave-js/files/5486699/zwave-3679.log)

Something strange is going on here...

robertsLando, 2020-11-04:
@darkbasic I have submitted an ui improvment for associations that you will like :) pull

robertsLando, 2020-11-04:
> Something strange is going on here...

Zwave is good, zwave devices are strange :rofl:

AlCalzone, 2020-11-04:
> z2m:App Zwave api call: addAssociations [ 2, 4, [ { nodeId: 1 } ] ] +5s
>   z2m:Zwave Assocaitions: Adding Node 1 to Group 4 of  Node 2 +5s
>   z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 [...]

Looks like the device did not accept this change. `addAssociations` performs an add command and then requests the updated list. Wonder why the node association showed up on the next attempt though, maybe the device needed time to update?

> Zwave is good, zwave devices are strange 🤣

yeah and Z-Wave associations are a f*cking mess.

> That's really strange as it is has multiChannel flag to true
If the config file forbids multi channel associations then the driver respects that. You could try and change your config file locally:
In `node_modules/@zwave-js/config/config/devices/0x0159/zmnhdd.json` comment out the lines with `noEndpoint: true`. Just be warned that this could cause the unsolicited updates to stop working.

darkbasic, 2020-11-04:
> I have submitted an ui improvment for associations that you will like :) pull

Unfortunately the backend is doing something fishy, because I had both a node association and a group association and after deleting one of them it deleted both...

```
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +10s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 }, { nodeId: 1, endpoint: 0 } ] +13s
  z2m:App Zwave api call: removeAssociations [ 2, 4, [ { nodeId: 1 } ] ] +1h
  z2m:Zwave Assocaitions: Removing associations from Node 2 Group 4: [{"nodeId":1}] +1h
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +388ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4 1 =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4 [object Object] => [object Object] +1ms
  z2m:Zwave Success zwave api call removeAssociations  +0ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +16ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4 [object Object] =>  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +594ms
```
> Wonder why the node association showed up on the next attempt though, maybe the device needed time to update?

I don't think so:

```
  z2m:App Zwave api call: addAssociations [ 2, 4, [ { nodeId: 1 } ] ] +2m
  z2m:Zwave Assocaitions: Adding Node 1 to Group 4 of  Node 2 +2m
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +389ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4  => 1 +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4  => [object Object] +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +45ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4 1 =>  +2ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4 [object Object] =>  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +561ms
  z2m:App User disconnected ndt4_zcnbK4pPDA0AAAC +10s

(waited some time)

  z2m:App Zwave api call: getAssociations [ 2, 4 ] +12s
  z2m:Zwave Success zwave api call getAssociations [] +23s
```

darkbasic, 2020-11-04:
> In node_modules/@zwave-js/config/config/devices/0x0159/zmnhdd.json comment out the lines with noEndpoint: true

It looks like despite the error it already created the group association, because after commeting them out it shows both:

```
  z2m:App Zwave api call: getAssociations [ 2, 4 ] +7s
  z2m:Zwave Success zwave api call getAssociations [] +12s
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +2s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 }, { nodeId: 1, endpoint: 0 } ] +2s
```

@robertsLando I tried the new "Remove all", but it doesn't remove all the associations:

```
  z2m:App Zwave api call: removeAllAssociations [ 2 ] +25s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +26s
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1 [object Object] =>  +2ms
  z2m:Zwave Assocaitions: Removed 2 associations from Node 2 group 1 +4ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-2 16 => 16 +337ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-2  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-2  =>  +0ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 2 +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +252ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-3 16 => 16 +93ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-3  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-3  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 3 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-4 16 => 16 +337ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-4  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-4  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 4 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-5 16 => 16 +326ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-5  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-5  =>  +0ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 5 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-6 16 => 16 +344ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-6  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-6  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 6 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-7 16 => 16 +336ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-7  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-7  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 7 +0ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-8 16 => 16 +352ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-8  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-8  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 8 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-9 16 => 16 +364ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-9  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-9  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 9 +2ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-10 16 => 16 +344ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-10  =>  +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-10  =>  +1ms
  z2m:Zwave Assocaitions: Removed 0 associations from Node 2 group 10 +1ms
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-11 16 => 16 +321ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-11  =>  +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-11 [object Object] =>  +0ms
  z2m:Zwave Assocaitions: Removed 1 associations from Node 2 group 11 +1ms
  z2m:Zwave Success zwave api call removeAllAssociations  +0ms
```

AlCalzone, 2020-11-04:
Hmm that does indeed look very fishy.

> z2m:App Zwave api call: removeAllAssociations [ 2 ] +25s
>   z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +26s
>   z2m:Zwave Node 2: value updated: 142-0-nodeIds-1  =>  +1ms
>   z2m:Zwave Node [...]

This looks like an endpoint association was removed (2nd to last line)

>   z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
>   z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +252ms

And this is really weird. @robertsLando does z2m cache these results somehow? If not, this does seem like we have a node association created by the `Association CC` and nothing for `Multi Channel Association CC`. Removing that only uses `Multi Channel Association CC`, so the `Association CC` association is probably unaffected. 🤔

robertsLando, 2020-11-04:
> does z2m cache these results somehow? 

I only cache groups, associations are fetched on every request when the user selects a group

robertsLando, 2020-11-04:
@darkbasic Please use `dev` branch, I will add edits there

AlCalzone, 2020-11-04:
Ok, then I'll have to investigate

robertsLando, 2020-11-04:
@AlCalzone Add `investigating` flag :rofl:

AlCalzone, 2020-11-04:
@darkbasic could you share your cache file (`something.values.jsonl`) please? If @robertsLando didn't change the location, it should be in `node_modules/zwave-js/cache`

robertsLando, 2020-11-04:
It's in "store" folder now

darkbasic, 2020-11-04:
> It's in "store" folder now

Yes, it's in store:

```
zwavejs2mqtt]$ git status --ignored
Sul branch master
Il tuo branch è aggiornato rispetto a 'origin/master'.

File ignorati:
  (usa "git add -f <file>..." per includere l'elemento fra quelli di cui verrà eseguito il commit)
        bin/zwave-10331.log
        bin/zwave-16501.log
        bin/zwave-3679.log
        dist/
        node_modules/
        store/d743ed1a.json
        store/d743ed1a.metadata.jsonl
        store/d743ed1a.values.jsonl
        store/settings.json
```

@AlCalzone here it is, I also attached latest log

[d743ed1a.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/5488725/d743ed1a.values.jsonl.txt)

[zwave-16501.log](https://github.com/zwave-js/node-zwave-js/files/5488734/zwave-16501.log)

AlCalzone, 2020-11-04:
Yup, that confirms my suspicion in https://github.com/zwave-js/node-zwave-js/issues/1090#issuecomment-721699693

AlCalzone, 2020-11-04:
I'll probably have some time for a fix later tonight. Will let you know.

AlCalzone, 2020-11-04:
@darkbasic https://github.com/zwave-js/node-zwave-js/pull/1094 should enable the controller to remove both the "normal" and the multi channel association. Can you test this from github or should I create an alpha release for you?

Testing from github should work as follows:
* clone the repo
* in the cloned folder run:
  * `npx lerna bootstrap`
  * `npm run build`
  * `npx lerna exec -- npm link`
* in the z2m folder run `npm link zwave-js`

darkbasic, 2020-11-04:
I'm using zwavejs2mqtt from the `dev` branch and node-zwave-js from the `association-delete-all` branch.
I added a node association and then tried to add a multi-channel association, but it doesn't show up, is it normal?

```
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [] +676ms
  z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1 } ] ] +8s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +8s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +320ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 [] => [ 1 ] +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1 [] => [] +0ms
  z2m:Zwave Success zwave api call addAssociations  +2ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +678ms
  z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +4s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +4s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +284ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 [ 1 ] => [ 1 ] +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1 [] => [] +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1 } ] +719ms
```

darkbasic, 2020-11-04:
By the way, did you consider using yarn workspaces?

AlCalzone, 2020-11-04:
> z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +4s
>   z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +4s
>   z2m:Zwave Node 2: value updated: [...]

The multi channel association is added as a node association. Not sure if this is due to the config file forcing a node association or if z2m is mixing this up? @robertsLando 

> By the way, did you consider using yarn workspaces?

I have never worked with yarn. `npm@7` is also getting a workspace feature, but I'm not sure yet what this changes exactly.

darkbasic, 2020-11-04:
> Btw, can I interpret the above log as confirmation that the association-delete-all does what it is supposed to?

I've been able to remove a node association or multichannel association, but I couldn't try to delete them when they where both present, because I can only have a node **or** a multichannel now.

@robertsLando I've found a new bug: if you edit the channel id field (for example because you added a multi channel association) and then you clear it up, you won't be able to create new associations until you refresh the page:
```
z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: '' } ] ] +14s
```
It basically targets the "empty string" endpoint.

AlCalzone, 2020-11-04:
> I've been able to remove a node association or a multi channel association

That is enough for me. At least it seems like your phantom association is gone.

robertsLando, 2020-11-04:
I dunno what you mean with "mixing it up" but I'm just calling the controller methods as you see in the logs, I don't cache anything about associations 😕 

About the empty string bug I will fix it tomorrow thanks

AlCalzone, 2020-11-04:
I mean this specifically:

> z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +4s
> z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of Node 2 +4s

The call in the first line should add an endpoint association, the 2nd line looks like it tries to add a node association. Or is this just a display issue? I'm pretty sure that **adding** endpoint associations works in zwave-js.

robertsLando, 2020-11-04:
> Or is this just a display issue

I think I forgot to add the endpoint to the log message

AlCalzone, 2020-11-04:
The thing is it also does add a node association, not an endpoint association.

robertsLando, 2020-11-04:
Check it here: https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js 

Line 1097, sorry I'm from the phone and I cannot find how to give you the direct line link

AlCalzone, 2020-11-04:
Hm, this all doesn't make sense. @darkbasic I think I need one more logfile from your attempt to add the association.

darkbasic, 2020-11-05:
Here it is:

[zwave-121980.log](https://github.com/zwave-js/node-zwave-js/files/5492648/zwave-121980.log)

```
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +9s
  z2m:Zwave Success zwave api call getAssociations [] +14s
  z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1 } ] ] +13s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +13s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +388ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 [] => [] +0ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1 [] => [ { nodeId: 1, endpoint: 0 } ] +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1, endpoint: 0 } ] +612ms
  z2m:App Zwave api call: addAssociations [ 2, 1, [ { nodeId: 1, endpoint: 0 } ] ] +4s
  z2m:Zwave Assocaitions: Adding Node 1 to Group 1 of  Node 2 +4s
  z2m:Zwave Node 2: value updated: 142-0-maxNodes-1 1 => 1 +367ms
  z2m:Zwave Node 2: value updated: 142-0-nodeIds-1 [] => [] +1ms
  z2m:Zwave Node 2: value updated: 142-0-endpoints-1 [ { nodeId: 1, endpoint: 0 } ] => [ { nodeId: 1, endpoint: 0 } ] +1ms
  z2m:Zwave Success zwave api call addAssociations  +1ms
  z2m:App Zwave api call: getAssociations [ 2, 1 ] +1s
  z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1, endpoint: 0 } ] +631ms
```

AlCalzone, 2020-11-05:
There's something weird happening with the device...

During the interview, the controller tries to assign itself as a multi channel association. The device first acknowledges this but then sends another report a few milliseconds later where the association is missing again.

That's why the first get above seems to yield no associations
> z2m:App Zwave api call: getAssociations [ 2, 1 ] +9s
> z2m:Zwave Success zwave api call getAssociations [] +14s

Then you try to add a node association, but the device already has the multi channel association from before. Since the limit of associations in the group is 1, the node association is not added. The GET which should confirm the add now shows the multi channel association that was added earlier.

Adding the multi channel association again is effectively a no-op

petrovcicklemen, 2020-11-09:
**> Both OZW and this driver automatically configure lifeline associations during the device interview. Missing reports are almost always related to misconfigured associations.**
A general rule for Qubino Flush devices is, that the ordinary lifeline has to be configured (using ASSOCIATION_SET(groupId=1, nodeId=1)), whereas the multi channel lifeline association must be configured (using MULTI_CHANNEL_ASSOCIATION_SET(groupId=1, mcnodeId=1, endpoint=0), if CC_MULTI_CHANNEL is listed, which means the temperature sensor is connected or an additional input is configured). 

The majority of Qubino devices are, in their default configurations, single channel devices, so they require an ordinary association to be configured. They expect the mc lifeline to be configured only, when additional endpoints are enabled. 

These two mentioned commands are enough to configure unsolicited reporting.

**> `zwave-js` automatically uses Multi Channel Assocations (with a target endpoint) if possible, otherwise "normal" node assocations.**
Like explained above, some Qubino devices will not send any unsolicited values, if a mc lifeline is configured in their default configurations, where they are represented as single channel (root endpoint only) devices.

AlCalzone, 2020-11-09:
> Like explained above, some Qubino devices will not send any unsolicited values, if a mc lifeline is configured in their default configurations, where they are represented as single channel (root endpoint only) devices.

Just to make sure I understand: Do these devices support MC associations?  
If I remember correctly from reading the specs (but I might have mixed this up), Multi Channel Associations should be preferred over Associations starting with MC version 3.

[29 more replies]

closed 2020-11-19 by AlCalzone