zwave-js/zwave-js #1100: Support for 700 series / Gen7 stick and devices

apaperclip, 2020-11-05:
I didn't see any mention of other inquires or support for 700 series / gen 7 devices.  Aeotec as a stick and a few devices and it looks like Inovelli has or will have some too.

https://aeotec.freshdesk.com/support/solutions/articles/6000225289-gen7-engineering-sheets
https://aeotec.freshdesk.com/support/solutions/articles/6000230552

robertsLando, 2020-11-05:
Some other useful links: 

https://www.silabs.com/documents/public/user-guides/INS12350-Serial-API-Host-Appl.-Prg.-Guide.pdf
https://www.silabs.com/documents/public/release-notes/SRN13181.pdf
https://www.silabs.com/documents/public/application-notes/APL14440.pdf

AlCalzone, 2020-11-23:
In general I would assume that the serial API stays backwards-compatible. If not, please mention it here.

If there is any new functionality which is required, please open separate issues for each feature.

fxl90, 2020-12-31:
@AlCalzone just want to make sure I provide this feedback in the right place. Zooz released their 700 series stick. It currently shows up as unknown manufacturer and product. What needs to be done to support this? I have it on hand and can help provide as much info as necessary. Link to the stick below.

[Zooz ZST10 700](https://www.thesmartesthouse.com/products/zooz-usb-700-series-z-wave-plus-s2-stick-zst10-700)

AlCalzone, 2021-01-01:
@fxl90 The Z-Wave protocol has currently no means of discovering the name of a product. We need configuration files for that, which map the combination of manufacturer id, product id and product type (those are discoverable) to a product name and potentially config options, associations and compatibility settings.

The [docs](https://zwave-js.github.io/node-zwave-js/#/development/config-files) describe how such a config file looks like. For a controller stick, you just need the bare minimum - here's an example (that needs to be changed to match your device):

```jsonc
// Sigma Designs (Former Zensys) HUSBZB-1
// QuickStick Combo
{
    "manufacturer": "Sigma Designs (Former Zensys)",
    "manufacturerId": "0x0000",
    "label": "HUSBZB-1",
    "description": "QuickStick Combo",
    "devices": [
        {
            "productType": "0x0001",
            "productId": "0x0001"
        }
    ],
    "firmwareVersion": {
        "min": "0.0",
        "max": "255.255"
    }
}
```

fxl90, 2021-01-01:
@AlCalzone that makes sense. I assumed it was a compatibility issue since I was not able to include devices successfully using the stick.

Is the configuration file needed in order for the stick to work properly? Or should it still function normally without an existing config file?

In the mean time I will reach out to Zooz to get the needed documentation to add the new ZST10 700 stick.

AlCalzone, 2021-01-01:
> since I was not able to include devices successfully using the stick.

I would assume that it should work normally. How does this manifest itself? Do you have a log? The product type etc should also be visible in the log.

fxl90, 2021-01-02:
Just for reference:
- node 1 = Zooz ZST10 700 
- node 3 = Fibargroup FGWPB-121

[image]

AlCalzone, 2021-01-02:
I'll need the log that zwave-js creates

fxl90, 2021-01-02:
@robertsLando I'm using the zwavejs2mqtt docker image. I enabled the zwave logging. Would the zwave-js log @AlCalzone needs be the JSON files that are created in the store directory?

AlCalzone, 2021-01-02:
It should have a "log" extension. Depending on the version it is either in zwavejs2mqtt's bin directory or in the store directory

fxl90, 2021-01-02:
Ahh well that makes sense why I don't see the logs. I'm not mapping the bin directory, and I'm using the dev version of the docker image which must not be storing the log in the store directory.

fxl90, 2021-01-02:
@AlCalzone here is the log file:

[zwave-1.log](https://github.com/zwave-js/node-zwave-js/files/5760746/zwave-1.log)

AlCalzone, 2021-01-02:
This is Node 1:
```
manufacturer ID:     0x00
product type:        0x04
product ID:          0x04
```

The responses from Node 3 seem to come in an unsupported message, which is why you're not seeing any progress:
```
18:50:28.682 DRIVER « [REQ] [undefined]
                        payload: 0x000103087205010f1401200000c1
18:50:28.683 DRIVER   handling request undefined (168)
```
I'll investigate

AlCalzone, 2021-01-02:
That was a variant of the normal ApplicationCommandRequest specifically for the Bridge API. #1276 should fix this - full support of the Bridge API including virtual slave nodes is a completely different beast. Time will tell if it is worth going down that rabbit hole...

fxl90, 2021-01-03:
Awesome, thanks for looking into it. Let me know if I can test or provide any other logs.

AlCalzone, 2021-01-03:
You could try out the `bridge-application-command` branch. If you're using zwavejs2mqtt with Docker, there should be a build argument to use that branch instead of master.

jcam, 2021-01-05:
Just wanted to say, @AlCalzone it is awesome how quickly you're working on this! Patreon incoming. I'll have a Zooz 700 stick in a few days, just waiting on the post. If it hasn't been merged by the time it arrives, I will report back my findings

fxl90, 2021-01-08:
Well @AlCalzone, either I didn't build the docker image correctly or it's still not working. I got the same result. I would lean towards a bad docker build on my side. I might wait to test this again once it's merged into an official z2m image just in case.

jcam, 2021-01-08:
@fxl90 you're using the Zooz 700 stick?
I didn't get anywhere near as far as you...
02:33:46.256 DRIVER   version 5.7.0                                                                  
02:33:46.257 DRIVER                                                                                  
02:33:46.258 DRIVER   starting driver...                                                             
02:33:46.290 DRIVER   opening serial port /dev/ttyUSB0                                               
02:33:46.345 DRIVER   serial port opened                                                             
02:33:46.348 SERIAL .. [NAK]                                                                   (0x15)                                                                              
02:33:47.856 DRIVER   loading configuration...                                                                                                                                     
02:33:50.153 DRIVER   beginning interview...                                                                                                                                       
02:33:50.156 DRIVER   added request handler for AddNodeToNetwork (0x4a)...                                                                                                         
                      1 registered                                                                                                                                                 
02:33:50.157 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...                                                                                                    
                      1 registered                                                                   
02:33:50.159 DRIVER   added request handler for ReplaceFailedNode (0x63)...                          
                      1 registered                                                                   
02:33:50.161 CNTRLR   beginning interview...                                                         
02:33:50.162 CNTRLR   querying version info...                                                       
02:33:50.221 SERIAL .. 0x01030015e9                                                         (5 bytes)
02:33:50.225 DRIVER .. [REQ] [GetControllerVersion]                                                                                                                                
02:33:51.243 CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i                                                                               
                      n 100 ms.                                                                                                                                                    
02:33:51.350 SERIAL .. 0x01030015e9                                                         (5 bytes)                                                                              
02:33:51.353 DRIVER .. [REQ] [GetControllerVersion]                                                                                                                                
02:33:52.367 CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i                                                                               
                      n 1100 ms.                                                                     
02:33:53.475 SERIAL .. 0x01030015e9                                                         (5 bytes)
02:33:53.478 DRIVER .. [REQ] [GetControllerVersion]                                                 
02:33:54.507 DRIVER   Failed to initialize the driver: Timeout while waiting for an ACK from the con
                      troller                                                                      
02:33:54.515 DRIVER   destroying driver instance...

jcam, 2021-01-08:
I rebased and built `bridge-application-command` and am no longer getting the unhandled request 168 errors. At pairing, I was able to see the manufacturer and model information for my device, but after a restart it no longer reports the info, and running the request info command has no visible effect.
The device (a Fibaro door sensor) was successfully added to Home Assistant, and I can see the state change for door open/close. And it is blazing fast.

Any info I can gather for why the info display isn't showing up?

The bin/zwave-1.log is no longer created in this build, and there are no logs showing up in the store folder either.

jcam, 2021-01-08:
Also, I tried adding a controller config to address the startup error Unknown manufacturer 0 - Unknown product 4 (4) but there was no effect, still get the error.
My branch is over here: https://github.com/jcam/node-zwave-js/tree/bridge-application-command

jcam, 2021-01-08:
@fxl90 if you want to try it, it's on docker at docker.io/hikeit/zwavejs2mqtt:zwave700

AlCalzone, 2021-01-08:
> Any info I can gather for why the info display isn't showing up?

@robertsLando I think that's a zwavejs2mqtt bug. I'm seeing this pretty often lately. zwave-js detects the device info but somehow they don't make it into the UI.

> there are no logs showing up

The way the log is configured now is not really compatible with the way zwavejs2mqtt uses the driver. We're refactoring the logging in zwave-js and should have a working copy by tomorrow. Then we'll see more.

jcam, 2021-01-08:
Okay cool, I'll rebuild once that's ready. Once I can see more detailed logs and add more of (or, the rest of) the devices from my ST zwave mesh, I'll feel better about giving this PR the thumbs up. Looking good so far, though :)

Any idea why I might still be getting Unknown manufacturer 0 - Unknown product 4 (4) from the controller?

jcam, 2021-01-09:
@AlCalzone / @robertsLando do either of you know... the below issue, where this battery powered device doesn't ever get nodeinfo after initial pairing, is this a bug in the bridge-application-command code, in zwave-js in general, or in zwave2mqtt?

[image]

AlCalzone, 2021-01-09:
@jcam That might be #1307

PR #1276 (2021-01-10): feat(driver): handle BridgeApplicationCommandRequests
For now this seems to be enough to support the 700 series devices

fixes: #1100
closed 2021-01-10 by AlCalzone