zwave-js/zwave-js #1105: Node property to indicate if the node is being interviewed / 'Failed' InterviewStatus

RoboPhred, 2020-11-08:
Currently, when a node fails its interview, there is no indication to the outside world beyond the "interview failed" event, and interviewStage remains on the failing interview step.  I would like to be able to check if the interview of a node has failed without relying on capturing the event. Adding a "Failed" stage to InterviewStage would be useful in this regard.

AlCalzone, 2020-11-09:
Good point - however the `InterviewStage` is used to remember how far the interview has progressed, so that some steps can be skipped when repeating an interview. Even after a failed interview it is not necessary to repeat all steps - the proposed solution would discard this information.

I've been planning to [rewrite the interview](https://github.com/zwave-js/node-zwave-js/issues/903) process as a whole. My preferred approach will avoid the problem of multiple concurrent interview processes. I'll be adding your requirement of better transparency there.

AlCalzone, 2020-11-09:
duplicate of #903

closed 2020-11-09 by AlCalzone